pymodbus-dev/pymodbus #73: RemoteSlaveContext not supporting unit

pakerfeldt, 2015-02-23:
Is there a reason why, specifically setValues and getValues in, RemoteSlaveContext doesn't supported **kwargs to enable for the unit (what I typically call slave id) parameter? I modified this to accommodate for a scenario where I have a bottle instance backed by a RemoteSlaveContext which communicates with multiple slaves. Should I send a PR for this or is this supposed to be handled in some other way?

kevinee, 2015-03-09:
Hi Patrik, I am new here but am having the same issue as you. I am interested in using pymodbus/example/contrib/serial-forwarder.py as a bridge from modbus-tcp to modbus-rtu(485). The example almost works for this purpose, except the unit-id is not passed from the tcp side to the rtu side. As you pointed out (https://groups.google.com/forum/#!topic/pymodbus/HkE_5gQ-gT8) RemoteSlaveContext in remote.py doesn't support unit_id's.

I'm not fluent in Python, but as far as I can tell the datastore model doesn't fit well with our scenarios. It allows one datastore for all requests or a datastore for each unit_id. I need a one datastore (i.e. the rtu-serial port) which is accessed for all unit_ids, with the unit_id being passed through to the serial port.

I'd like to help with this, but I think we both need some guidance from the more experienced people here. Is this the correct way to build a tcp-rtu bridge (or REST api - rtu bridge) with pymodbus?

pakerfeldt, 2015-03-09:
I decided to base my project on NodeJS instead, using a completely different library. So to me, personally, this is no longer an issue. However, here's a patch on the remote.py changes I did to make it handle unit ids. 

https://gist.github.com/pakerfeldt/9671aaf6e39e3c7912ac

bashwork, 2015-04-01:
There was no reason, it was just an omission. I will incorporate your patch. Thanks!

commit fec7ac37 (2015-04-02): closes #73
closed 2015-04-02 by bashwork