pymodbus-dev/pymodbus #75: Modbus server. How to generate error?

SergeyDubovitsky, 2015-03-16:
I have created a server example: http://pymodbus.readthedocs.org/en/latest/examples/asynchronous-server.html  I need a timer to generate an error Modbus client (eg 81 8F)? How can this be implemented?

xvart, 2015-03-24:
you need to ask the main pymodbus developer

bashwork, 2015-04-01:
I'm sorry I don't understand the question. Can you explain more?

marokamaro, 2017-07-14:
I suppose I have a similar problem as an author of the thread. I would like to create some registers which will respond with an error (bad frame format or only incorrect CRC). How can I do something like this? I start the server with StartSerialServer function.

dhoomakethu, 2021-03-06:
For any one who wants this featuere, it is part of `pymodbus.server` console script when you install pymodbus 2.5.0 and above and is available only with python3.6 and above.

```
pymodbus git:(dev) ✗ pymodbus.server run --modbus-server tcp --modbus-port 5020 --modbus-unit-id 1
Pymodbus server...

SERVER > help
Available commands:
clear                                        Clears screen
manipulator                                  Manipulate response from server.
Usage: manipulator response_type=|normal|error|delayed|empty|stray
	Additional parameters
		error_code=<int>
		delay_by=<in seconds>
		clear_after=<clear after n messages int>
		data_len=<length of stray data (int)>

	Example usage:
	1. Send error response 3 for 4 requests
	   manipulator response_type=error error_code=3 clear_after=4
	2. Delay outgoing response by 5 seconds indefinitely
	   manipulator response_type=delayed delay_by=5
	3. Send empty response
	   manipulator response_type=empty
	4. Send stray response of lenght 12 and revert to normal after 2 responses
	   manipulator response_type=stray data_len=11 clear_after=2
	5. To disable response manipulation
	   manipulator response_type=normal
SERVER >

```

dhoomakethu, 2021-03-06:
Note, the documentation is still not updated.

closed 2021-03-06 by dhoomakethu