pymodbus-dev/pymodbus #76: RTU sync client always waits for timeout (with possible solution)

kevinee, 2015-03-16:
The RTU serial port client always waits for the port timeout value, e.g.;
client = ModbusClient(port = 'dev/ttyUSB0', ..... , timeout = 1)
will cause every request to take one second size the read size is always set to 1024 bytes, larger than an actual response. One possible solution is to read the first few bytes of the response and from that determine the complete response size. This is possible for all the common function codes.

This is a modified /client/[sync.py](https://gist.github.com/kevinee/713687d8c00c2bdc2722)
All changes in last 128 lines of the file. Only affects RTU clients.
Only tested for holding register reads, which is all my hardware is set up for right now.

If the response is corrupted somehow the client will either return after reading too few bytes or timeout waiting for too many. A buffer flush would likely be required to recover from reading too few bytes, not sure where to implement that (I'm EEng not Python coder unfortunately).

Cougar, 2015-03-16:
I proposed patch that I have successfully used myself in issue #55

dhoomakethu, 2017-11-26:
Fix already available in pymodbus 1.3 and above

closed 2017-11-26 by dhoomakethu