pymodbus-dev/pymodbus #84: Stop async Server

Loris1123, 2015-05-18:
I didn't find a way to stop an async TCP Server. I need this to test the server with different configuration in a single script.

To stop the async server I added the following code to async.py:

    def StopTcpServer(is_mainthread=True):
        from twisted.internet import reactor
        if is_mainthread:
            reactor.stop()
        else:
            reactor.callFromThread(reactor.stop)

It seems like it is working for me. Maybe you should include something similar to pymodbus.

tmcolby, 2017-04-06:
I needed the ability to stop the server also.  Thank you, Loris1123.  Your solution works for me as well.  I agree that it would be nice to have this incorporated into pymodbus.

closed 2017-11-24 by dhoomakethu

MM8704, 2018-07-23:
Hello, I have a question if I have a button that stops the server with the Stopserver () function, how can I start it again without closing the program?

dhoomakethu, 2018-07-24:
The `reactor` is not restartable . So I am not sure if there is way to restart it again with out calling `StartTcpServer` method.

MM8704, 2018-07-24:
Thanks for your answer dhoomakethu, I have to implement a method able to change the IP address of the server, without closing the main program.

Initially try to stop the server and start it again with the StarTcpServer method with a new Ip address, but when you stop the reactor and start it again, an error appears. Any recommendation?.

dhoomakethu, 2018-07-24:
@MM8704 could you share the error and any code to reproduce your use case?

MM8704, 2018-07-24:
@dhoomakethu sure, is a small example of a server mounted on kivy

Please download it in google drive: 
https://drive.google.com/open?id=10_VHhRWViiBYwp_OLX20VHzIF8Jo7WVk

dhoomakethu, 2018-07-24:
You can take a look at [modbus-simulator](https://github.com/riptideio/modbus-simulator. It is also based on pymodbus and kivy supports both TCP and Serial servers (uses pymodbus sync server and threads though). You can find the backend code [here](https://github.com/riptideio/modbus-simulator/blob/master/modbus_simulator/utils/pymodbus_server.py)

MM8704, 2018-07-26:
Hi @dhoomakethu, without any positive, have you found something?

dhoomakethu, 2018-07-27:
@MM8704 sorry but your code throws me error while running
`     on_press: Server_TCP.stop(self)
 TypeError: unbound method stop() must be called with Server_TCP instance as first argument (got ToggleButton instance instead)`. I would suggest to use the modbus-simulator mentioned above for the time being. I will take a closer look in to your code when I get some free time.

MM8704, 2018-07-27:
@dhoomakethu I tried to execute the code that you mentioned but I could not (the launcher file does not really know how to execute it), I just read the code to implement it in my program ... without results.

PR #240 (2017-11-28): Test dev

Fixes #84 , #135, #139, #233, #107,  #190