pymodbus-dev/pymodbus #85: pymodbus.client.async.ModbusSerialClient seems to be missing

sjlongland, 2015-05-26:
Hi,

I'm in the process of writing an asynchronous Modbus interface using pymodbus, I went to try out some of the functional tests, namely this one:

https://github.com/bashwork/pymodbus/blob/46c900832aa35a00955f2fd12537f4963687bb87/examples/functional/asynchronous-rtu-client.py

I get the following error:
    RC=0 stuartl@rikishi ~/vrt/projects/metermaster/deps/pymodbus/examples/functional $ python asynchronous-rtu-client.py 
    Traceback (most recent call last):
      File "asynchronous-rtu-client.py", line 3, in <module>
        from pymodbus.client.async import ModbusSerialClient as ModbusClient
    ImportError: cannot import name ModbusSerialClient

Where do I find the asynchronous `ModbusSerialClient`?

Cougar, 2015-12-06:
AFAIK there is none. I just opened case #108 where I implemented one possible solution.

sjlongland, 2015-12-06:
Ahh, using Tornado too, I like! :-)

closed 2017-12-22 by dhoomakethu