pymodbus-dev/pymodbus #91: pip installs old version (with bugs)

vthielen, 2015-06-26:


vthielen, 2015-06-26:
When using pip-3.2 on beaglebone to check out pymodbus3, it installs the 1.0.0 version of the software which has bugs.
I copied from the githup python3 branch and installed with: "python3 setup.py install" to get the latest version

Humphreybas, 2015-12-31:
I had other problems with pip, described below, but this is my workaround and I think would work for you as well:
sudo pip3 install https://github.com/bashwork/pymodbus/archive/python3.zip
Just point pip to the zip file of the python3 branch.

I first tried 
sudo pip3 install pymodbus
but it failed to install since I was installing the python 2 version apparently. Then I used the workaround above.
But installing the python2 package from pip3 is also unexpected behaviour (although this could be more of a pip issue) and it would be nice to have a remark about the pymodbus3 package in the readme. But indeed when I tested pip3 install pymodbus3 I also got the 1.0.0 version.

closed 2017-06-28 by dhoomakethu