pymodbus-dev/pymodbus #95: PYMOBUS returning none

tabishibly, 2015-07-22:
I am using pymodbus library to set up communication over tcp. When I try reading registers, it is returning "none" How do i solve this?

WillyKaze, 2016-02-12:
Just hit the same bug. I'll investigate this.

WillyKaze, 2016-02-12:
@tabishibly Which branch are you using ?

oetcoleman, 2016-02-13:
I'm getting this on the python3 branch.  Suspect it might be to do with the zope.interface dependency not installing correctly.

ellepdesk, 2016-02-22:
I'm also seeing this for ModbusSerialClient, read_holding_registers returns None for python3/pymodbus3, and a correct response for pyfon2.7/pymodbus

WillyKaze, 2016-03-09:
Thanks for the answer, I since ported my code to use pylibmodbus. I didn't run any benchmarks between this two libs but libpymodbus with ctypes (not cffi) seems quite reactive and easy to cross-compile for BeagleBone Black

closed 2017-10-03 by dhoomakethu