pymodbus-dev/pymodbus #97: No communication to Emerson Commander SE drive (RTU/Serial)

Sprinterfreak, 2015-08-13:
Dear,
im Trying to talk to an Emerson Commander SE drive using modbus RTU.
Tryed different implementations sync and async. Both didn't work.
But minimalmodbus can talk to the drive, so I assume the hardware setup ist correct.

I digged around a bit and found out that the answers from the drive didn't get fully received.
The device manual can be found here:
 http://www.transbearco.com/Tech_Support_Files/Emerson_Manuals/Commander_SE_Advanced_Users_Guidepdf.pdf

If I'm trying to route to the Serial forwarder Example it showes this
---
DEBUG:pymodbus.server.sync:Started thread to serve client at ('127.0.0.1', 50042)
DEBUG:pymodbus.server.sync:Client Connected [127.0.0.1:50042]
DEBUG:pymodbus.server.sync:0x0 0x1 0x0 0x0 0x0 0x9 0x1 0x10 0x0 0x71 0x0 0x1 0x2 0x0 0x5 0x0 0x2 0x0 0x0 0x0 0x9 0x1 0x10 0x0 0x75 0x0 0x1 0x2 0x0 0x0 0x0 0x3 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x80 0x0 0x1 0x2 0x0 0x7f 0x0 0x4 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x9 0x1 0x10 0x0 0x68 0x0 0x1 0x2 0x0 0x0
DEBUG:pymodbus.factory:Factory Request[16]
DEBUG:pymodbus.datastore.remote:validate[16] 113:1
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 1
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'NoneType' object has no attribute 'function_code'; Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/server/sync.py", line 59, in execute
    response = request.execute(context)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/register_write_message.py", line 176, in execute
    if not context.validate(self.function_code, self.address, self.count):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/datastore/remote.py", line 42, in validate
    return result.function_code < 0x80
AttributeError: 'NoneType' object has no attribute 'function_code'

ERROR:pymodbus.pdu:Exception Response F(16) E(4)
DEBUG:pymodbus.server.sync:send: 000100000003019004
DEBUG:pymodbus.factory:Factory Request[16]
DEBUG:pymodbus.datastore.remote:validate[16] 117:1
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 2
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'NoneType' object has no attribute 'function_code'; Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/server/sync.py", line 59, in execute
    response = request.execute(context)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/register_write_message.py", line 176, in execute
    if not context.validate(self.function_code, self.address, self.count):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/datastore/remote.py", line 42, in validate
    return result.function_code < 0x80
AttributeError: 'NoneType' object has no attribute 'function_code'
---

The drive definitly answers wich I have seen on the scope but pymodbus didn't receive it. With same setup but using minimalmodbus it works perfectly.

My hello world application looks like this
---
#!/usr/bin/env python
'''
Pymodbus Asynchronous Processor Example
--------------------------------------------------------------------------

The following is a full example of a continuous client processor. Feel
free to use it as a skeleton guide in implementing your own.
'''
#---------------------------------------------------------------------------#
# import the neccessary modules
#---------------------------------------------------------------------------#
from twisted.internet import serialport, reactor, defer
from twisted.internet.protocol import ClientFactory
from pymodbus.factory import ClientDecoder
from pymodbus.client.async import ModbusClientProtocol

#---------------------------------------------------------------------------#
# Choose the framer you want to use
#---------------------------------------------------------------------------#
#from pymodbus.transaction import ModbusBinaryFramer as ModbusFramer
#from pymodbus.transaction import ModbusAsciiFramer as ModbusFramer
[... 226 lines cut ...]
Unhandled error in Deferred:

Traceback (most recent call last):
Failure: pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
Unhandled error in Deferred:

Traceback (most recent call last):
Failure: pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
Unhandled error in Deferred:

Traceback (most recent call last):
Failure: pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
Unhandled error in Deferred:

Traceback (most recent call last):
Failure: pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
Unhandled error in Deferred:

Traceback (most recent call last):
Failure: pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Connection lost during request
---

--
Best regards,
Jan

karlp, 2015-08-13:
I haven't looked at this closely, but are you really sure you're doing the right thing?  You seem to have a mix of socket (TCP) _and_ serial (RTU) configuration in the same code.  Your description talks of trying to do RTU.

I've never tried any of the twisted stuff either personally.  Do you have the minimal modbus code that does work?

Sprinterfreak, 2015-08-13:
### The minimalmodbus code wich works
```python
#!/usr/bin/env python
import minimalmodbus

class Commander(object):
    def __init__(self,tty,unit):
        self.drive=minimalmodbus.Instrument(tty,unit)
        self.drive.CLOSE_PORT_AFTER_EACH_CALL = True
        self.drive.mode = 'rtu'
        self.drive.serial.baudrate = 19200   # Baud
        self.drive.serial.parity   = minimalmodbus.serial.PARITY_NONE
        self.drive.serial.bytesize = 8
        self.drive.serial.stopbits = 2
        self.drive.serial.timeout = 40

    def read_registers(self,base,len=1):
        ret=self.drive.read_registers(base-1,len)
        return ret

    def write_registers(self,base,value):
        self.drive.write_registers(int(base)-1,value)

drive = Commander('/dev/ttyUSB0', 0x01)

drive.write_registers(106,[1000])
drive.write_registers(114,[5])
drive.write_registers(641,[127,0,1])

while True:
    temp = drive.read_registers(704,1)[0]
    speed = temp * 5
    drive.write_registers(105,[speed])
    drive.write_registers(642,[7])
```

Yes I have tried to go different ways talking to my drive.
First I have used the code above with ModbusRtuFramer and SerialModbusClient commented in and SocketFramer vs connectTCP commented out. Same result. Nothing...
At least I'am trying to pass the traffic to
### Pymodbus Synchronous Serial Forwarder
*similar to the Example http://pymodbus.readthedocs.org/en/latest/examples/serial-forwarder.html*
except the line:
```
- client = ModbusClient(method='ascii', port='/dev/pts/14')
+ client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=19200, stopbits=2, bytesize = 8, parity = 'N')
```
The serial forwarder is running in a seperate process.
Even if I send commands using QModMaster via Network it gives me the same errors as if I execute the client.async code above to feed the forwarder.
#### Serial Forwarder Output
```
DEBUG:pymodbus.server.sync:Started thread to serve client at ('***', 1246)
DEBUG:pymodbus.server.sync:Client Connected [***:1246]
DEBUG:pymodbus.server.sync:0x0 0x1 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7
DEBUG:pymodbus.factory:Factory Request[16]
DEBUG:pymodbus.datastore.remote:validate[16] 641:1
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 1
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'NoneType' object has no attribute 'function_code'; Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/server/sync.py", line 59, in execute
    response = request.execute(context)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/register_write_message.py", line 176, in execute
    if not context.validate(self.function_code, self.address, self.count):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/datastore/remote.py", line 42, in validate
    return result.function_code < 0x80
AttributeError: 'NoneType' object has no attribute 'function_code'

ERROR:pymodbus.pdu:Exception Response F(16) E(4)
DEBUG:pymodbus.server.sync:send: 000100000003019004
DEBUG:pymodbus.server.sync:0x0 0x2 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x3 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x4 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7
[... 44 lines cut ...]
AttributeError: 'NoneType' object has no attribute 'function_code'

ERROR:pymodbus.pdu:Exception Response F(16) E(4)
DEBUG:pymodbus.server.sync:send: 000400000003019004
DEBUG:pymodbus.server.sync:0x0 0x5 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x6 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x7 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x8 0x0 0x0 0x0 0x9 0x1 0x10 0x2 0x81 0x0 0x1 0x2 0x0 0x7 0x0 0x9 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x81 0x0 0x2 0x0 0xa 0x0 0x0 0x0 0x6 0x1 0x3 0x2 0x81 0x0 0x2
DEBUG:pymodbus.factory:Factory Request[16]
DEBUG:pymodbus.datastore.remote:validate[16] 641:1
DEBUG:pymodbus.transaction:Running transaction 5
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 5
DEBUG:pymodbus.server.sync:Datastore unable to fulfill request: 'NoneType' object has no attribute 'function_code'; Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/server/sync.py", line 59, in execute
    response = request.execute(context)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/register_write_message.py", line 176, in execute
    if not context.validate(self.function_code, self.address, self.count):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus-1.3.0-py2.7.egg/pymodbus/datastore/remote.py", line 42, in validate
    return result.function_code < 0x80
AttributeError: 'NoneType' object has no attribute 'function_code'
```
The interesting part is, the Pymodbus library receives incomplete garbage wich is different everytime to the same request. Perhaps a timing issue?
But minimalmodbus runs perfectly stable over time.
Therfore I didn't think about an hardware issue.

Thanks,
Jan

karlp, 2015-08-13:
If you're trying to work out why pymodbus isn't working and minimal modbus is, can you at least start by trying some direct comparisons?  You have minimal modbus working on serial RTU.  Great.  Now do the same with pymodbus.  That _should_ be easy.  If that doesn't work, you can make a nice clear bug report about what doesn't work, and you can leave out all of this socket relay noise.

If that _does_ work, then you should be in a position to provide much clearer, simpler explanations of what is and is not working.  It's possible that the serial forwarding example code just doesn't work, or at least, not the way you expect, but that is then completely separate from RTU with an emerson drive controller.

Sprinterfreak, 2015-08-13:
Thanks Karl.
As I mentioned before the rtu standalone-version of my programm without "socket things" differs in two lines of code:
assigning framer module
```python
from pymodbus.transaction import ModbusRtuFramer as ModbusFramer
#from pymodbus.transaction import ModbusSocketFramer as ModbusFramer
```

and in _main()_ chosing the serial client as factory
```python
    SerialModbusClient(factory, SERIAL_PORT, reactor, baudrate=19200, stopbits=2, bytesize=8, parity=serialport.PARITY_NONE)
    #factory = reactor.connectTCP("127.0.0.1", 502, factory)
```
Now no socket things are involved anymore. The result is already posted above.
This runs until ` DEBUG:pymodbus.transaction:adding transaction 5 ` and hangs forever.
Only interuptable with Ctrl+c. As I said before things get send to the drive and the drive answers back. I did not have a real scope or logic analyzer but a fluke 124 scopemeter wich showes me traffic on the line. But the rtu framer skips mostly all the returned packages.
In transaction.py: ModbusRtuFramer.processIncomingPacket() data from the drive getting in and dropped because checkFrame() often returns False and the packet gets dropped quietly.
This is not the issue itself I think. I don*t know why but the received packet length in the buffer differs at any repeated request and most requests (not all, ~20% varying success rate) hang in the
`while self.isFrameReady()` loop until application is killed.

My oppinion is that we have an issue while filling the buffer 
Possibly wait timers that packets didn't get captured completely.
Im pretty new to python and modbus anyway so I'll try my best to help.

vkooistra, 2016-03-30:
Did you ever get it to work properly? I'm running in the exact same issue.

Sprinterfreak, 2016-03-30:
Yes. But only synchronous.
The best experiences I've made using pySimpleModbus.
This was pretty reliable and fast as i can remember.

vkooistra, 2016-03-31:
Got it to work with async. The ModbusRtuFramer in transaction.py is bugged.

In function ProcessIncomingPacket a call is made to self.checkFrame(). This function creates a crc buffer based on frame_size (frame_size - 2: frame_size) and does a check on it using the first 2 positions of the buffer. However in almost all situations this will result in a buffer with size 0, simply because not enough data has been received yet.

For now I just added an if/else to checkFrame to verify there is enough data available to work with. This seems to fix it

hpatel826, 2016-10-12:
@vkooistra Would you mind sharing a diff of your changes? I might be having related issues and would like to observe you fix

Sprinterfreak, 2016-10-12:
I switched to modbus simple then.
Didn't get the async version to run.
With the sync version I got pretty slow communication too.

My setup doesn't exists anymore so in my case I can't tell if the problem
still persists at the current master.

vkooistra, 2016-10-12:
[transaction.py.txt](https://github.com/bashwork/pymodbus/files/525469/transaction.py.txt)

I can't remember it exactly, I have to look it up. Anyway, the attached file is what I altered to get it to work. It has been stripped down to RTU only, which was easier to figure things out and all.

The main items changed are at line 160. Where the framereset has been removed (well more moved elsewhere). The other part is in the lines "checkFrame" function (from line 101 upwards). You probably want to edit this is the library, in which the lines will differ. Or just use this file as external instead, which should also work.

Note: I don't have the checkCRC function on this computer, I recovered the files from my e-mail so I can't see what it does exactly. But this should fix the dropping of a partly received message. A partly message will fail CRC check and will be disposed, while it should first receive the entire message and then validate the CRC.

janiversen, 2022-04-01:
Is this issue still relevant, if not it will be closed in a couple of days.

dhoomakethu, 2022-04-01:
@janiversen  we can remove the lable `invetigating` so that the bot will auto-close if no issue activity. This can be closed as there is no activity.