pymodbus-dev/pymodbus #98: Bad packet creation

kseagrav, 2015-08-21:
I am trying to use a serial server to acquire packets from a transmitter over RS485 (using ModbusRtuFramer). Whether I use a synchronous server or an asynchronous server, and no matter how I manipulate the timeout, the creation of the packets seems to be inconsistent. I really need to use async, because I need an updating function to update the registers in the context(s), but in an attempt to get pymodbus to work with what I have I tried both:
For async, a few bytes are read in at a time, and a majority of the time they are thrown away before a good packet can be built. Eventually it will build a good packet and responds correctly, but it takes much too long to do this and most packets are lost. 
Using sync, way too much data is read in at a time, and then it grabs one good packet out of this and throws everything else away. As a result, necessary packets are lost in the process and only a few are received and responded to. 
I managed to write my own code to receive and respond to packets, just using your CRC checking code to validate them, and got it to work perfectly fine so I know it is possible with my transmitter (which may have an odd timing issue). 

Here is the bulk of my code, if using async (mostly same, but without LoopingCall, for sync):
```python
if __name__ == "__main__":
    # Create slaves, server context, and identity
    slaves = {}
    for boards in config.boards:
        slaves[string_to_hex(boards)] = ModbusSlaveContext(hr=ModbusSequentialDataBlock.create())

    server_context = ModbusServerContext(slaves=slaves, single=False)
    identity = ModbusDeviceIdentification()

    # Start up looping function to update the context
    time = config.polling_interval
    loop = LoopingCall(f=updating_writer,
                       a=(server_context,),
                       address=config.client_address,
                       port=config.client_port)
    loop.start(time, now=False)

    StartSerialServer(server_context,
                      identity=identity,
                      framer=ModbusRtuFramer,
                      port=config.port,
                      timeout=0,
                      baudrate=9600)
```

dmosberger, 2017-10-30:
Hmmh, I see this issue is from 2015 and yet I appear to run into the same problem.  I'm using the modbus server example from the documentation, except modified to enable RTU via serial like this:

```
serial_port = '/dev/ttyUSB0'
StartSerialServer(context, framer=ModbusRtuFramer, identity=identity,
                  port=serial_port, stopbits=1, bytesize=8, parity='N',
                  baudrate=9600)
```

The server very quickly would die with an uncaught exception trying to decode a packet (struct.unpack threw the exception because the packet didn't have the expected length.

I think the problem is in ModbusRtuFramer.processIncomingPacket().  Whenever it receive a partial packet (which can happen a lot over serial), self.checkFrame() would return False and then the packet would be treated as an erroneous packet.  The problem then gets made worse by ModbusRtuFramer._process() because it attempts to decode the partial packet (and then that triggers aforementioned exception).

I changed processIncomingPacket() to this:
```
def processIncomingPacket(self, data, callback):
        self.addToFrame(data)
        while self.checkFrame():
            self._process(callback)
```

and now it has been working just fine for hours.  This code will get stuck on a bogus Modbus packet, however (since checkFrame() never returns True in that case).  Perhaps a better fix would be to check if len(self.__buffer)>=self.__header['len'] and if checkFrame() returns False, handle it as an error.

dmosberger, 2017-10-31:
Here is a better version:
```
    def processIncomingPacket(self, data, callback):
        self.addToFrame(data)
        while self.isFrameReady():
            if self.checkFrame():
                self._process(callback)
            else:
                buffer_len = len(self.__buffer)
                expected_len = self.__header.get('len', buffer_len + 1)
                if buffer_len > expected_len:
                    self.resetFrame()
                break
```

dhoomakethu, 2017-12-22:
@dmosberger Could you please verify with the latest changes on `dev` branch or you can also try the pre-release 1.4.0rc2 either from github or from pypi `pip install pymodbus==1.4.0rc2` or `pip install https://github.com/riptideio/pymodbus/archive/v1.4.0rc2.tar.gz`