pymodbus-dev/pymodbus #103: What's the best way of writing the float value to ModbusServerContext ?

mindmixer, 2015-10-26:
I'm using the pymodbus library for creating the Modbus server. I've discovered all examples and faced with problem related to writing float value on the server (slave) side. There is an updating_writer function in which I want to write the float value = 22.34 to 2 holding resitors:

    def updating_writer(a):
        context  = a[0]
        register = 3
        slave_id = 0x00
        address  = 0x0
        builder = BinaryPayloadBuilder(endian=Endian.Little)
        builder.add_32bit_float(22.34)
        res = builder.build()
        context[slave_id].setValues(register, address, res)

    store = ModbusSlaveContext(
        di = ModbusSequentialDataBlock(0, [17]*100),
        co = ModbusSequentialDataBlock(0, [17]*100),
        hr = ModbusSequentialDataBlock(0, [17]*100),
        ir = ModbusSequentialDataBlock(0, [17]*100))
    context = ModbusServerContext(slaves=store, single=True)

But when I have a connection from Modbus master (client) and try to read these 2 registor I get an error:

        data = message.encode()
      File "/home/workstation/devel/modbus_example/env_modbus/local/lib/python2.7/site-packages/pymodbus/register_read_message.py", line 71, in encode
        result += struct.pack('>H', register)
    struct.error: cannot convert argument to integer

How I should write the float value to some context? I cannot find any examples.

closed 2015-10-28 by mindmixer

byhisdeeds, 2016-04-28:
I have the same problem. What was your solution why you closed it.

FlaminMad, 2016-06-13:
Quite a way down the line now but I also am experiencing this problem. Any ideas on the solution?

shedfly, 2018-09-15:
For the next person looking, its discussed here:
https://groups.google.com/forum/#!topic/pymodbus/X4vtEg9Hq10

in summary, use builder.to_registers()

Example here:
https://github.com/riptideio/pymodbus/blob/master/examples/common/modbus_payload_server.py