pymodbus-dev/pymodbus #105: Install fails on Windows Python 3.3 - pyserial

LorenAmelang, 2015-10-28:
I thought it might be faster to run my test script on my Windows machine with local Wireshark than to set up a promiscuous capture from Wi-Fi via Linux Wireshark. Guess not. My Win7 only has Python 3.3...  

Select Branch = Python3
<https://github.com/bashwork/pymodbus/tree/python3>
Download Zip
Move contained folder to C:\Apps\Python33\Lib\site-packages
C:\Apps\Python33\Lib\site-packages\pymodbus-python3>python setup.py install

Seemed to work except:
---
Installed c:\apps\python33\lib\site-packages\pymodbus-1.2.0-py3.3.egg
Processing dependencies for pymodbus==1.2.0
Searching for pyserial>=2.6
Reading https://pypi.python.org/simple/pyserial/
Best match: pyserial 2.7
Downloading https://pypi.python.org/packages/any/p/pyserial/pyserial-2.7.win32.exe#md5=21555387937eeb79126cde25abee4b35
Processing pyserial-2.7.win32.exe
File "c:\users\loren\appdata\local\temp\easy_install-kgavir\pyserial-2.7-py3.3-win32.egg.tmp\serial\rfc2217.py", line 385
    except Exception, msg:
                    ^
SyntaxError: invalid syntax
...
[lots more version errors]
...
creating 'c:\users\loren\appdata\local\temp\easy_install-kgavir\pyserial-2.7-py3.3-win32.egg' and adding 'c:\users\loren\appdata\local\temp\easy_install-kgavir\pyserial-2.7-py3.3-win32.egg.tmp' to it
Moving pyserial-2.7-py3.3-win32.egg to c:\apps\python33\lib\site-packages
Adding pyserial 2.7 to easy-install.pth file
Installing miniterm.py script to C:\Apps\Python33\Scripts

Installed c:\apps\python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg
Searching for twisted>=13.1.0
Reading https://pypi.python.org/simple/twisted/
Best match: Twisted 15.4.0
Downloading https://pypi.python.org/packages/source/T/Twisted/Twisted-15.4.0.tar.bz2#md5=5337ffb6aeeff3790981a2cd56db9655
Processing Twisted-15.4.0.tar.bz2
Writing c:\users\loren\appdata\local\temp\easy_install-hi3kp2\Twisted-15.4.0\setup.cfg
Running Twisted-15.4.0\setup.py -q bdist_egg --dist-dir c:\users\loren\appdata\local\temp\easy_install-hi3kp2\Twisted-15.4.0\egg-dist-tmp-wur7si
file twisted.py (for module twisted) not found
file twisted\application.py (for module twisted.application) not found
file twisted\application\test.py (for module twisted.application.test) not found
file twisted\cred.py (for module twisted.cred) not found
...
[lots more not found]
...
file twisted\web\_auth.py (for module twisted.web._auth) not found
file twisted\web\test.py (for module twisted.web.test) not found
creating c:\apps\python33\lib\site-packages\twisted-15.4.0-py3.3.egg
Extracting twisted-15.4.0-py3.3.egg to c:\apps\python33\lib\site-packages
Adding twisted 15.4.0 to easy-install.pth file
Installing trial script to C:\Apps\Python33\Scripts

Installed c:\apps\python33\lib\site-packages\twisted-15.4.0-py3.3.egg
---

No luck:
---
C:\Users\Loren\Documents\Projects\Computing\BeagleBone Black\BBB Projects\SunSpec Modbus\Pymodbus for SunSpec>python suns311debug.py3
Traceback (most recent call last):
  File "suns311debug.py3", line 6, in <module>
    from pymodbus.client.sync import ModbusTcpClient
  File "<frozen importlib._bootstrap>", line 1558, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1525, in _find_and_load_unlocked
  File "C:\Apps\Python33\lib\site-packages\pymodbus-1.2.0-py3.3.egg\pymodbus\client\sync.py", line 2, in <module>
  File "<frozen importlib._bootstrap>", line 1558, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1525, in _find_and_load_unlocked
  File "C:\Apps\Python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg\serial\__init__.py", line 19, in <module>
  File "<frozen importlib._bootstrap>", line 1558, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1525, in _find_and_load_unlocked
  File "C:\Apps\Python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg\serial\serialwin32.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 1558, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1525, in _find_and_load_unlocked
  File "C:\Apps\Python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg\serial\win32.py", line 196
    MAXDWORD = 4294967295L # Variable c_uint
                         ^
SyntaxError: invalid syntax
---
--> That '^' actually points to the 'L' in "4294967295L"

I barely know what I'm doing in Python 2.7, and haven't had much luck with Python3, especially on Windows. Maybe that is a problem in pyserial, and not your responsibility? Looks like a version issue to me...  But I thought I'd report it, in case you can get it fixed. Or tell me what I've done wrong...  

Off to do this on Linux...  Where it just works...  Thanks!

LorenAmelang, 2015-11-20:
Just saw this, with almost the exact error message I get:
http://stackoverflow.com/questions/25082577/netbeans-python-pyserial-import-not-found

Their solution didn't help:
>pip3 install pyserial
Requirement already satisfied (use --upgrade to upgrade): pyserial in c:\apps\python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg
>pip3 install --upgrade  pyserial
Requirement already up-to-date: pyserial in c:\apps\python33\lib\site-packages\pyserial-2.7-py3.3-win32.egg

One contributor said there is no Python 3 version of pyserial, another said there is. The filenames seem to suggest there is, but the error shows it is not operational here...

dhoomakethu, 2017-05-27:
cleaning up old issues, there is a universal pip installable pymodbus available . I have no windows machines to test , please use it and reopen the issue if you are still having trouble installing it on windows. Any support to make pymodbus compatible with windows is highly appreciated.

closed 2017-05-27 by dhoomakethu

LorenAmelang, 2018-04-08:
@dhoomakethu
I ended up here again, love it when Google finds me my own questions...  Your pip suggestion worked perfectly in Win 10 Python 3.5.0 :: Anaconda 2.4.0 (64-bit)! 

`pip install pymodbus
Collecting pymodbus
  Downloading pymodbus-1.4.0-py2.py3-none-any.whl (82kB)
    100% |################################| 86kB 483kB/s
Collecting pyserial>=2.6 (from pymodbus)
  Downloading pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |################################| 196kB 64kB/s
Installing collected packages: pyserial, pymodbus
Successfully installed pymodbus-1.4.0 pyserial-3.4`