pymodbus-dev/pymodbus #110: State per client-IP

cudeso, 2015-12-11:
Hi,

This isn't an "issue" but rather a question. Is there a way to have the state of the coils and HR independent per client IP? 
F.e.: a server starts with a predefined set of values. If client A changes the value of a HR then this is only visible to client A. Client B still sees the original value set during server startup.

Thanks.
koen

karlp, 2015-12-11:
Bear in mind that there's no way you could do this on modbus-rtu, it's only something you think you're used to having from IP applications.  You _might_ be able to hack something together, but you'd be confusing anyone else using your software who expects "modbus"

cudeso, 2015-12-11:
I understand. 
The idea is to setup something ('classroom-purpose') where one can interact with the slave/master without getting interfered with actions done by other participants. 
Figured out I should get "somewhere" (store.py I assume) access to the client_address set in ModbusBaseRequestHandler and then add in the file something for storing the data in a sqlitedb or similar.

karlp, 2015-12-12:
Why not just have separate virtual devices per participant instead? less hassle, more standard, less programming.

closed 2018-05-03 by dhoomakethu