pymodbus-dev/pymodbus #114: Wrong data by Reassembled TCP Segments

HFlor, 2016-02-22:
I have this script:

`#!/usr/bin/env python

from pymodbus.client.sync import ModbusTcpClient

client = ModbusTcpClient('192.168.75.27')
client.connect
result = client.read_input_registers(1000, 62)
print result
client.close()
`

The response from the client comes in 2 packages. The recording is attached. As error message I get the following:

`Traceback (most recent call last):
  File "./read2.py", line 7, in <module>
    result = client.read_input_registers(1000, 62)
  File "/usr/lib/python2.7/dist-packages/pymodbus/client/common.py", line 122, in read_input_registers
    return self.execute(request)
  File "/usr/lib/python2.7/dist-packages/pymodbus/client/sync.py", line 79, in execute
    return self.transaction.execute(request)
  File "/usr/lib/python2.7/dist-packages/pymodbus/transaction.py", line 71, in execute
    self.client.framer.processIncomingPacket(result, _set_result)
  File "/usr/lib/python2.7/dist-packages/pymodbus/transaction.py", line 247, in processIncomingPacket
    result = self.decoder.decode(self.getFrame())
  File "/usr/lib/python2.7/dist-packages/pymodbus/factory.py", line 190, in decode
    return self._helper(message)
  File "/usr/lib/python2.7/dist-packages/pymodbus/factory.py", line 212, in _helper
    response.decode(data[1:])
  File "/usr/lib/python2.7/dist-packages/pymodbus/register_read_message.py", line 82, in decode
    self.registers.append(struct.unpack('>H', data[i:i + 2])[0])
struct.error: unpack requires a string argument of length 2
`

The "decode" in the module "register_read_message.py" in line 79 gets only 120 bytes of data, and not 125 as it should be. The data of the second package is missing.

[wireshark.txt](https://github.com/bashwork/pymodbus/files/140626/wireshark.txt)

MayukhSobo, 2017-03-07:
Any solution to this found ?? Please let me know

HFlor, 2017-03-07:
There is no solution. Before I could investigate the problem, the manufacturer of the heat pump has installed an update. After that, the packages were bigger and more complete.

dhoomakethu, 2017-12-19:
Will be fixed as part of 1.4.0 Refer #251

closed 2017-12-19 by dhoomakethu