pymodbus-dev/pymodbus #116: ModbusTcpClient.read_holding_register return None

ccatterina, 2016-03-10:
Hi, 
I'm trying to read a device with modbus rtu over tcp, but read_holding_registers function return always None. If i edit transaction.py adding a time.sleep(0.5) between send and recv like suggested in an other similar issue here, read_holding_registers return me the right values. Is there an other cleaner way to make it works?

`self.client = ModbusTcpClient(host=ip, port=port, framer=ModbusRtuFramer)`
   ```
 def read (self, attribute):
        attr = self.data[attribute]

        if self.client.connect ():
            value = self.client.read_holding_registers (attr['reg'], attr['reg_num'], unit = self.unit)
            self.client.close ()
            regs=list(reversed(value.registers))
            packed = pack ('>' + ('H' * attr['reg_num']), *regs)
            return unpack (attr['type'], packed)[0]

        else:
            return None

```

CabritoIncognito, 2016-03-23:
@ccatterina I'm getting a very similar response. Have you had any luck figuring this issue out and (@bashwork ) does anyone see anything blatantly wrong with what I'm attempting? I've been able to pull back input register 32605 from the device I'm connected to with both C# and java but would rather use Python for the project.

from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.transaction import ModbusRtuFramer

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient('10.100.xxx.xxx', port=xxxx, framer=ModbusRtuFramer) 

client.connect() 

rr = client.read_input_registers(2605, 1)
print rr

client.close()

**Output**
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:Transaction failed. (timed out) 
DEBUG:pymodbus.transaction:Transaction failed. (timed out) 
DEBUG:pymodbus.transaction:Transaction failed. ([Errno 10053] An established connection was aborted by the software in your host machine)

ccatterina, 2016-03-23:
hi @CabritoIncognito i figured this issue out using this version of pymodbus https://github.com/mesias/pymodbus edited by @mesias . By the way i didn't receive the same output as yours, did you try to specify the unitid in client.read_input_register? i think it's necessary when you use modbus rtu over tcp.

PS: ty @mesias your version solved me this problem after many days searching for a solution!

CabritoIncognito, 2016-03-24:
Thanks Claudio! The Mesias version worked well for me and I was able to get the results I was looking for with the addition of the unitid.

I appreciate it!

Justin

vincent1190, 2017-02-22:
This is a simple solution. It returns none because of the delay in the PLC(atleast that was the case for us) . It takes a second for the PLC to actually respond so you just have to setup while loop along with  time.sleep() to allow some time for the response to come in

ccatterina, 2017-06-22:
Should be fixed by PR #180

closed 2017-12-23 by dhoomakethu