pymodbus-dev/pymodbus #117: Install breakage with latest setuptools?

meeas, 2016-03-10:
I think there may have been something in the latest setuptools that is choking on something in pymodbus/__init__.py.  It doesn't appear to install through pip nor by download with manual install.  Here is the error:

==========
Traceback (most recent call last):
  File "setup.py", line 26, in <module>
    from pymodbus import __version__, __author__
  File "/private/var/root/pymodbus/pymodbus/__init__.py", line 37
    True, False = (1 == 1), (0 == 1)
       ^
SyntaxError: can't assign to keyword
==========

Commenting out the try/except section which makes up the last four lines of that file fixes the issue allowing pymodbus to install cleanly.

barik, 2016-03-15:
Depending on the version of Python you are using, you may wish to try `pip install pymodbus3`.

geoffe, 2016-03-15:
Ah, thanks.  Although I found a more active project minimalmodbus which may suit just as well.

Tutunkommon, 2016-08-25:
Install is still broken.  Looked at minimalmodbus but it doesn't support TCP.  Hoping pymodbus3 covers everything!

dhoomakethu, 2017-06-28:
fixed with pymodbus >= 1.3.0

closed 2017-06-28 by dhoomakethu