pymodbus-dev/pymodbus #119: No Check for ascii data

DJJo14, 2016-03-30:
While doing:
`result = client.read_input_registers(0x00,0x13, unit=adrr)`

I get:

`  File "C:\Python27\lib\site-packages\pymodbus\client\common.py", line 120, in read_input_registers
    return self.execute(request)
  File "C:\Python27\lib\site-packages\pymodbus\client\sync.py", line 83, in execute
    return self.transaction.execute(request)
  File "C:\Python27\lib\site-packages\pymodbus\transaction.py", line 65, in execute
    self.client.framer.processIncomingPacket(result, self.addTransaction)
  File "C:\Python27\lib\site-packages\pymodbus\transaction.py", line 671, in processIncomingPacket
    if self.checkFrame():
  File "C:\Python27\lib\site-packages\pymodbus\transaction.py", line 597, in checkFrame
    self.__header['uid'] = int(self.__buffer[1:3], 16)
ValueError: invalid literal for int() with base 16: '\x00\x00'`

In `transaction.py` form line 604 until 606 there is no check for ascii data.

dhoomakethu, 2017-08-04:
Not able to reproduce, please provide the logs if you have any.

dhoomakethu, 2017-08-08:
Please try 1.3.2.rc1 for a potential solution

dhoomakethu, 2017-08-23:
pymodbus 1.3.2 is out with the fix for this issue. Please re-open if still facing the issue.

closed 2017-08-23 by dhoomakethu