pymodbus-dev/pymodbus #122: How can I read RTU value ? (as a Master or using pymodbus.client.sync

thunderbirdtr, 2016-04-26:
Hi ; 
I was trying to use pymodbus for TCP and RTU connections but I have small problem for reading value. When I use TCP I  can use "getRegisters","registers" this parameters but If I use ModbusSerialClient I can't call my that parameters and If I print the register value It's give me just "None" as a result and can't ready anything from it. Here is the code I use for simple RTU connection to slave. 

`import time
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

port = '/dev/ttyAMA0'
print port
client = ModbusClient(method='rtu', port=port, retries=1000, timeout=1, rtscts=True, parity='E', baudrate=9600, unit=1)

client.connect()

read_register = 1004
qty = 1
rr = client.read_holding_registers(read_register, qty)
print rr
time.sleep(1 / 1000000.0)
client.close()
`
And DEBUG result ; 

DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:getting transaction 1
None

Thank you

thompcd, 2016-07-19:
Sorry for kicking up an old thread - but did you ever get any results on this? I've been trying to get RTU to work for a week now to no avail and can't find any info on it.

semyont, 2016-07-27:
Hi,
Did you try to use the latest 1.3.0 build? or are you using the pip 1.2.0 version?

thompcd, 2016-07-28:
Hello,
Thanks for the response!
I was using the pip 1.2.0 version. I never made any progress on this, as I swapped over to minimalmodbus and it has all the features I needed. I've finished up on this project, so I don't believe I'll be able to revisit what my issue was.

hno, 2016-11-11:
After getting the serial layers worked out the version included in Debian works for me (1.2.0-2).

For it to work the serial port must alreadby be configured as an RS485 port with the right settings, and the serial driver used need to properly support RS485.  On BeagleBone black the TI serial driver (/dev/ttyO4) works. The 8250 serial driver (/dev/ttyS4) do not work properly (part of the request is echoed back even with rs485 loopback disabled). A workaround for the 8250 driver is possible by enabling RS485 loopback and modifying pymodbus to read and discard the echoed request before reading the response.

Only issue is that the lack of proper frame collection when reading the response frame makes communication quite slow, always waiting for the maximum timeout before trying to decode response.

I have not yet tried newer version of pymodbus as the 1.2.0 version shipped by Debian does what I need right now.

hno, 2016-11-11:
These are my client connection settings that works:

note: pyserial is from Debian jessie-backports as the older version in jessie lacks RS485 mode setting support. If the port is already configured in rs485 mode then rs485_mode setting is not needed I think.

```
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
import serial.rs485

client = ModbusClient(method='rtu', port='/dev/ttyO4', timeout=0.5, baudrate=9600)
client.connect()
rs485_mode = serial.rs485.RS485Settings(delay_before_tx = 0, delay_before_rx = 0, rts_level_for_tx=True, rts_level_for_rx=False, loopback=False)
client.socket.rs485_mode = rs485_mode
```

tusharvb19, 2017-05-19:
****My Setup********
Arduino--->RS485 slave example---->RS485 TTL module---->pair of wire---->RS485 to USB module---->Nano Pi NEO with dietpi(jessie) working
(I have tested the above setup via USB on PC using modbus master simulator and it is working!

Here is my code I want to read holding registers for NanoPi NEO:

import pymodbus
import serial
from pymodbus.pdu import ModbusRequest
from pymodbus.client.sync import ModbusSerialClient as ModbusClient #initialize a serial RTU client instance
from pymodbus.transaction import ModbusRtuFramer

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#count= the number of registers to read
#unit= the slave unit this request is targeting
#address= the starting address to read from

client= ModbusClient(method = "rtu", port="/dev/ttyUSB0", stopbits = 1, bytesize = 8, parity = 'N', baudrate= 19200)

#Connect to the serial modbus server
connection = client.connect()
print connection

#Starting add, num of reg to read, slave unit.
#result= client.read_holding_registers(0x00,2,unit= 0x1)
result= client.read_input_registers(40001, 16, unit=0x01)

print(result.registers) 

#temps  = client.read_holding_registers(0x00, 4, unit=0x1) # address, count, slave address
#print temps.registers
#Closes the underlying socket connection
client.close()

The response I am getting:
True
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x84 0x83 0xf8 0xfe
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
Traceback (most recent call last):
  File "test8.py", line 26, in <module>
    print result.registers 
AttributeError: 'ReadCoilsResponse' object has no attribute 'registers'

Can anyone help me to resolve this issue?

tusharvb19, 2017-05-19:
@hno 

After adding your code got this issue:

 client.socket.rs485_mode = rs485_mode
  File "/usr/local/lib/python2.7/dist-packages/serial/serialutil.py", line 496, in rs485_mode
    self._reconfigure_port()
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 445, in _reconfigure_port
    self._set_rs485_mode(self._rs485_mode)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 158, in _set_rs485_mode
    raise ValueError('Failed to set RS485 mode: {}'.format(e))
ValueError: Failed to set RS485 mode: [Errno 22] Invalid argument

gordio, 2018-01-22:
@tusharvb19 `ReadCoilsResponse` not contain`registers`, check `coils` property.

JinTran89, 2018-08-20:
Hi,
I am trying to connect Modbus RTU from RasPi to PowerMeter and read holding register. But I'm having problems printing register: "AttributeError: 'ModbusIOException' object has no attribute 'registers'". I hope someone can help me.
Thanks!
Here is my code :
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.register_read_message import ReadInputRegistersResponse

client= ModbusClient(method = "rtu", port="/dev/ttyUSB0", stopbits = 1, bytesize = 8, parity = 'N', baudrate= 19200)
connection = client.connect()
print connection

result= client.read_holding_registers(30001, 2, unit= 1)
print(result.registers)

client.close()

The response I am getting:
True
Traceback (most recent call last):
  File "Modbus-RS485-8.py", line 9, in <module>
    print(result.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

dhoomakethu, 2018-08-20:
@JinTran89 Please check if raspberry pi you are using support RS-485 , if your communication settings are good and the device supports RS-485 communication you should see the response. You can enable debug logs for more insights on what's happening at the bus level.

JinTran89, 2018-08-20:
Thanks for the quick answer
I'm using a USB-RS485 converter: https: //www.ebay.co.uk/itm/111944569765? _Trksid = p2057872.m2749.l2649 & ssPageName = STRK% 3AMEBIDX% 3AIT

And after adding debug logs I get the result:
True
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x1 0x3 0x75 0x31 0x0 0x1 0xcf 0xc9
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x1 0x80 0x2 0xc0 0x1
DEBUG:pymodbus.framer.rtu_framer:Getting Frame - 0x80 0x2
DEBUG:pymodbus.factory:Factory Response[128]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 128
ERROR:pymodbus.transaction:Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/transaction.py", line 164, in execute
    request.unit_id)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/framer/rtu_framer.py", line 219, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/framer/rtu_framer.py", line 298, in _process
    raise ModbusIOException("Unable to decode request")
ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "Modbus-RS485-8.py", line 14, in <module>
    print(result.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

I'm starting with RasPi and python. Hope to help. Thanks!

dhoomakethu, 2018-08-20:
Your response `0x1 0x80 0x2 0xc0 0x1` is not valid, it could be that your line is too noisy or you are not using the correct serial settings.  Try using another adapter (may be a different brand ?) Check the raspberry pi forums for the working ones.

JinTran89, 2018-08-20:
Thanks you! I'm trying

closed 2021-03-09 by github-actions[bot]