pymodbus-dev/pymodbus #128: Writing booleans on async server results in struct.unpack error

muldereric, 2016-07-13:
When a boolean is written to the server, the following error occurs:
`Unhandled Error
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 101, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/log.py", line 84, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/local/lib/python2.7/dist-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 597, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 209, in doRead
    return self._dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 215, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/server/async.py", line 59, in dataReceived
    self.framer.processIncomingPacket(data, self._execute)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/transaction.py", line 349, in processIncomingPacket
    result = self.decoder.decode(self.getFrame())
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/factory.py", line 106, in decode
    return self._helper(message)
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/factory.py", line 133, in _helper
    request.decode(data[1:])
  File "/usr/local/lib/python2.7/dist-packages/pymodbus/bit_write_message.py", line 68, in decode
    self.address, value = struct.unpack('>HH', data)
struct.error: unpack requires a string argument of length 4
`

muldereric, 2016-07-28:
Does anyone have an idea on what is going wrong here? 
I'm not a Python guru and just know some basics, but would love to get this one solved with some help.

muldereric, 2016-08-02:
Figured out that the error is caused by the client writing data to the server.
When I use Kepware to simulate my client, everything works as expected, however the other software package does not use/send the CRC byte

dhoomakethu, 2017-06-24:
Pymodbus need complete packet (with CRC) to decode a request, I will close this issue as this is not related to pymodbus.

closed 2017-06-24 by dhoomakethu