pymodbus-dev/pymodbus #134: Not working on Python 3.5.2

eer-act, 2016-10-13:
The example code is not working on python 3.5.2:

```
from pymodbus.client.sync import ModbusTcpClient
client = ModbusTcpClient('127.0.0.1')
client.write_coil(1, True)
result = client.read_coils(1,1)
print(result.bits[0])
client.close()
```
Result:
```
---------------------------------------------------------------------------
ConnectionException                       Traceback (most recent call last)
<ipython-input-9-0e2e7db9a87b> in <module>()
      1 client = ModbusTcpClient('127.0.0.1')
----> 2 client.write_coil(1, True)
      3 result = client.read_coils(1,1)
      4 print(result.bits[0])
      5 client.close()

D:\Programs\Anaconda3\lib\site-packages\pymodbus-1.2.0-py3.5.egg\pymodbus\client\common.py in write_coil(self, address, value, **kwargs)
     63         '''
     64         request = WriteSingleCoilRequest(address, value, **kwargs)
---> 65         return self.execute(request)
     66 
     67     def write_coils(self, address, values, **kwargs):

D:\Programs\Anaconda3\lib\site-packages\pymodbus-1.2.0-py3.5.egg\pymodbus\client\sync.py in execute(self, request)
     80         '''
     81         if not self.connect():
---> 82             raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
     83         return self.transaction.execute(request)
     84 

ConnectionException: Modbus Error: [Connection] Failed to connect[127.0.0.1:502]
```

smurfix, 2016-11-26:
Looks like part of a larger problem.

Is anybody actually working on combining the various Py3 porting efforts out there back into one repository (preferably this one's python3 branch)?

dhoomakethu, 2017-05-17:
addressed with the latest pymodbus v1.3.0rc1 which is compatible with both python2 and python3

closed 2017-05-17 by dhoomakethu

opencv-github, 2020-11-13:
I have  also  this  poblem.Even i  OFF  my  firewall  and  still  it  gives  me  the   connection  error

dhoomakethu, 2020-11-13:
@opencv-github what version of pymodbus are you using? The latest versions should have these issues already fixed. Share the logs and code if you have.