pymodbus-dev/pymodbus #139: Regısters ın Database DataStore cant be read from remote client

sansal54, 2016-11-14:
I try to implement a database backed TCP server so that my registers will be kept in database. For that task I modulated "Updating Server" example . When I run the code I can query regısters from inside. But I cant read same registers from a client located on another computer. With normal ModbusSlaveContext I was able to do that. Results are attached. and here is my code:

```
'''
Pymodbus Server With Updating Thread
--------------------------------------------------------------------------
This is an example of having a background thread updating the
context while the server is operating. This can also be done with
a python thread::
    from threading import Thread
    thread = Thread(target=updating_writer, args=(context,))
    thread.start()
'''
#---------------------------------------------------------------------------# 
# import the modbus libraries we need
#---------------------------------------------------------------------------# 
from pymodbus.server.async import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusServerContext
from database_store import DatabaseSlaveContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer

#---------------------------------------------------------------------------# 
# import the twisted libraries we need
#---------------------------------------------------------------------------# 
from twisted.internet.task import LoopingCall

#---------------------------------------------------------------------------# 
# configure the service logging
#---------------------------------------------------------------------------# 
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#---------------------------------------------------------------------------# 
# define your callback process
#---------------------------------------------------------------------------# 
def updating_writer(a):
    ''' A worker process that runs every so often and
    updates live values of the context. It should be noted
    that there is a race condition for the update.
    :param arguments: The input arguments to the call
    '''
    log.debug("updating the context")
    context  = a[0]
    readfunction = 0x03 # read holding registers
    writefunction = 0x10 # wrıte holding registers
    slave_id = 0x00 # slave address
    address  = 16 # adress : 400017

    values = context[slave_id].getValues(readfunction, address, count=3)

    log.debug("new values: " + str(values))

#---------------------------------------------------------------------------# 
# initialize your data store
#---------------------------------------------------------------------------#
block = ModbusSequentialDataBlock(0x00, [0]*0xff)
store = DatabaseSlaveContext(block)

context = ModbusServerContext(slaves=store, single=True)

#---------------------------------------------------------------------------# 
# initialize the server information
#---------------------------------------------------------------------------# 
identity = ModbusDeviceIdentification()
identity.VendorName  = 'pymodbus'
identity.ProductCode = 'PM'
identity.VendorUrl   = 'http://github.com/bashwork/pymodbus/'
identity.ProductName = 'pymodbus Server'
identity.ModelName   = 'pymodbus Server'
identity.MajorMinorRevision = '1.0'

#---------------------------------------------------------------------------# 
# run the server you want
#---------------------------------------------------------------------------# 
time = 5 # 5 seconds delay
loop = LoopingCall(f=updating_writer, a=(context,))
loop.start(time, now=False) # initially delay by time
StartTcpServer(context, identity=identity, address=("", 5007))

```
[image]
[image]

dhoomakethu, 2017-05-27:
I am sorry if this is late, but are you still having this problem ?

sansal54, 2017-11-24:
Yes I am still having this problem, @dhoomakethu

rahulraghu94, 2017-11-28:
@sansal54 , check out the `dev` branch for a fix. Please get back to us if you still face problems

deverakondabalakrishna, 2018-09-20:
Hi All, Could you please someone help me how to map custom based registers to kept in the database for Modbus?  I am very new to python and working on Modbus. Please help me in this regards.
@sansal54, In the above code "from database_store import DatabaseSlaveContext" -- could you please help me how to create own database according to Modbus registers and add to the server example?

dhoomakethu, 2018-09-21:
@deverakondabalakrishna You need to be more specific on what exactly you would like us to help with. The examples have the explanantion on various datastores and datablocks (refer  [synchronous_server](https://github.com/riptideio/pymodbus/blob/master/examples/common/synchronous_server.py) ). I am pasting the content here for quick reference. 
```
# ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- # 
    # The datastores only respond to the addresses that they are initialized to
    # Therefore, if you initialize a DataBlock to addresses of 0x00 to 0xFF, a
    # request to 0x100 will respond with an invalid address exception. This is
    # because many devices exhibit this kind of behavior (but not all)::
    #
    #     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
    #
    # Continuing, you can choose to use a sequential or a sparse DataBlock in
    # your data context.  The difference is that the sequential has no gaps in
    # the data while the sparse can. Once again, there are devices that exhibit
    # both forms of behavior::
    #
    #     block = ModbusSparseDataBlock({0x00: 0, 0x05: 1})
    #     block = ModbusSequentialDataBlock(0x00, [0]*5)
    #
    # Alternately, you can use the factory methods to initialize the DataBlocks
    # or simply do not pass them to have them initialized to 0x00 on the full
    # address range::
    #
    #     store = ModbusSlaveContext(di = ModbusSequentialDataBlock.create())
    #     store = ModbusSlaveContext()
    #
    # Finally, you are allowed to use the same DataBlock reference for every
    # table or you you may use a seperate DataBlock for each table.
    # This depends if you would like functions to be able to access and modify
    # the same data or not::
    #
    #     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
    #     store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
    #
    # The server then makes use of a server context that allows the server to
    # respond with different slave contexts for different unit ids. By default
    # it will return the same context for every unit id supplied (broadcast
    # mode).
    # However, this can be overloaded by setting the single flag to False and
    # then supplying a dictionary of unit id to context mapping::
    #
    #     slaves  = {
    #         0x01: ModbusSlaveContext(...),
    #         0x02: ModbusSlaveContext(...),
    #         0x03: ModbusSlaveContext(...),
    #     }
    #     context = ModbusServerContext(slaves=slaves, single=False)
    #
    # The slave context can also be initialized in zero_mode which means that a
    # request to address(0-7) will map to the address (0-7). The default is
    # False which is based on section 4.4 of the specification, so address(0-7)
    # will map to (1-8)::
    #
    #     store = ModbusSlaveContext(..., zero_mode=True)
    # ----------------------------------------------------------------------- # 
```

deverakondabalakrishna, 2018-10-08:
@dhoomakethu! Thank you for the above suggestions! I am able to read and write from the database in the slave side and able print on console but when I am trying from client side I am not able to read any data from registers. When I am trying to read hold register address from the client I am getting "Illegal Address". The below are the codes and console output. 
my objective is to get database datastore register values ( server machine ) from client machine.
I am bit of confused with modbusslavecontest and sqlslavecontext. Please help me in this regards.

slave:
def updating_writer(a):

    log.debug("Updating the database context")
    context  = a[0]
    readfunction = 0x03 # read holding registers
    writefunction = 0x10
    slave_id = 0x01 # slave address
    #address = 40010
    #count = 50

    Context[slave_id].setValues(writefunction, 40000, [111]*0x08)
    values1 = context[slave_id].getValues(readfunction, 40000, count=7)
   #log.debug("Values from datastore: " + str(values1))
    print(values1)
    
def run_dbstore_update_server():
    # ----------------------------------------------------------------------- #
    # initialize your data store
    # ----------------------------------------------------------------------- #
    block = ModbusSequentialDataBlock(0x01, [888]*0xFF)
    #store = ModbusSlaveContext(hr=block) 
    store = SqlSlaveContext(block)
    #store = DatabaseSlaveContext(block)
    slaves  = {
                       0x01: store,
                    }
    context = ModbusServerContext(slaves={1: store}, single=False)
    #context = ModbusServerContext(slaves=slaves, single=False)

    # initialize the server information
    # ----------------------------------------------------------------------- #
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'PyModbusServer'
    identity.ProductCode = 'PM'
    #identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Energy Hub'
    identity.ModelName = 'MPP'
    identity.MajorMinorRevision = '1.0'
    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #
    time = 5  # 5 seconds delay
    loop = LoopingCall(f=updating_writer, a=(context,))
    loop.start(time, now=False)  # initially delay by time
    StartTcpServer(context, identity=identity, address=("localhost", 5058))
if __name__ == "__main__":
    run_dbstore_update_server()

console output:
-----------------
 [111, 111, 111, 111, 111, 111, 111, 111]
========================================================
Client:
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.device import ModbusDeviceIdentification
UNIT = 0x01
def run_sync_client():
    client = ModbusClient('localhost', port=5058)
    client.connect()
    log.debug("Reading Device Information")
    information = {}
    rr = None

    while not rr or rr.more_follows:
        next_object_id = rr.next_object_id if rr else 0
        rq = ReadDeviceInformationRequest(read_code=0x03, unit=UNIT,
                                          object_id=next_object_id)
        rr = client.execute(rq)
        information.update(rr.information)
        log.debug(rr)

    print("Device Information : ")
    for key in information.keys():
        print(key, information[key])

    print client.read_holding_registers(40000, 2, unit=UNIT)
    rh1 = client.read_holding_registers(0x16,1, unit=UNIT)
    print(rh1)
    #print(rh1.registers)
 client.close()
if __name__ == "__main__":
    run_sync_client()

console output:
------------------
Device Information : 
(0, 'PyModbusServer')
(1, 'PM')
(2, '1.0')
(4, 'Energy Hub')
(5, 'MPP')
Exception Response(131, 3, IllegalAddress)
Exception Response(131, 3, IllegalAddress)

deverakondabalakrishna, 2018-10-09:
@dhoomakethu Thank you for your prompt reply. I have tested with the above-mentioned branch, even I am getting the same error "IllegalAddress". 
Please suggest me If there are any other ways to access data from database/file to Modbus registers(slave side) and then read from client.

dhoomakethu, 2018-10-09:
@deverakondabalakrishna There are two things you are doing wrong.
1. The starting address for your context is wrong , you are using `40000` , you will have to use zero based index . 
2. In your client, the offsets you are using is wrong ( 40000 == 80000 and 0x16 == 40022) both of them are illegal as your context is created with size (0x08) and the valid registers offsets are between (0 -7)

I have tried your code with the below modifications and it works fine for me.

```
# Updated server code
count = 50
Context[slave_id].setValues(writefunction, 0, [111]*count )
 values1 = context[slave_id].getValues(readfunction, 0, count=count)
```
```
# Client code
print(client.read_holding_registers(0, 2, unit=UNIT))
 print( client.read_holding_registers(0x16, 1, unit=UNIT))
```

deverakondabalakrishna, 2018-10-10:
@dhoomakethu. Thank you very much!
I am able to write and read from the database using "store = ModbusSlaveContext(SqlSlaveContext())". Using this, it will create pymodbus.db file and able to write "db" file and able to read "db" values from the client. The below is code snippets for slave and client:
===================================================================
slave: count=8
    context[slave_id].setValues(writefunction, 0, [111]*count )
    values1 = context[slave_id].getValues(readfunction, 0, count=count)
    print(values1)
    store = ModbusSlaveContext(SqlSlaveContext())
DEBUG:pymodbus.datastore.context:setValues[16] 1:8
DEBUG:pymodbus.datastore.context:getValues[3] 1:8
[111, 111, 111, 111, 111, 111, 111, 111] 
client:
   rh=client.read_holding_registers(0, 4, unit=UNIT)
[111, 111, 111, 111]
===================================================================
but if I am not writing any values to "db" and just I want to read existing "db" file then it giving zeros at both slave and client as shown below console output:
------------------------------------------------------------
slave:
    values1 = context[slave_id].getValues(readfunction, 0, count=count)
DEBUG:pymodbus.datastore.context:getValues[3] 1:8
[0, 0, 0, 0, 0, 0, 0, 0]
client:
    rh=client.read_holding_registers(0, 4, unit=UNIT)
[0, 0, 0, 0]
------------------------------------------------------------
From the above I have a question, Is it possible to read from an existing database(pymodbus.db) file without writing? In my case I have a database file which contineoulsy updating values from other devices and I have to assign this file with modbus slave and read this values from modbus client (PC). It will be great help for me if you give me some inputs on this. 
Thanks!!

dhoomakethu, 2018-10-10:
Just remove the the calls to the `updating_writer` and it works like a normal server .

```
#    loop = LoopingCall(f=updating_writer, a=(context,))
 #   loop.start(time, now=False)  # initially delay by time
```

deverakondabalakrishna, 2018-10-10:
@dhoomakethu. I have tried with removing the calls to the updating_writer and with different combinations, but I am able to read zeros at both slave/client or with some comibinations getting an error "Unhandled error in Deferred:". The below are the slave and client codes, I am a bit of confusion that maybe I am doing wrong. Please once go through my codes and give me your suggestion or modifications in the code. 
slave:
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def updating_writer(a):
    log.debug("Updating the database context")
    context  = a[0]
    readfunction = 0x03 # read holding registers
    writefunction = 0x10
    slave_id = 0x01 # slave address
    count = 8

    #context[slave_id].setValues(writefunction, 0, [111]*count )
    values1 = context[slave_id].getValues(readfunction, 0, count=count)
    print(values1)
    
def run_dbstore_update_server():
    #block = ModbusSequentialDataBlock(0x00, [0]*0xFF)
    #store = ModbusSlaveContext(block) 
    store = ModbusSlaveContext(SqlSlaveContext())
    #store = SqlSlaveContext(block)
    #store = DatabaseSlaveContext(block)
    slaves  = {
                       0x01: store,
                    }
    #context = ModbusServerContext(slaves={1: store}, single=False)
    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'PyModbusServer'
    identity.ProductCode = 'PM'
    #identity.VendorUrl = 'http://github.com/riptideio/pymodbus/'
    identity.ProductName = 'Energy Hub'
    identity.ModelName = 'MPP'
    identity.MajorMinorRevision = '1.0'

    time = 5  # 5 seconds delay
    loop = LoopingCall(f=updating_writer, a=(context,))
    loop.start(time, now=False)  # initially delay by time
    StartTcpServer(context, identity=identity, address=("localhost", 5176))
if __name__ == "__main__":
    run_dbstore_update_server()

[slave.txt](https://github.com/riptideio/pymodbus/files/2463599/slave.txt)

client:
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.device import ModbusDeviceIdentification
UNIT = 0x01
def run_sync_client():
    client = ModbusClient('localhost', port=5176)
    client.connect()
    #log.debug("Reading Device Information")
    information = {}
    rr = None
    while not rr or rr.more_follows:
        next_object_id = rr.next_object_id if rr else 0
        rq = ReadDeviceInformationRequest(read_code=0x03, unit=UNIT,
                                          object_id=next_object_id)
        rr = client.execute(rq)
        information.update(rr.information)
        #log.debug(rr)
    print("Device Information : ")
    for key in information.keys():
        print(key, information[key])

    #print(client.read_holding_registers(0, 2, unit=UNIT))
    rh=client.read_holding_registers(0, 4, unit=UNIT)
    print(rh.registers)
    client.close()
if __name__ == "__main__":
    run_sync_client() 
[client.txt](https://github.com/riptideio/pymodbus/files/2463600/client.txt)

deverakondabalakrishna, 2018-10-11:
@dhoomakethu! I have tested with different scenarios and test cases with "SqlSlaveContext", "ModbusSlaveContext" and "ModbusSlaveContext(SqlSlaveContext())", but I am not able to read values from existing db file using client. The below are the results of different test cases. Please let me know if I am doing anything wrong in the below test cases.

Test1. In this case I used "ModbusSlaveContext(SqlSlaveContext(block))" and the below are the result:
using this "store = ModbusSlaveContext(SqlSlaveContext(block))" able to read values from client when we are setvalues to db. Here It is creating "pymodbus.db" file but it is empty, not updating any values to the file but able to read values from client.

slave:
    `context[slave_id].setValues(writefunction, 0, [111]*count )
    values1 = context[slave_id].getValues(readfunction, 0, count=count)
    print(values1)
    store = ModbusSlaveContext(SqlSlaveContext(block))
    time = 5  # 5 seconds delay
    loop = LoopingCall(f=updating_writer, a=(context,))
    loop.start(time, now=False)  # initially delay by time
    StartTcpServer(context, identity=identity, address=("localhost", 5006))
if __name__ == "__main__":
    run_dbstore_update_server()`
slave console:
DEBUG:root:Updating the database context
DEBUG:pymodbus.datastore.context:setValues[16] 1:8
DEBUG:pymodbus.datastore.context:getValues[3] 1:8
[111, 111, 111, 111, 111, 111, 111, 111]
client:
    `print(client.read_holding_registers(0, 4, unit=UNIT))
    rh1=client.read_holding_registers(0, 4, unit=UNIT)
    print(rh1.registers)`
client console:
ReadRegisterResponse (4)
[111, 111, 111, 111]

It will create pymodbus.db file but it is empty!!

console ouput at slave when connection occur:

DEBUG:pymodbus.server.async:Client Connected [IPv4Address(type='TCP', host='127.0.0.1', port=5013)]
DEBUG:pymodbus.server.async:Data Received: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.factory:Factory Request[43]
DEBUG:pymodbus.server.async:send: 000100000032012b0e0383000005000e50794d6f646275735365727665720102504d0203312e30040a456e657267792048756205034d5050
DEBUG:pymodbus.server.async:Data Received: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:4
DEBUG:pymodbus.datastore.context:getValues[3] 1:4
DEBUG:pymodbus.server.async:send: 00020000000b010308006f006f006f006f
DEBUG:pymodbus.server.async:Data Received: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:4
DEBUG:pymodbus.datastore.context:getValues[3] 1:4
DEBUG:pymodbus.server.async:send: 00030000000b010308006f006f006f006f
DEBUG:pymodbus.server.async:Data Received: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x4 0x0 0x8
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x4 0x0 0x8
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 5:8
DEBUG:pymodbus.datastore.context:getValues[3] 5:8
DEBUG:pymodbus.server.async:send: 000400000013010310006f006f006f006f0000000000000000
DEBUG:pymodbus.server.async:Data Received: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x10
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x10
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 17:16
DEBUG:pymodbus.datastore.context:getValues[3] 17:16
DEBUG:pymodbus.server.async:send: 0005000000230103200000000000000000000000000000000000000000000000000000000000000000
DEBUG:pymodbus.server.async:Client Disconnected: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
DIfferent Scenarios:
case1: if not setting any values to pymodbus.db then it giving zero's at client and slave. (this makes sense may be due to empty pymodbus.db file which pointed out in the above case not updating values to file)
slave:
    `#context[slave_id].setValues(writefunction, 0, [111]*count )
    values1 = context[slave_id].getValues(readfunction, 0, count=count)
    print(values1)
[... 67 lines cut ...]
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:4
DEBUG:pymodbus.datastore.context:getValues[3] 1:4
DEBUG:pymodbus.server.async:send: 00020000000b010308006f006f006f006f
DEBUG:pymodbus.server.async:Data Received: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x4
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:4
DEBUG:pymodbus.datastore.context:getValues[3] 1:4
DEBUG:pymodbus.server.async:send: 00030000000b010308006f006f006f006f
DEBUG:pymodbus.server.async:Data Received: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x4 0x0 0x8
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x4 0x0 0x8
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 5:8
DEBUG:pymodbus.datastore.context:getValues[3] 5:8
[... cut ...]
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x5 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x10 0x0 0x10
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 17:16
DEBUG:pymodbus.datastore.context:getValues[3] 17:16
DEBUG:pymodbus.server.async:send: 0005000000230103200000000000000000000000000000000000000000000000000000000000000000
DEBUG:pymodbus.server.async:Client Disconnected: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]

dhoomakethu, 2018-10-11:
@deverakondabalakrishna apologies for late reply. Your question is too vague and lengthy. With out any code formatting its hard to read . Would you mind telling precisely what exactly are you trying to achieve or the problem you are facing? I would suggest you to use proper formatting of the comments so that its easier to read and people can suggest a solution.

deverakondabalakrishna, 2018-10-11:
@dhoomakethu. Thank you for the reply! In my case, I have a database file which continuously updating values from other devices and I have to assign this file with Modbus slave(beaglebone) and read this values from Modbus client (PC). 
The below code which can able to write(SetValues) values to db file(pymodbus.db) and read(getvalues) values from db file but I am not able to read these registers from client(PC), and getting error "Exception Response(131, 3, IllegalAddress)". Not able to read database values from client.
Getting error in DEBUGG"ERROR:pymodbus.pdu:Exception Response F(3) E(2)" when client connect to slave to read data. (I have copied this output in the below)

slave
```
from pymodbus.server.async import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusServerContext
from pymodbus.datastore.database import SqlSlaveContext
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer
from twisted.internet.task import LoopingCall

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def updating_writer(a):
    log.debug("Updating the database context")
    context  = a[0]
    readfunction = 0x03 # read holding registers
    writefunction = 0x10
    slave_id = 0x01 # slave address
    count = 50

    context[slave_id].setValues(writefunction, 0, [111]*count ) # writes values into pymodbus.db
    values1 = context[slave_id].getValues(readfunction, 0, count=count) #reads values from pymodbus.db
    print(values1)
    
def run_dbstore_update_server():

    block = ModbusSequentialDataBlock(0x00, [0]*0xFF)
    #store = ModbusSlaveContext(block) 
    store = SqlSlaveContext(block) # it will create pymodbus.db with values but not able to read at client

    slaves  = {   0x01: store,
                    }
    context = ModbusServerContext(slaves=slaves, single=False)

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'PyModbusServer'
    identity.ProductCode = 'PM'
    identity.ProductName = 'Energy Hub'
    identity.ModelName = 'MPP'
    identity.MajorMinorRevision = '1.0'

    time = 5  # 5 seconds delay
    loop = LoopingCall(f=updating_writer, a=(context,))
    loop.start(time, now=False)  # initially delay by time
    StartTcpServer(context, identity=identity, address=("localhost", 5020))
if __name__ == "__main__":
    run_dbstore_update_server()
```
slave output:
----------------
DEBUG:pymodbus.datastore.database.sql_datastore:set-values[16] 1:50
DEBUG:pymodbus.datastore.database.sql_datastore:get-values[3] 1:50
[111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111, 111]

client 
```from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.mei_message import ReadDeviceInformationRequest
from pymodbus.device import ModbusDeviceIdentification
UNIT = 0x01
def run_sync_client():
    client = ModbusClient('localhost', port=5020)
    client.connect()
    information = {}
    rr = None
    while not rr or rr.more_follows:
        next_object_id = rr.next_object_id if rr else 0
        rq = ReadDeviceInformationRequest(read_code=0x03, unit=UNIT,
                                          object_id=next_object_id)
        rr = client.execute(rq)
        information.update(rr.information)
        #log.debug(rr)
    print("Device Information : ")
    for key in information.keys():
        print(key, information[key])

    print(client.read_holding_registers(0, 10, unit=UNIT))
    rh1=client.read_holding_registers(0, 10, unit=UNIT) # read hold registers
    print(rh1.registers)

    client.close()
if __name__ == "__main__":
    run_sync_client() 
```
client output:
-----------------
Device Information : 
(0, 'PyModbusServer')
(1, 'PM')
(2, '1.0')
(4, 'Energy Hub')
(5, 'MPP')
Exception Response(131, 3, IllegalAddress)
Traceback (most recent call last):
  File "cdb_v2.py", line 29, in <module>
    run_sync_client() 
  File "cdb_v2.py", line 25, in run_sync_client
    print(rh1.registers)
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
_________
console ouput at slave when connection occure from client: 
---------------------
DEBUG:pymodbus.server.async:Client Connected [IPv4Address(type='TCP', host='127.0.0.1', port=5020)]
DEBUG:pymodbus.server.async:Data Received: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.factory:Factory Request[43]
DEBUG:pymodbus.server.async:send: 000100000032012b0e0383000005000e50794d6f646275735365727665720102504d0203312e30040a456e657267792048756205034d5050
DEBUG:pymodbus.server.async:Data Received: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.database.sql_datastore:validate[3] 1:10
**_ERROR:pymodbus.pdu:Exception Response F(3) E(2)_**
DEBUG:pymodbus.server.async:send: 000200000003018302
DEBUG:pymodbus.server.async:Data Received: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.database.sql_datastore:validate[3] 1:10
**_ERROR:pymodbus.pdu:Exception Response F(3) E(2)_**
DEBUG:pymodbus.server.async:send: 000300000003018302
DEBUG:pymodbus.server.async:Client Disconnected: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]

dhoomakethu, 2018-10-12:
Are you running these examples on the correct branch. Please share the pymodbus version (`pip freeze | grep pymodbus`).
Your example works perfectly on my machine. Also from the server logs I see that request is successfully being processed and sent back to the master. Here are the logs for your reference.

Server
```
DEBUG:pymodbus.server.async:Client Connected [IPv4Address(TCP, '127.0.0.1', 5020)]
DEBUG:pymodbus.server.async:Data Received: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x2b 0xe 0x3 0x0
DEBUG:pymodbus.factory:Factory Request[43]
DEBUG:pymodbus.server.async:send: 000100000032012b0e0383000005000e50794d6f646275735365727665720102504d0203312e30040a456e657267792048756205034d5050
DEBUG:pymodbus.server.async:Data Received: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.database.sql_datastore:validate[3] 1:10
DEBUG:pymodbus.datastore.database.sql_datastore:get-values[3] 1:10
DEBUG:pymodbus.server.async:send: 000200000017010314006f006f006f006f006f006f006f006f006f006f
DEBUG:pymodbus.server.async:Data Received: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.database.sql_datastore:validate[3] 1:10
DEBUG:pymodbus.datastore.database.sql_datastore:get-values[3] 1:10
DEBUG:pymodbus.server.async:send: 000300000017010314006f006f006f006f006f006f006f006f006f006f
DEBUG:pymodbus.server.async:Client Disconnected: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
```

Client:
```
Device Information :
(0, 'PyModbusServer')
(1, 'PM')
(2, '1.0')
(4, 'Energy Hub')
(5, 'MPP')
ReadRegisterResponse (10)
[111, 111, 111, 111, 111, 111, 111, 111, 111, 111]
```

Make sure you are working on the correct branch, If you are running these tests in a separate virtualenv , you can build the dist with `python setup.py sdist` from root directory and do a PIP install with `pip install dist/pymodbus-2.1.0.tar.gz`

deverakondabalakrishna, 2018-10-15:
@dhoomakethu Thank you very much. Now it is working good and I am started working on Sunspec based slave controller. Once again Thanks a lot. 
The problem was I am running in a separate virtualenv.

closed 2018-10-15 by dhoomakethu

Tanjimuddin, 2019-06-14:
Hi, I am trying to send a command from CasModbus client software to Modbus server  running on raspberry pi to write 40001 register value to 50 and save it to the database. Is it possible in this way that you were discussing above?

dhoomakethu, 2019-06-18:
@Tanjimuddin You will have to use a [CustomDatablock](https://github.com/riptideio/pymodbus/blob/dev/examples/common/custom_datablock.py) along with `SqlSlaveContext` to achieve what you are hoping.

Tanjimuddin, 2019-06-18:
@dhoomakethu Thanks for your reply. I used below code and run this in pi. when I send command from CasModbus to write value to database. It shows an illegal data address.Could you please tell me where is the problem?
from __future__ import print_function
from pymodbus.server.asynchronous import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.datastore import SqlSlaveContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer

# --------------------------------------------------------------------------- # 
# configure the service logging
# --------------------------------------------------------------------------- # 

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

# --------------------------------------------------------------------------- # 
# create your custom data block here
# --------------------------------------------------------------------------- # 

class CustomDataBlock(ModbusSparseDataBlock):
    """ A datablock that stores the new value in memory
    and performs a custom action after it has been stored.
    """

    def setValues(self, address, value):
        """ Sets the requested values of the datastore
        :param address: The starting address
        :param values: The new values to be set
        """
        super(ModbusSparseDataBlock, self).setValues(address, value)
        print("wrote {} to {}".format(value, address))

def run_custom_db_server():
    # ----------------------------------------------------------------------- # 
    # initialize your data store
    # ----------------------------------------------------------------------- # 
    block  = CustomDataBlock([0]*100)
    store  = SqlSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)

    # ----------------------------------------------------------------------- #
    # initialize the server information
    # ----------------------------------------------------------------------- #

    identity = ModbusDeviceIdentification()
    identity.VendorName = 'pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'pymodbus Server'
    identity.ModelName = 'pymodbus Server'
    identity.MajorMinorRevision = '2.2.0'

    # ----------------------------------------------------------------------- #
    # run the server you want
    # ----------------------------------------------------------------------- #

    # p = Process(target=device_writer, args=(queue,))
    # p.start()
    StartTcpServer(context, identity=identity, address=("192.168.1.10", 502))

if __name__ == "__main__":
    run_custom_db_server()

[image]

dhoomakethu, 2019-06-19:
Could you please attach the logs as well ?

dhoomakethu, 2019-06-19:
Ok here is the reason why you are getting `invalidAddress` error.
 
The `SqlContext` would create a database by name `pymodbus` in the directory where you are running the script and if you check the db, there are no entries in it by default. You will have to add rows to your table for the register address you want to access/store before trying to write to the db.

```
$ sqlite3 pymodbus.db
SQLite version 3.24.0 2018-06-04 14:10:15
Enter ".help" for usage hints.
sqlite> .tables
pymodbus
sqlite> .headers on
sqlite> select * from pymodbus;
sqlite> .schema pymodbus
CREATE TABLE pymodbus (
	type VARCHAR(1),
	"index" INTEGER,
	value INTEGER,
	CONSTRAINT "key" UNIQUE (type, "index")
);

```
The `types` supported are `h, i, d, c` corresponding to `holding register, input registers, discreteinputs and coils`, the `index` column is the actual offset and `value` is the column where the value is stored. You can create new rows of your something like this `insert into pymodbus(type, "index", value) values('c', 1, 0);` and then your read and write should work.

The context right now does not create new entries if there is matching row available. This is an enhancement that is pending.

Tanjimuddin, 2019-06-27:
@dhoomakethu Thanks for your reply. It is now working after creating rows into the database.

Tanjimuddin, 2019-08-02:
@dhoomakethu is it possible to save sql database 32bit long integer?

dhoomakethu, 2019-08-02:
Yes, it should be possible technically but considering the fact that each
word is 16bit long it is advised to store them as two successive entry in
the dB(16bit).

ಶುಕ್ರ, ಆಗ 2, 2019 13:31 ದಿನಾಂಕದಂದು Tanjim <notifications@github.com> ಅವರು
ಬರೆದಿದ್ದಾರೆ:

Tanjimuddin, 2019-08-02:
@dhoomakethu Thank you for your reply. I am having difficulties to insert unsigned 32bit integer in sql using sqlslavecontext and customdatablock. here attached is a client command and  pymodbusdb. code is also here.

` from __future__ import print_function
from pymodbus.server.asynchronous import StartTcpServer
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import  ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
from pymodbus.datastore.database import SqlSlaveContext
from pymodbus.transaction import ModbusRtuFramer, ModbusAsciiFramer
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)
class CustomDataBlock( ModbusSequentialDataBlock):
    """ A datablock that stores the new value in memory
    and performs a custom action after it has been stored.
    """

    def setValues(self, address, value):
        """ Sets the requested values of the datastore
        :param address: The starting address
        :param values: The new values to be set
        """
        super( ModbusSequentialDataBlock, self).setValues(address, value)
        print("wrote {} to {}".format(value, address))

def run_custom_db_server():
    block  = CustomDataBlock(0,[0]*100)
    store  = SqlSlaveContext(di=block, co=block, hr=block, ir=block)
    context = ModbusServerContext(slaves=store, single=True)
    identity = ModbusDeviceIdentification()
    identity.VendorName = 'pymodbus'
    identity.ProductCode = 'PM'
    identity.VendorUrl = 'http://github.com/bashwork/pymodbus/'
    identity.ProductName = 'pymodbus Server'
    identity.ModelName = 'pymodbus Server'
    identity.MajorMinorRevision = '2.2.0'
    StartTcpServer(context, identity=identity, address=("192.168.1.12", 502))

if __name__ == "__main__":
        run_custom_db_server()`
    
[image]
[image]

PR #351 (2018-10-15): #139 sql db context
Fix #139