pymodbus-dev/pymodbus #147: read_holding_registers give exception after few hours.

mohitbansal111, 2017-01-10:
Hi , 

I am trying to run my application to read register values from a PLC. 
It's loop after 10-20 sec to read registers values and send back to server. This work successfully for couple of hour but after that it's give me exception error 'exceptions.IndexError' : 'list index out of range'.
The code snippet is following: 
self.client = ModbusClient(method='ascii', port='/dev/ttyUSB0', stopbits=1, bytesize=8, parity='N',baudrate=38400, timeout=0.3)
self.client.connect()
try:
rr = self.client.read_holding_registers(list[0], list[1] - list[0] + 1, unit=0x01) #list contain continuous range of registers
if isinstance(rr, ReadHoldingRegistersResponse): 
        reg_val = rr.registers
    else
         reg_val = ''
except Exception, inst:
    print 'exception type:',type(inst)
    print 'exception:', inst
    reg_val = ''

Is my code have some problem or there is some issue in pymodbus code.

dhoomakethu, 2017-05-12:
Could you please attach the traceback?

dhoomakethu, 2017-06-24:
should be fixed with pymodbu 1.3.1 , please reopen in case the issue still persists

closed 2017-06-24 by dhoomakethu