pymodbus-dev/pymodbus #148: BinaryPayloadBuilder object has no attribute to_registers()

milanthakker, 2017-01-10:
I'm trying to use the binary payload builder to initialize a complex data store, but the issue I constantly am running into is that it has no to_registers attribute.  I've tried working around using pack and unpack functions in python but I tend to get a bunch of errors that way.

jotathebest, 2017-01-25:
Have the same issue, could you fix it in some way ?

milanthakker, 2017-01-25:
It looks like that function is broken in some way - I got around it by splitting my complex data into two registers (by default initialized as uint_16s) and configuring my client to look at the registers properly.

To split a float into two uint_16s of binary data: 
`def float_to_uint16(number):
    i1, i2 = unpack('HH', pack('f', number))
    return i1, i2`

Assuming you set your values with: 
`context[slave_id].setValues(register, address, <array of your values>)`
your complex data type is now split over two consecutive registers, and on the other end you'll just have to combine the binary data of the two uint16s on the other end and you'll have the binary of your float.

Complicated, and I probably did a bad job explaining that, but I got it to work.

egarl004, 2017-03-02:
any progress on this? just ran into this issue today.

jjchromik, 2017-03-29:
if your problem was that this function was not existing - it is because in last releases it was not there. I found it e.g. herehttps://github.com/bashwork/pymodbus/blob/master/pymodbus/payload.py :
`    def to_registers(self):
        ''' Convert the payload buffer into a register
        layout that can be used as a context block.

        :returns: The register layout to use as a block
        '''
        fstring = self._endian + 'H'
        payload = self.build()
        return [unpack(fstring, value)[0] for value in payload]`

I added it back to payload - BinaryPayloadBuilder and tested it with few values and so far - it worked... Did anyone experience problems with that function?

egarl004, 2017-04-05:
This helped so much. Working late and you saved the day! Awesome!

Emerson Garland

dhoomakethu, 2017-05-17:
fixed with version 1.3.0.rc1

closed 2017-05-17 by dhoomakethu