pymodbus-dev/pymodbus #151: Some mistake in /exapmle/common/updating-server.py

xiaomingfm, 2017-02-06:
i find some mistake in branch python3's example code:

/exapmle/common/updating-server.py

def updating_writer(a):
    log.debug("updating the context")
    context  = a[0]
    register = 3
    slave_id = 0x00
    address  = 0x10
    values   = context[slave_id].getValues(register, address, count=5)
    values   = [v + 1 for v in values]
    log.debug("new values: " + str(values))
    context[slave_id].setValues(register, address, values)

It should be like this：

def updating_writer(a):
    log.debug("updating the context")
    context  = a
    register = 3
    slave_id = 0x00
    address  = 0x10
    values   = context[slave_id].get_values(register, address, count=5)
    values   = [v + 1 for v in values]
    log.debug("new values: " + str(values))
    context[slave_id].set_values(register, address, values)

i dont know what about other branches,but for pymodbus3 the function name should be 'set_values() 'and 'get_values()'

Astarry, 2017-04-12:
Why do you say get_values(), set_values() should be called? getValues(), setValues() are methods belonging to the class ModbusSlaveContext which is the type of context[slave_id]

dhoomakethu, 2017-05-27:
Not an issue

closed 2017-05-27 by dhoomakethu