pymodbus-dev/pymodbus #155: Publish pymodbus on pypi

andrey-git, 2017-04-15:
Could you publish this package on Pypi, so it can be fetched without specifying the github URL?

home-assistant/home-assistant#7069

Thanks!

DaveTrussel, 2017-05-04:
+1 Or is there a reason only the python 2 version is on PyPI?

dhoomakethu, 2017-05-12:
pymodbus is already available in PyPi.
[PyModbus](https://pypi.python.org/pypi/pymodbus)

closed 2017-05-12 by dhoomakethu

andrey-git, 2017-05-12:
@dhoomakethu That pypi version is 4 years old. Does it support python3?

dhoomakethu, 2017-05-12:
Ah, you are talking about python3, no there is not one on the pypi, you could try this branch .[python3](https://github.com/riptideio/pymodbus/tree/python3) , I will see if there is a version made available for python3 , but I do not have a ready timeline.

MartinBorg, 2017-05-12:
It would be nice to have **one** pymodbus for both 2.7 and 3.4+ at some point.
Are there any plans for a merge?

dhoomakethu, 2017-05-12:
I am currently working on it, I do not have a definite timeline though.

dhoomakethu, 2017-05-17:
A new universal version is now available on pypi

jonadem, 2018-11-09:
It should be nice to state clearly in the package 'pymodbus3' on pypi that 'pymodbus' should be used instead.
https://pypi.org/project/pymodbus3/