pymodbus-dev/pymodbus #159: FifoTransactionManager: Transaction never removed after errorCallback

submerged15, 2017-04-30:
Hi,

I'm using pymodbus over RS485 with RtuFramer and AsyncModbusProtocol based on the twisted serialport. (Python 2.7)

Whenever there are multiple slaves attached and one of them is disabled, there is a reponse/transaction mismatch. The request of the disabled slave never return any response and any further requests with any other slaves too. No timeout is ever reached and I have to call the errorCallback manually.

I figured out that, after the errorCallback is called, the TransactionManager (RtuFramer uses FifoTransactionManager) still contains the failed deferred. Any future request with valid response tries to match with that failed deferred and therefore discard the response. The FifoTransactionManager tries always to match the first element of the list and not the x element based on the Transaction-ID. 

def getTransaction(self, tid):
     _logger.debug("getting transaction %s" % str(tid))
    return self.transactions.pop(0) if self.transactions else None

Workaround: Delete the transaction in the errorCallback method.

Solution: Add handling for failed requests.

dhoomakethu, 2017-05-12:
I believe there is a fix in one of the raised PR's, It will be available as part of 1.3.0 release

dhoomakethu, 2017-05-17:
fixed with pymodbus v1.3.0rc1

closed 2017-05-17 by dhoomakethu