pymodbus-dev/pymodbus #160: Serial + Async + RTU: Callback is never fired

submerged15, 2017-04-30:
Hi again,

I'm using pymodbus over RS485 with RtuFramer and AsyncModbusProtocol based on the twisted serialport. (Python 2.7)

The callback of the method transaction/ModbusRtuFramer/processIncomingPacket is never fired. In this case the method async/ModbusClientProtocol/_handleresponse.

Workaround: Derivate the ModbusClientProtocol, override the dataReceived method and insert the complete content of the processsIncomingPacket and replace the callback with the actuall method _handleresponse.

dhoomakethu, 2017-05-12:
If you have a working solution could you please raise a PR, so that it could be reviewed and merged if necessary?
Thanks

wexi, 2017-05-18:
I probably encounter the same problem with long RTU queries. It is not "firing" since client/async.py dataReceived() receives incomplete data. Indeed, serial.Serial is using non blocking timeout=0, so why would we receive the complete packet if it is long... 
I will try to use timeout=None with a short inter_byte_timeout and report if I am successful.

dhoomakethu, 2017-05-18:
I believe with the latest release 1.3.0rc1, the issue should be fixed. Just give a try and update your findings.

wexi, 2017-05-18:
Thanks, however with pymodbus-1.3.0rc1 I encounter the following unhandled error.

 ~~~~
   File "/home/wexi/.local/lib/python2.7/site-packages/pymodbus/client/async.py", line 93, in dataReceived
     self.framer.processIncomingPacket(data, self._handleResponse)
   File "/home/wexi/.local/lib/python2.7/site-packages/pymodbus/transaction.py", line 604, in processIncomingPacket
     self._process(callback, error=True)
   File "/home/wexi/.local/lib/python2.7/site-packages/pymodbus/transaction.py", line 631, in _process
     raise ModbusIOException("Unable to decode request")
 pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
~~~~

dhoomakethu, 2017-05-19:
@wexi do see this error always ? I am not able to reproduce this at my end. Could you give more info regarding your setup ?

wexi, 2017-05-19:
I solved my problem by simplifying/correcting transaction.py processIncomingPacket(self, data, callback). I can't say that I am satisfied: With a given Modbus query the normal response length is known in advance and this information is not utilized. To your question, I am using the Kivy modified twisted reactor. My OS is Ubuntu. Thanks.

[patch.txt](https://github.com/riptideio/pymodbus/files/1015176/patch.txt)

wexi, 2017-05-27:
Adding to my previous comment regarding async RTU (improper repacking, a criticism which still stands):

To add timeout protection to async RTU we have to addTimeout() to the deferred.
addTimeout, when fired, causes a CancelledError response errback.
**It is pymodbus responsibility** then to remove the transaction from its FifoTransactionManager instance, which it doesn't... bad things then happen, need I explain?

As far as async RTU is concerned we are not at RC quality yet. Please fix :-)

dhoomakethu, 2017-05-27:
If you think you have a working solution for this issue, could you please
raise a PR ?

ಮೇ 27, 2017 11:36 ಅಪರಾಹ್ನ ರಂದು, "Enoch" <notifications@github.com> ಅವರು
ಬರೆದಿದ್ದಾರೆ:

wexi, 2017-05-27:
A proper solution for the addTimeout() bug requires pymodbus to supply a canceller function during the construction of its deferred. This canceller should remove the transaction from the appropriate manager. Since **I am only interested in async RTU now** I took a shortcut. I leave it for those with write access to this repository (aka the maintainers) to do the hard work.

Again, processIncomingPacket(), also requires a rewrite...

dhoomakethu, 2017-05-27:
Thanks for the hints. I do not mind adding more contributors to the
project. Let me know if you need write access to the repo. Anyhow, You can
always fork and raise a PR upstream. This is a community driven project and
more helping hands are highly appreciated.

ಮೇ 28, 2017 2:58 ಪೂರ್ವಾಹ್ನ ರಂದು, "Enoch" <notifications@github.com> ಅವರು
ಬರೆದಿದ್ದಾರೆ:

A proper solution for the addTimeout() bug requires pymodbus to supply a
canceller function during the construction of its deferred. This canceller
should remove the transaction from the appropriate manager. Since *I am
only interested in async RTU now* I took a shortcut. I leave it for those
with write access to this repository (aka the maintainers) to do the hard
work.

Again, processIncomingPacket(), also requires a rewrite...

bhagn, 2017-10-14:
@submerged15, @wexi can you share the code that instantiates the async client? Is the reactor running in its own thread?

wexi, 2017-10-14:
You are welcome to examine my 'local' branch:
https://github.com/wexi/pymodbus/tree/local
See Foreword in README.rst

rahulraghu94, 2018-01-04:
Hello, I'm unable to reproduce this problem as of the new `pymodbus 1.4.0` release and will be closing this now.

If you are still facing this issue, feel free to reopen this and make sure you leave your **code excerpts**, **logs**, **expected outputs** and anything else that can help us track the problem.

Thanks :)

closed 2018-01-04 by rahulraghu94