pymodbus-dev/pymodbus #165: 'WriteMultipleRegistersResponse' object has no attribute 'registers'

oryxr, 2017-05-17:
I use RTU serial sync communication with pymodbus on master .
It works perfectly but sometime this error occur :
`'WriteMultipleRegistersResponse' object has no attribute 'registers'`
I can't repeat error, but when I restart code (master or slave) the error disappear
What do you think
Thanks and sorry for my poor english

oryxr, 2017-05-17:
I this error too, same condition:
`'WriteMultipleCoilsResponse' object has no attribute 'bits':`

dhoomakethu, 2017-05-22:
what version of pymodbus are you working with ?

oryxr, 2017-05-22:
pymodbus 1.2.0 branch python3, same error with python2

dhoomakethu, 2017-05-22:
Please use the latest v 1.3.0rc1 , it is supposed to have a bug fix for similar problem. Let me know your findings

oryxr, 2017-05-24:
I test with the new version and I have the same error

oryxr, 2017-06-08:
always this errors with rc2

dhoomakethu, 2017-06-08:
Are you trying this against any real modbus device ? Can you give some more information regarding your setup and baudrate?

commit f5f9f2c1 (2017-06-22): fix #165, #174, #169, #175, #147, #146
commit e8ef2ef6 (2017-06-24): 1.3.1 release (#185)

* #184 prepare for v1.3.1

* Make examples compatible withg python3

* fix #165, #174, #169, #175, #147, #146

* #184 updated changelog

* #142 move
MaskWriteRegisterRequest/MaskWriteRegisterResponse  to register_write_message.py from file_message.py
closed 2017-06-24 by dhoomakethu