pymodbus-dev/pymodbus #169: 'NoneType' object has no attribute 'registers'

tusharvb19, 2017-05-20:
***Here is my Code

#!/usr/bin/python2.7

from pymodbus.client.sync import ModbusSerialClient as ModbusClient
client = ModbusClient(method='rtu', port='/dev/ttyS1',
                      stopbits=1, bytesize=8, timeout=0.3,
                      baudrate=19200)
client.connect()
debug=True
#enable hardware handshake IMPORTANT
client.socket.rtscts=True

#specify which slave you want to talk to
myUnit=1

#read  Version
reg = client.read_holding_registers(0,1, unit = myUnit)
print reg
print reg.registers

client.close()

***The output

None
Traceback (most recent call last):
  File "test12.py", line 22, in <module>
    print reg.registers
AttributeError: 'NoneType' object has no attribute 'registers'

The code is working I guess but not able to extract the data!
Help needed!

dhoomakethu, 2017-05-22:
Your `read_holding_registers` method is not returning any values. This could either be a problem with your serial device or serial settings you are using to setup the connection. Enabling logs could throw more light on the actual problem. If you think your setup is working fine , try increasing the read timeout and see if that helps.

tusharvb19, 2017-05-23:
@dhoomakethu 
When I am enabling the debug I got the packets and after decoding them they are correct one!

DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 0x1 0x4 0x10 0x3 0xef 0xfe 0xff 0xcf 0x40 0x10 0x2 0x7c 0xbf 0xff 0x80 0xe8 0xff
DEBUG:pymodbus.factory:Factory Response[1]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 1
Traceback (most recent call last):
  File "test.py", line 26, in <module>
    print result.registers 
AttributeError: 'ReadCoilsResponse' object has no attribute 'registers'

I think the data is coming but I am not able to extract it..is it a bug?

ccatterina, 2017-05-23:
I think ReadCoilsResponse has ```bits``` attribute, not ```registers```.
Try this:
```python
print result.bits
```

tusharvb19, 2017-05-23:
But I am using read registers and not coils

dhoomakethu, 2017-05-24:
@tusharvb19 what version of pymodbus are you using? There is a newer version available 1.3.0rc1. please give it a try, the issue you are facing has a fix in there

dhoomakethu, 2017-06-24:
should be fixed with pymodbus==1.3.1 , please reopen in case the issue still persists

closed 2017-06-24 by dhoomakethu