pymodbus-dev/pymodbus #170: ord() expected string of length 1, but int found

oryxr, 2017-05-24:
When I run pymodbus-1..0-rc1 with debug activate, I have this error:
`ord() expected string of length 1, but int found`

I remove ord function in transaction.py and sync.py same as:
`# _logger.warning("cleanup recv buffer before send: " + " ".join([hex(ord(x)) for x in result]))`
to
`_logger.warning("cleanup recv buffer before send: " + " ".join([hex(x) for x in result]))`

and it run.
Why do you add the ord() ?

dhoomakethu, 2017-05-24:
what version of python are you using ? Could you provide the code snippet, which is causing you this error ?

dhoomakethu, 2017-05-24:
If you are using python3, there is a bugfix on the way.

oryxr, 2017-05-24:
I using python3 and when I remove only ord( it's run correctly)

dhoomakethu, 2017-05-24:
A version with the fix will be available by this week.

commit ed0d54d0 (2017-05-25): Fix #170 and other misc updates
closed 2017-05-27 by dhoomakethu