pymodbus-dev/pymodbus #171: struct.error: argument for 's' must be a bytes object

Lstick, 2017-05-25:
When I run pymodbus-1.3.0-rc1 with debug activate, I have this error:

>   builder.add_string('abcdefgh')
>   File "/usr/lib64/python3.4/site-packages/pymodbus-1.3.0rc2-py3.4.egg/pymodbus/payload.py", line 179, in add_string
> struct.error: argument for 's' must be a bytes object
Code is:
`builder = BinaryPayloadBuilder(endian=Endian.Little)
builder.add_string('abcdefgh')
builder.add_32bit_float(22.34)
builder.add_16bit_uint(0x1234)
builder.add_8bit_int(0x12)
builder.add_bits([0,1,0,1,1,0,1,0])
payload = builder.build()`

dhoomakethu, 2017-05-26:
if you are using python3 , pass the string as bytestring , 
`builder.add_string(b'abcdefgh')`

Lstick, 2017-05-26:
Thanks a lot!

closed 2017-05-26 by Lstick