pymodbus-dev/pymodbus #174: What's the Python 3 status?

timothywalsh, 2017-05-29:
Can someone in the know please give an overview of the status of pymodbus and Python 3?

I have managed to get pymodbus to work on python 2.7, however I have already developed the rest of my application using Python 3, and I'm not sure I want to go to the trouble to downgrade to 2.7 if there is another way around it.

I have installed pymodbus3 using pip3, however it doesn't work for me with Python 3.6. It only returns NoneType objects

When I try to install pymodbus for python3 using "`sudo pip3 install pymodbus`", I get the following error

```
Collecting pymodbus
  Using cached pymodbus-1.2.0.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-t1wyyv9y/pymodbus/setup.py", line 26, in <module>
        from pymodbus import __version__, __author__
      File "/tmp/pip-build-t1wyyv9y/pymodbus/pymodbus/__init__.py", line 37
        True, False = (1 == 1), (0 == 1)
           ^
    SyntaxError: can't assign to keyword
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-t1wyyv9y/pymodbus/
```

Which I have seen mentioned in other issues here. 

So, where do I go from here? 

--Tim

timothywalsh, 2017-05-29:
Update: I downloaded the latest source code from Github (version 1.3.0rc2) and installed it with `sudo python3 setup.py install`. It seems to be working now.

Correction: I'm using Python version 3.4.5 here, not 3.6 as stated earlier

dhoomakethu, 2017-05-29:
You can install also with `pip3 install pymodbus` , which should install the latest version on pypi

ptendolkar, 2017-05-30:
@dhoomakethu Tried to do this on fresh python 3 install windows 10 64 bit and got the same error that @timothywalsh had, "can't assign to keyword".

Cloning the repo and running `python setup.py` in the pymodbus directory worked.

dhoomakethu, 2017-05-30:
I will take a look in that case and update

ಅವರು ಬರೆದಿದ್ದಾರೆ:

dhoomakethu, 2017-05-31:
For some reason PIP install was downloading the older version of pymodbus , when you do `pip install pymodbus` , to overcome this you can install explicitly with `pip install pymodbus==1.3.0rc2`

timothywalsh, 2017-06-05:
Hi dhoomakethu,

That's not really a fix... more of a workaround. I would suggest that this issue should be reopened until the root problem is fixed properly, and `pip3 install pymodbus` just works.

--Tim

dhoomakethu, 2017-06-24:
fixed with pymodbus 1.3.1

closed 2017-06-24 by dhoomakethu