pymodbus-dev/pymodbus #181: Issue running Custom Datablock Example

bnjroos, 2017-06-07:
Good evening,

When I try to run the Custom Datablock Example, I get the following error: 
```
Traceback (most recent call last):
  File "modbus-custom-action.py", line 55, in <module>
    block   = CustomDataBlock()
TypeError: __init__() takes exactly 2 arguments (1 given)
```
I have not modified anything, any idea on what needs to be done to make it run ?

Thank you in advance,

Bnjroos

dhoomakethu, 2017-06-08:
Not sure which example are you running ,if your custom datablock is inherited from `ModbusSequentialDataBlock` , you need to pass starting address and values while creating object instance , in case you are creating custom datablock based on `ModbusSparseDataBlock`, you need to pass the values for your datastore (either as list or a dict). You can refer [synchronous server example](https://github.com/riptideio/pymodbus/blob/master/examples/common/synchronous-server.py) for more info on creating datastores.

bnjroos, 2017-06-08:
Thank you for the help, will have a deeper look at it.

closed 2017-06-08 by dhoomakethu