pymodbus-dev/pymodbus #190: Fix examples

dhoomakethu, 2017-07-06:


tryer3000, 2017-08-28:
`examples/functional/synchronous-rtu-client.py` fail to pass on ubuntu 16.04

dhoomakethu, 2017-08-28:
@tryer3000 could you please paste the error trace here ?

tryer3000, 2017-08-29:
@dhoomakethu  FYI

EEEEEE
======================================================================
ERROR: testReadWriteCoil (__main__.SynchronousRtuClient)
----------------------------------------------------------------------
```
Traceback (most recent call last):
  File "/home/joshua/ws/open/pymodbus/examples/functional/base_runner.py", line 33, in testReadWriteCoil
    self.__validate(rq, lambda r: r.function_code < 0x80)
  File "/home/joshua/ws/open/pymodbus/examples/functional/base_runner.py", line 81, in __validate
    else: self.assertTrue(test(result))
  File "/home/joshua/ws/open/pymodbus/examples/functional/base_runner.py", line 33, in <lambda>
    self.__validate(rq, lambda r: r.function_code < 0x80)
AttributeError: 'ModbusIOException' object has no attribute 'function_code'
```

dhoomakethu, 2017-08-29:
Thanks, will have a look

rahulraghu94, 2017-11-24:
@dhoomakethu, are there more examples that are broken apart from the ones that got fixed this week?

closed 2018-01-03 by dhoomakethu