pymodbus-dev/pymodbus #191: [Sync Serial RTU

vermut, 2017-07-09:


vermut, 2017-07-09:
P.S. I'm not sure if `time.sleep(0.000668402777778)` will do something meaningful instead of being rounded up to 0.

dhoomakethu, 2017-07-11:
I agree , its been on my plate for sometime.
`// Modbus states that a baud rate higher than 19200 must use a fixed 750 us 
// for inter character time out and 1.75 ms for a frame delay.
// For baud rates below 19200 the timeing is more critical and has to be calculated.
// E.g. 9600 baud in a 10 bit packet is 960 characters per second
// In milliseconds this will be 960characters per 1000ms. So for 1 character
// 1000ms/960characters is 1.04167ms per character and finaly modbus states an
// intercharacter must be 1.5T or 1.5 times longer than a normal character and thus
// 1.5T = 1.04167ms * 1.5 = 1.5625ms. A frame delay is 3.5T.    

if (baud > 19200)
{
    T1_5 = 750; 
    T3_5 = 1750; 
}
else 
{
    T1_5 = 15000000/baud; 
    T3_5 = 35000000/baud; 
}`

dhoomakethu, 2017-07-11:
Thats a good thing to have but not a high priority item to be considered for development. Though the timings on python is not guaranteed, it should work for most of the cases considering the fact that majority of the hardwares/OS combination on which pymodbus is being run is good.

xtofl, 2017-07-11:
Ok.  Thanks for the quick reply!

dhoomakethu, 2017-08-24:
fixed with 1.3.2

closed 2017-08-24 by dhoomakethu