pymodbus-dev/pymodbus #192: Pymodbus in pyscada

singhaisid0, 2017-07-10:
I am not getting functionality of following program which is using pymodbus library.
Please can anubody help me in this
`# -*- coding: utf-8 -*-
  | from pyscada import log
  |  
  | try:
  | from pymodbus.client.sync import ModbusTcpClient
  | from pymodbus.client.sync import ModbusSerialClient
  | from pymodbus.client.sync import ModbusUdpClient
  | from pymodbus.constants import Defaults
  | driver_ok = True
  | except ImportError:
  | driver_ok = False
  |  
  | from math import isnan, isinf
  | from time import time
  | import traceback
  |  
  | def find_gap(self,L,value):
  | """
  | try to find a address gap in the list of modbus registers
  | """
  | for index in range(len(L)):
  | if L[index] == value:
  | return None
  | if L[index] > value:
  | return index
  |  
  |  
  | def _default_decoder(value):
  | return value[0]
  |  
  |  
  | class RegisterBlock:
  | def __init__(self):
  | self.registers_data = {} # dict of modbus register addresses
  | self.registers = []
  | self.variables = {}
  | self.register_size = 16 # in bits
  |  
  |  
  | def insert_item(self,variable_id,variable_address,decode_value=_default_decoder,variable_length=1):
  | self.variables[variable_id]= {
  | 'decode_function':decode_value,\
  | 'len':variable_length,\
  | 'registers': []\
  | }
  | for idx in range(variable_length/self.register_size):
  | if not self.registers_data.has_key(variable_address+idx):
  | # register will not be queried add register
  | self.registers_data[variable_address+idx] = None
  |  
  | self.variables[variable_id]['registers'].append(variable_address+idx)
  | self.registers = self.registers_data.keys()
  | self.registers.sort()
  |  
  | def check(self):
  | '''
  | check the registerblock
  | '''
  | # todo do check for gaps
  | return True
  |  
  | def _request_data(self,slave,unit,first_address,quantity):
  | return slave.read_input_registers(first_address,quantity,unit=unit)
  |  
  | def request_data(self,slave,unit=0x00):
  | quantity = max(self.registers)-min(self.registers)+1
  | first_address = min(self.registers)
  |  
[... 268 lines cut ...]
  | self._disconnect()
  | return True
  | else:
  | log.info(("device with id: %d is now accessible")%(self.device.pk))
  | return False
  | else:
  | log.error('Modbus Address %d out of range'%self.variables[variable_id].modbusvariable.address)
  | return False
  | elif self.variables[variable_id].modbusvariable.function_code_read == 1:
  | # write coil
  | if 0 <= self.variables[variable_id].modbusvariable.address <= 65535:
  | if self._connect():
  | self.slave.write_coil(self.variables[variable_id].modbusvariable.address,bool(value),unit=self._unit_id)
  | self._disconnect()
  | return True
  | else:
  | log.info(("device with id: %d is now accessible")%(self.device.pk))
  | return False
  | else:
  | log.error('Modbus Address %d out of range'%self.variables[variable_id].modbusvariable.address)
  | else:
  | log.error('wrong type of function code %d'%self.variables[variable_id].modbusvariable.function_code_read)
  | return False

`

dhoomakethu, 2017-07-11:
what exactly is the problem ? Please share any error tracebacks if available , also please provide details regarding the pymodbus version.

singhaisid0, 2017-07-11:
The following link from where I get this code. 
[Pyscada-modbus-device](https://github.com/trombastic/PyScada/blob/dev/0.7.x/pyscada/modbus/device.py)
Actually it's Pyscada program which is using pymodbus library in which I am not getting which type of format it's sending for request to device and which type of format it's getting in reponse.

dhoomakethu, 2017-07-11:
please go through the documentation of [pyscada](http://pyscada.readthedocs.io/en/dev-0.7.x/) or get in touch with the pyscada maintainers on how to use Modbus with pyscada.  Please raise an issue if there is any particular issue with pymodbus .

closed 2017-07-11 by dhoomakethu