pymodbus-dev/pymodbus #196: Slaves sending responses to eachother

jonathan-daniel, 2017-07-21:
Hello,

I created two slaves and one master (RTU), and when I request something from slave id 1 both the slaves get the request, so one of them signals slave 1 doesnt exist and the other one recognizes the id and sends the data back. But when the data is sent back to the master the other slave tries to parse it which leads to an error. 

SLAVE 2 LOG:
DEBUG:pymodbus.server.sync:Started thread to serve client
DEBUG:pymodbus.server.sync:Client Connected [/dev/ttyUSB2:/dev/ttyUSB2]
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 1
DEBUG:pymodbus.server.sync:0x1 0x3 0x2 0x0 0x11 0x78 0x48
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4

SLAVE 1 LOG:
DEBUG:pymodbus.server.sync:Started thread to serve client
DEBUG:pymodbus.server.sync:Client Connected [/dev/ttyUSB1:/dev/ttyUSB1]
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 01030200117848
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 01030200117848
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa

How can I make slave 2 ignore the packets starting with 0x1?
I dont know if this is a bug in the library or Im doing something wrong.

This is the code for slave 2:
```
#!/usr/bin/env python
'''
Pymodbus Synchronous Server Example
--------------------------------------------------------------------------

The synchronous server is implemented in pure python without any third
party libraries (unless you need to use the serial protocols which require
pyserial). This is helpful in constrained or old environments where using
twisted just is not feasable. What follows is an examle of its use:
'''
#---------------------------------------------------------------------------# 
# import the various server implementations
#---------------------------------------------------------------------------# 
from pymodbus.server.sync import StartSerialServer

from pymodbus.device import ModbusDeviceIdentification
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

from pymodbus.transaction import ModbusRtuFramer
#---------------------------------------------------------------------------# 
# configure the service logging
#---------------------------------------------------------------------------# 
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

#---------------------------------------------------------------------------# 
# initialize your data store
#---------------------------------------------------------------------------# 
# The datastores only respond to the addresses that they are initialized to.
# Therefore, if you initialize a DataBlock to addresses of 0x00 to 0xFF, a
# request to 0x100 will respond with an invalid address exception. This is
# because many devices exhibit this kind of behavior (but not all)::
#
#     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
#
# Continuing, you can choose to use a sequential or a sparse DataBlock in
# your data context.  The difference is that the sequential has no gaps in
# the data while the sparse can. Once again, there are devices that exhibit
# both forms of behavior::
#
#     block = ModbusSparseDataBlock({0x00: 0, 0x05: 1})
#     block = ModbusSequentialDataBlock(0x00, [0]*5)
#
# Alternately, you can use the factory methods to initialize the DataBlocks
# or simply do not pass them to have them initialized to 0x00 on the full
# address range::
#
#     store = ModbusSlaveContext(di = ModbusSequentialDataBlock.create())
#     store = ModbusSlaveContext()
#
# Finally, you are allowed to use the same DataBlock reference for every
# table or you you may use a seperate DataBlock for each table. This depends
# if you would like functions to be able to access and modify the same data
# or not::
#
#     block = ModbusSequentialDataBlock(0x00, [0]*0xff)
#     store = ModbusSlaveContext(di=block, co=block, hr=block, ir=block)
#
# The server then makes use of a server context that allows the server to
# respond with different slave contexts for different unit ids. By default
# it will return the same context for every unit id supplied (broadcast
# mode). However, this can be overloaded by setting the single flag to False
# and then supplying a dictionary of unit id to context mapping::
#
#     slaves  = {
#         0x01: ModbusSlaveContext(...),
#         0x02: ModbusSlaveContext(...),
#         0x03: ModbusSlaveContext(...),
#     }
#     context = ModbusServerContext(slaves=slaves, single=False)
#
# The slave context can also be initialized in zero_mode which means that a
# request to address(0-7) will map to the address (0-7). The default is
# False which is based on section 4.4 of the specification, so address(0-7)
# will map to (1-8)::
#
#     store = ModbusSlaveContext(..., zero_mode=True)
#---------------------------------------------------------------------------# 
store = ModbusSlaveContext(
    di = ModbusSequentialDataBlock(0, [17]*100),
    co = ModbusSequentialDataBlock(0, [17]*100),
    hr = ModbusSequentialDataBlock(0, [17]*100),
    ir = ModbusSequentialDataBlock(0, [17]*100))

store2 = ModbusSlaveContext(
    di = ModbusSequentialDataBlock(0, [17]*100),
    co = ModbusSequentialDataBlock(0, [17]*100),
    hr = ModbusSequentialDataBlock(0, [18]*100),
    ir = ModbusSequentialDataBlock(0, [17]*100))

slaves_stores = {
    0x02: store2
}

context = ModbusServerContext(slaves=slaves_stores, single=False)

#---------------------------------------------------------------------------# 
# initialize the server information
#---------------------------------------------------------------------------# 
# If you don't set this or any fields, they are defaulted to empty strings.
#---------------------------------------------------------------------------# 
identity = ModbusDeviceIdentification()
identity.VendorName  = 'Pymodbus'
identity.ProductCode = 'PM'
identity.VendorUrl   = 'http://github.com/riptideio/pymodbus/'
identity.ProductName = 'Pymodbus Server'
identity.ModelName   = 'Pymodbus Server'
identity.MajorMinorRevision = '1.0'

#---------------------------------------------------------------------------#
# run the server you want
#---------------------------------------------------------------------------# 
# Tcp:
#StartTcpServer(context, identity=identity, address=("localhost", 5020))

# Udp:
#StartUdpServer(context, identity=identity, address=("localhost", 502))

# Ascii:
#StartSerialServer(context, identity=identity, port='/dev/pts/3', timeout=1)

# RTU:
StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='/dev/ttyUSB2', stopbits=2, bytesize=8, parity='O', ignore_missing_slaves=True, timeout=.005, baudrate=921600)
```

dhoomakethu, 2017-07-21:
are you running both the servers on the same port ? Please share the client code as well . Also do mention the version of pymodbus on which this is observed.

jonathan-daniel, 2017-07-21:
Thanks for the reaction

The master code:

```
#!/usr/bin/env python

from pymodbus.client.sync import ModbusSerialClient as ModbusClient

import time

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(method='rtu', port='/dev/ttyUSB0', timeout=1, stopbits=1, bytesize=8, parity='N', baudrate=921600)
client.connect()

log.debug("Reading Coils")
rr = client.read_coils(1, 1, unit=0x01)

log.debug("Write to a Coil and read back")
rq = client.write_coil(0, True, unit=1)
rr = client.read_coils(0, 1, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rr.bits[0] == True)          # test the expected value

log.debug("Write to multiple coils and read back- test 1")
rq = client.write_coils(1, [True]*8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
rr = client.read_coils(1, 21, unit=1)
assert(rr.function_code < 0x80)     # test that we are not an error
resp = [True]*21

resp.extend([False]*3)
assert(rr.bits == resp)         # test the expected value

log.debug("Write to multiple coils and read back - test 2")
rq = client.write_coils(1, [False]*8, unit=1)
rr = client.read_coils(1, 8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rr.bits == [False]*8)         # test the expected value

log.debug("Read discrete inputs")
rr = client.read_discrete_inputs(0, 8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error

log.debug("Write to a holding register and read back")
rq = client.write_register(1, 10, unit=1)
rr = client.read_holding_registers(1, 1, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rr.registers[0] == 10)       # test the expected value

log.debug("Write to multiple holding registers and read back")
rq = client.write_registers(1, [10]*8, unit=1)
rr = client.read_holding_registers(1, 8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rr.registers == [10]*8)      # test the expected value

log.debug("Read input registers")
rr = client.read_input_registers(1, 8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error

arguments = {
    'read_address':    1,
    'read_count':      8,
    'write_address':   1,
    'write_registers': [20]*8,
}

log.debug("Read write registeres simulataneously")
rq = client.readwrite_registers(unit=1, **arguments)
rr = client.read_holding_registers(1, 8, unit=1)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rq.registers == [20]*8)      # test the expected value
assert(rr.registers == [20]*8)      # test the expected value

time.sleep(1)

# This is where it crashes, it doesnt matter if i start with 2 or 1, both will work untill switching.

log.debug("======= Read write registeres simulataneously ON THE OTHER SLAVE =====")
rq = client.readwrite_registers(unit=2, **arguments)
rr = client.read_holding_registers(1, 8, unit=2)
assert(rq.function_code < 0x80)     # test that we are not an error
assert(rq.registers == [20]*8)      # test the expected value
assert(rr.registers == [20]*8)      # test the expected value

#---------------------------------------------------------------------------# 
# close the client
#---------------------------------------------------------------------------# 
client.close()
```

One slave is on dev/ttyUSB1 another slave is on dev/ttyUSB2 and the master is on dev/ttyUSB0.
I dont know much about how ports work on Linux, but Im guessing its a different port since its a different dev.

pymodbus version 1.3.1

jonathan-daniel, 2017-07-21:
So these are probably the most useful logs.

As you can see  It seems slave 2 is interpreting the first byte of the slave 1's response as a function code:

SLAVE 1 
```

DEBUG:pymodbus.server.sync:Started thread to serve client
DEBUG:pymodbus.server.sync:Client Connected [/dev/ttyUSB1:/dev/ttyUSB1]
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 01030200123849
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 01030200123849
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.datastore.context:validate[3] 1:1
DEBUG:pymodbus.datastore.context:getValues[3] 1:1
DEBUG:pymodbus.server.sync:send: 01030200123849
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 2
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 2
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 2
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 2
```

SLAVE 2

```
DEBUG:pymodbus.server.sync:Started thread to serve client
DEBUG:pymodbus.server.sync:Client Connected [/dev/ttyUSB2:/dev/ttyUSB2]
DEBUG:pymodbus.server.sync:0x1 0x3
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x0 0x0 0x0 0x1 0x84 0xa 0x1 0x3 0x2 0x0 0x12 0x38 0x49
DEBUG:pymodbus.factory:Factory Request[3]
DEBUG:pymodbus.server.sync:requested slave does not exist: 1
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x1 0x3 0x2 0x0 0x12 0x38 0x49
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x1 0x3 0x0 0x0 0x0 0x1 0x84 0xa 0x1 0x3 0x2 0x0 0x12 0x38 0x49
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
DEBUG:pymodbus.server.sync:0x2 0x3 0x0 0x0 0x0 0x1 0x84 0x39
DEBUG:pymodbus.factory:Factory Request[1]
ERROR:pymodbus.server.sync:Socket error occurred unpack requires a string argument of length 4
```

dhoomakethu, 2017-07-21:
I see your serial configuration are different for server and client 

`StartSerialServer(context, framer=ModbusRtuFramer, identity=identity, port='/dev/ttyUSB2', stopbits=2, bytesize=8, parity='O', ignore_missing_slaves=True, timeout=.005, baudrate=921600)`

`client = ModbusClient(method='rtu', port='/dev/ttyUSB0', timeout=1, stopbits=1, bytesize=8, parity='N', baudrate=921600)`

Also I am not sure how /dev/ttyUSB0 is connected to /dev/ttyUSB2 , ideally the master and slave should be on the two ends of same port `MASTER-------/dev/ttyUSBn-----------SLAVE1---SLAVE2` + you are using high baudrate , I would recommend to use the same serial configuration (same stopbits/parity etc ) and try at lower baudrates to see if everything is working as expected.

jonathan-daniel, 2017-07-26:
My configurations are the same I copied the wrong source, the masters and slave all have each own ports, and I tested various baudrates.

I tried using libmodbus in c and it works, so I know for sure its not the settings or my setup.

dhoomakethu, 2017-07-26:
If it works with libmodbus, it should work with pymodbus as well.I can not think of any reasons why your code is not working If your serial configurations are same for both master and slave (parity, baudrate etc) . Are you using any real modbus device as your slave? Also if possible please share the correct source code you are using (both python and libmodbus) , just share a link .

jonathan-daniel, 2017-07-26:
Very weird things going on, it was working the whole day with libmodbus until some point (I did not change anything in the code or hardware). So you might be right its not a problem in the library, I really dont know. I will create a repository soon with all the code I tried. 

To be clear, everything is one device, my laptop, im just using 3 USB ports with three USB to RS485 converters that are connected to each other. 

I can also make a picture of how the A, B, ground and resistors are connected.
But again, it worked the whole day and stopped (afaik) randomly. same symptoms as before.

closed 2017-07-27 by jonathan-daniel