pymodbus-dev/pymodbus #199: Serial stops reading before full response has been received

Jostikas, 2017-07-31:
On a Delta DVP-SV2, the following:

```python
if client.connect():
    try:
        log.info('Connected.')
        rq = client.write_coil(1, True, unit=1)
        log.info(f'rq: {rq}')
        rr = client.read_coils(1, 1, unit=0x01)
        log.info(f'rr: {rr}')
    except Exception as e:
        log.error(e)
    client.close()
    log.info('Disconnected.')
```

Returns this:
```
INFO:root:Connected.
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x3a 0x30 0x31 0x30 0x35 0x30 0x30 0x30 0x31 0x46 0x46 0x30 0x30 0x46 0x41 0xd 0xa
DEBUG:pymodbus.transaction:recv: 0x3a 0x30 0x31 0x30 0x35 0x30 0x30 0x30 0x31
DEBUG:pymodbus.transaction:getting transaction 1
INFO:root:rq: Modbus Error: [Input/Output] No Response received from the remote unit
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x3a 0x30 0x31 0x30 0x31 0x30 0x30 0x30 0x31 0x30 0x30 0x30 0x31 0x46 0x43 0xd 0xa
WARNING:pymodbus.client.sync:cleanup recv buffer before send: 0x46 0x46 0x30 0x30 0x46 0x41 0xd 0xa
DEBUG:pymodbus.transaction:recv: 0x3a 0x30 0x31 0x30 0x31 0x30 0x31
DEBUG:pymodbus.transaction:getting transaction 2
INFO:root:rr: Modbus Error: [Input/Output] No Response received from the remote unit
INFO:root:Disconnected.```

So, the response arrives, but isn't read. Maybe related to https://github.com/riptideio/pymodbus/issues/68

dhoomakethu, 2017-08-01:
what is the read timeout you are using in your client ?

dhoomakethu, 2017-08-01:
I will also accept this as bug, as I found another issue debugging this issue, as of now the possible workaround is to increase the timeout of the client, when dealing with ascii protocol.

Jostikas, 2017-08-01:
If I understand it correctly, the timeout is the default 3s (which sounds right, considering that if I accidently used the wrong unit ID, it timed out after 3 s). The response comes much faster than this though.

Is there another timeout I need to check? Fact is, that it should not stop reading serial port until the 3s has actually elapsed or \r\n is read. Currently, it does.

dhoomakethu, 2017-08-01:
if you observe this `WARNING:pymodbus.client.sync:cleanup recv buffer before send` in your log, it indicates that the previous read was incomplete or the receive buffer had only partial data . T

Jostikas, 2017-08-01:
Then why did reading from the receive buffer stop? It didn't time out, as I said, both transactions were well under timeout (as in, around 0.2 s in total by eye-feel), and the second transaction already flushes what looks like a valid end of packet from the receive buffer (`WARNING:pymodbus.client.sync:cleanup recv buffer before send: 0x46 0x46 0x30 0x30 0x46 0x41 0xd 0xa`). Am I understanding the valid stop conditions for reading wrong? 1. When timeout (3s) is reached, or 2. When \r\n is received.

dhoomakethu, 2017-08-02:
I do not have any ASCII modbus device to give you an exact answer , My observations are based on the logs provided by you. As you can see the both recv transaction are incomplete , the only reason I could think is of timeout or wrong response length being calculated by pymodbus (ref: [transaction.py](https://github.com/riptideio/pymodbus/blob/master/pymodbus/transaction.py#L126), With the ASCII simulations,I could confirm that the expected response length calculated by pymodbus is correct. . If possible could you please configure the device in RTU mode and retry your transactions ?
I will try to reproduce this with simulations and see if I can get any lead.

Jostikas, 2017-08-05:
Currently, the expected length is calculated as ´self.base_adu_size + expected_pdu_size´. Checking the get_expected_pdu_size methods, it seems to return the size as it would be for a RTU message. In ASCII mode, each RTU byte is encoded as two bytes, which is not accounted for, it seems (or I couldn't find where it takes place). Except for the starting colon (0x3a) and the ending \r\n (0xd 0xa), The rest is the ASCII codes for letters of the hex representation, 0..9 (0x30..0x39) and A-F (0x41..0x46).

I'm currently doing something else, so I also don't have access, but this seems like a likely place to look for.
Citing http://www.simplymodbus.ca/ASCII.htm, 

Modbus Mode | data (ASCII) | data (hex) | data (binary)
-- | -- | -- | --
Modbus RTU | ® | AE | 1010 1110
Modbus ASCII | A , E | 41 , 45 | 100 0001 , 100 0101

So it seems to me, like ´base_adu_size´ for ascii framer should be ´self.base_adu_size = 7 # Start (1), Address(2), LRC(2), End(2)´ and ´_calculate_response_length´ should ´return self.base_adu_size + expected_pdu_size * 2´. That gives the correct count (17 for bit_write_message: 2*5 for pdu, 7 for start, lrc and stop).

Although isn't that supposed to be handled by the ...Framer? To be honest, I'm not 100% on the structure right now, which is the main reason I'm not sending this as a pull request (don't have a way to test it myself right now, either).

dhoomakethu, 2017-08-06:
good point, there is a fix for expected length for ASCII already , refer [this line](https://github.com/riptideio/pymodbus/blob/dev/pymodbus/transaction.py#L116)  in `dev` branch.  May be that is the reason why the code worked fine for  me as I generally will be on dev branch.
If you would like to give a try, there is an pre release available with this fix [1.3.2rc1](https://github.com/riptideio/pymodbus/releases/tag/v1.3.2rc1) , do give a try and let me know if that solves the issue. I

dhoomakethu, 2017-08-23:
pymodbus 1.3.2 is out with the fix for this issue. Please re-open if still facing the issue.

closed 2017-08-23 by dhoomakethu