pymodbus-dev/pymodbus #200: 'Unable to decode request' with Modbus rtu over TCP

ccatterina, 2017-08-02:
Hi, 
We use many RS485 to Ethernet converter to read data with Modbus RTU over TCP and I've noticed that with  the version ```1.3.1``` of pymodbus the failure rate has increased. Most of the time  i receive ```Modbus Error: [Input/Output] Modbus Error: [Input/Output] Unable to decode request```. 

I think is the result of calling ```socket.recv``` with the expected length of the response instead of use ```1024``` when there is noise on the serial line, in fact with ```1024``` the buffer was cleaned of the residual bytes of noise.

dhoomakethu, 2017-08-03:
starting with pymodbus 1.3.0, pymodbus implements three new feature to speedup transactions and reduce error rates.

- Clears read buffer before starting a transaction
- Calculates the expected response length beforehand to reduce the wait time on recv.
- retry recv in case of incomplete response recieved.

The decode error happens when the received packet fails CRC check or is corrupt. Just to make sure we are in the correct page, could you please enable the debug logs and paste the transactions for which you are having this error ? You could also enable `retry_on_empty` and increase `retries` and see if that has any positive effect.
``` 
# * retries - Specify how many retries to allow per transaction (default = 3)
--
# * retry_on_empty - Is an empty response a retry (default = False)
```

ccatterina, 2017-08-03:
I do some test with debug logs enabled, the problem seems different from what i expected:

```
correct responses from first transaction
...
...
DEBUG:pymodbus.transaction:Running transaction 23
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x53 0xc0 0xdd
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 23
ReadRegisterResponse (1)
DEBUG:pymodbus.transaction:Running transaction 24
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x53 0xc0 0xdd
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 24
ReadRegisterResponse (1)
DEBUG:pymodbus.transaction:Running transaction 25
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x53 0xc0 0xdd
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 25
ReadRegisterResponse (1)
DEBUG:pymodbus.transaction:Running transaction 26
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x53 0xc0 0xdd
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 26
ReadRegisterResponse (1)
DEBUG:pymodbus.transaction:Running transaction 27
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x46 0x40 0x60 0x60 0x4a 0x75 0xff
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x51 0x41 0x1c
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x51 0x41 0x1c
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
DEBUG:pymodbus.transaction:getting transaction 27
Modbus Error: [Input/Output] Modbus Error: [Input/Output] Unable to decode request
DEBUG:pymodbus.transaction:Running transaction 28
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x51 0x41 0x1c
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x51 0x41 0x1c
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
DEBUG:pymodbus.transaction:send: 0x33 0x3 0x3 0xe8 0x0 0x1 0x0 0x68
DEBUG:pymodbus.transaction:recv: 0x33 0x3 0x2 0x2 0x51 0x41 0x1c
DEBUG:pymodbus.factory:Factory Response[70]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 70
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
...
...
continue to return these errors.
```

ccatterina, 2017-08-03:
> Btw, what client are you using , sync or async ?

sync client.

ccatterina, 2017-08-08:
I tried the version ```v1.3.2rc1```, the issue seems fixed.

dhoomakethu, 2017-08-08:
Thanks, I will keep it open for some time for you to confirm .

closed 2017-08-11 by ccatterina