pymodbus-dev/pymodbus #201: ModbusRtuFramer.advanceFrame() is never called after transmission errors

MrLobs, 2017-08-04:
A message like `0x1 0x5 0x0 0x0 0xff 0x0 0x0 0x0` to `ModbusSerialServer` causes the following events:  

- In `processIncomingPacket()`:
- `checkFrame()` returns `False`
- in `_process(callback, error=True)` `data` is assigned to `getRawFrame()`
- in `self.decoder.decode(data)` it expects 4 bytes of data (`self.address, self.count = struct.unpack('>HH', data)`), but gets the entire frame. *Rises an exception*
- in `_process(callback, error=True)` `self.advanceFrame()` *does not get called*

Now the server is broken! Any incoming data appends to `ModbusRtuFramer.__buffer` and the same events happen over and over

dhoomakethu, 2017-08-04:
could you please check this issue with [1.3.2rc1](https://github.com/riptideio/pymodbus/releases/tag/v1.3.2rc1)

MrLobs, 2017-08-07:
Sorry for the delay  
Yes, the issue has gone, thank you!

closed 2017-08-07 by MrLobs