pymodbus-dev/pymodbus #204: Unable to read register values from Modbus rtu

anjamago, 2017-08-15:
Hello,
I am trying to read data from modbus rtu without success
I do not have much experience in modbus but I tried to resivir the data of my automata which is connected by serial to my raspberry atrabes of a usb to rs485 the only thing I notice is that the client connects but from now on I can not do more Nothing i can not remember

My program is as follows
`
#!/usr/bin/env python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(method='rtu', port='/dev/ttyUSB1', timeout=2, baudrate=9600, stopbits=1, bytesize=8, parity="N")
client.connect()

log.debug("Reading Coils")
rr = client.read_coils(0, 1, unit=0x01)

log.debug("Read input registers")
rr = client.read_input_registers(0, 4, unit=1)

client.close()
` 

my answer

`
DEBUG:root:Reading Coils
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1 0x1 0x0 0x0 0x0 0x1 0xfd 0xca
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 1
DEBUG:root:Read input registers
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x1 0x4 0x0 0x0 0x0 0x4 0xf1 0xc9
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv:
DEBUG:pymodbus.transaction:getting transaction 2

`

dhoomakethu, 2017-08-15:
I see no problem with  the communication from pymodbus side based on your logs, please ensure you are using the same serial settings on both raspberry pi as with pymodbus. Also make sure to enable the termination on the USB-RRS485 adapter if any. Check the logs of your modbus application on raspberry pi and see if the packets are reaching it.

anjamago, 2017-08-15:
The truth would not know how to configure my rs485 usb to resivir the data, could you give me an example of how to do it would be of great help

dhoomakethu, 2017-08-17:
I am sorry but without the details of the hardware you are using I am unable to help.

anjamago, 2017-08-24:
the usb conversion a [rs485 ](https://www.vistronica.com/comunicaciones/convertidor-usb-a-rs485-detail.html?gclid=Cj0KCQjw8vnMBRDgARIsACm_BhJoBwc_l2mpNOi711IIIovoavYeszmShO7JNJ0i6Eot7TgbqbfiQtQaAjCIEALw_wcB)
the plc is a AMP1-E  
Manufactured in the city of medellin colombia is not very commercial yet so there is no information on the internet

dhoomakethu, 2017-08-26:
This is something I do not have access to, Can you use some other USB-RS485 converter, something like this https://sgcdn.startech.com/005329/media/sets/ICUSB422IS_Manual/ICUSB422IS.pdf ?

dhoomakethu, 2017-10-03:
@anjamago Is there an update for this issue ?

dhoomakethu, 2017-11-06:
No update from the creator, closing this for the time being.

closed 2017-11-06 by dhoomakethu