pymodbus-dev/pymodbus #207: checksum error : rolling response or delay to answer

Grominet, 2017-08-17:
Hello,

Sorry, I am not an Modbus or python expert, but I don't know how continue to solve my issue :
I've an modbus equipment (Nilan combi 300), and use an FDTI USB plugin RPI to communicate.

My issue is I can't get a stable response, and that give checksum error. I've two script to explained issue :
- one the response is stable 
- second the response is "rolling", and I success to get one right response after 5-6times. 

I never have issue with transmitting a request, the equipment react always at first sending, but response is always... uncertain. 
I have also try another USB-serial chip, same issue.
Note the first byte from stable response is always an empty byte, so I've put some resistor for biasing but without more success, and with two type of cable

## First script output (stable, always the same):
```
sudo python nilan2.py
TESTING Nilan Connection
__main__.Nilan<id=0x76c99238, address=30, mode=rtu, close_port_after_each_call=False, precalculate_read_size=False, debug=True, serial=Serial<id=0x76ab0530, open=True>(port='/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_00000000-if00-port0', baudrate=19200, bytesize=8, parity='E', stopbits=1, timeout=1.1, xonxoff=False, rtscts=False, dsrdtr=False)>

MinimalModbus debug mode. Writing to instrument (expecting 1000 bytes back): '\x1e\x03\x03\xeb\x00\x01\xf6\x15' (1E 03 03 EB 00 01 F6 15)
MinimalModbus debug mode. No sleep required before write. Time since previous read: 1502978904014.4 ms, minimum silent period: 2.01 ms.
MinimalModbus debug mode. Response from instrument: '\x00\x1e\x03\x02\x00\x02\xacG\x00' (00 1E 03 02 00 02 AC 47 00) (9 bytes), roundtrip time: 1101.7 ms. Timeout setting: 1100.0 ms.

Traceback (most recent call last):
  File "nilan2.py", line 68, in <module>
    old_userVent_value = n.get_userVent()
  File "nilan2.py", line 45, in get_userVent
    return self.read_register(1003, numberOfDecimals=0, signed=False, functioncode=3)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 258, in read_register
    return self._genericCommand(functioncode, registeraddress, numberOfDecimals=numberOfDecimals, signed=signed)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 697, in _genericCommand
    payloadFromSlave = self._performCommand(functioncode, payloadToSlave)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 798, in _performCommand
    payloadFromSlave = _extractPayload(response, self.address, self.mode, functioncode)
  File "/usr/local/lib/python2.7/dist-packages/minimalmodbus.py", line 1075, in _extractPayload
    raise ValueError(text)
ValueError: Checksum error in rtu mode: 'G\x00' instead of '\\\x1b' . The response is: '\x00\x1e\x03\x02\x00\x02\xacG\x00' (plain response: '\x00\x1e\x03\x02\x00\x02\xacG\x00')
```

First script .py : 
```
#!/usr/bin/env python

#reference https://minimalmodbus.readthedocs.io/en/master/apiminimalmodbus.html#minimalmodbus.Instrument.read_register
import minimalmodbus
import serial

__author__  = "Nick Ma"

class Nilan( minimalmodbus.Instrument ):
    """Instrument class for nilan heat pump.
    communication via RS485
    """

    HOLDINGREG_OFFSET = 10000

    def __init__(self, portname, slaveaddress=30):
        minimalmodbus.Instrument.__init__(self, portname, slaveaddress)

        self.serial.baudrate = 19200
        self.serial.stopbits = 1
        self.serial.bytesize = 8

        self.serial.parity = serial.PARITY_EVEN
        self.serial.timeout = 1.1  #timeout to 1000ms because I discovered roundtrip times as high as 898.5 ms

        self.serial.rtscts = False

        self.mode = minimalmodbus.MODE_RTU
        self.precalculate_read_size = False
[... 174 lines cut ...]
        instrument.serial.timeout  = 1.1   # seconds
        instrument.serial.parity   = minimalmodbus.serial.PARITY_EVEN
        instrument.CLOSE_PORT_AFTER_EACH_CALL = True
        instrument.debug = True
        print time.strftime('%X %x %Z')

        ## Read temperature (PV = ProcessValue) ##
        temp_maison = lire_nilan(210, 2, 4, True) # Registernumber, number of decimals
        print 'Temp. maison = ', temp_maison
        if temp_maison < 5 :
            os.exit()

        temp_puls = lire_nilan(207, 2, 4, True) # Registernumber, number of decimals
        print 'Temp. air pulsee = ', temp_puls

        ventil_consigne = lire_nilan(1003, 0, 3, True)
        print 'Ventilation 1003 = ', ventil_consigne

#        ecrire_nilan(1003,2,0)

    except:
        print 'Exception'
        print traceback.format_exc()
        sys.exit(0)
```

dhoomakethu, 2017-08-17:
Hello, This repo deals with `pymodbus` and not `minimal modbus`, please raise the issue against [minimal-modbus](https://github.com/pyhys/minimalmodbus). feel free to raise an issue here, If you face any issues using pymodbus.

Grominet, 2017-08-17:
Hello, 
Sorry for first post, please see edited issue. 
And on fact i've same issue with minimalmodbus
Thanks

dhoomakethu, 2017-08-17:
could you please provide the complete client initialization code, the one in the snippet is incomplete
`client = ModbusClient(method='rtu', port='/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_00000000-if00-port0', timeout = 1 ,stopbits = 1, bytesize = 8, parity = 'E', baudrate = 19200, loopback$`

I want to know, what is loopback here and why are you using it?

dhoomakethu, 2017-08-17:
`DEBUG:pymodbus.transaction:send: 0x1e 0x3 0x0 0xcf 0x0 0x1 0xb6 0x5a
DEBUG:pymodbus.transaction:recv: 0x0 0x1e 0x83 0x2 0xf1 0x37 0x0`

If you ignore the first byte in your recv message (0x0), the rest of the data makes a valid modbus error response, which is for `0x02 Illegal Data Address` , so is `207` a valid address ? Could you check with the specs ?

Grominet, 2017-08-17:
Hi, thanks for looking.
for loopback, it's set to False (as default). I was trying to set it to True one time to see if it that was that parameter cause "rolling". 

207 is not valid for input header, only holding register, you're right.
Here I try with a valid one for the input header :
```
DEBUG:root:Reading Holding Regist
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x1e 0x3 0x0 0xc9 0x0 0x1 0x56 0x5b
DEBUG:pymodbus.transaction:recv: 0x0 0x1e 0x3 0x2 0x19 0x64 0x27
DEBUG:pymodbus.factory:Factory Response[0]
ERROR:pymodbus.factory:Unable to decode response Modbus Error: Unknown response 0
DEBUG:pymodbus.transaction:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
```
Always this empty byte... I can't get a "correct" answer, or a valid modbus error :)

dhoomakethu, 2017-08-21:
If you ignore the first byte in the response (`0x0`), you are actually getting a valid response (not verified though) ` 0x1e 0x3 0x2 0x19 0x64 0x27`. I am not sure why the device is responding with the extra `0x0` in your response.

Grominet, 2017-08-21:
Ok, maybe have you some idea to do it?
Which function I can use to?
Or which parameter could help ?
Thanks to bring me on the way.

dhoomakethu, 2017-08-21:
I am afraid this falls outside the purview of pymodbus, unless a proper response is recieved pymodbus could not do much. you could probably check if your connections are proper and try with other baudrate and see if the communication gets stable at any of the other baudrates.

dhoomakethu, 2017-10-03:
Not an issue with pymodbus

closed 2017-10-03 by dhoomakethu