pymodbus-dev/pymodbus #210: Release tar ball without generated files?

MartinBorg, 2017-08-27:
Hi,

while I think, that for some users it is convenient to have a release .tar.gz file with all documentation pre-build, it makes life slightly harder for redistributors (like me, working on the Debian pymodbus package). For Debian packages (and probably most other Linux distributions), it is required, that all non-source files are build from source. One can, of course, just remove the files and rebuild them, but having a pristine tar file from you would be preferable. (E.g. users can easily check, that Debian distributes the original sources by comparing the checksum of the tar.gz).

Many thanks in advance!

PS: Also, I wonder wether maintaining generated files in version control is a good idea? In my experience, it is not, but YMMV, of course.

dhoomakethu, 2017-08-28:
The release is entirely controlled by GITHUB as of now, All I am doing is just click the button `Draft a new release`, create a tag and publish. I will have to go through the release documentation to fine tune it (to what you are asking). The majority of the users use the package via PIP and I know only a few people who depends on the release here on the github repo (cc. @markmevans )

MartinBorg, 2017-08-28:
I think, that both the pypi and the github archive have the pre-compiled documentation included, which is probably, because it is already in the version control.

markmevans, 2017-08-29:
I also require a source tarball to build from.  I ignore the generated documentation so this specific issue does not directly affect me but agree that it really does not need to be under source control.

I'd vote to remove the generated documentation files under source control and fixing the hooks to  [readthedocs.org]https://readthedocs.org/projects/pymodbus/) but I don't know what that entails.

See: http://dont-be-afraid-to-commit.readthedocs.io/en/latest/documentation.html

dhoomakethu, 2017-08-30:
@markmevans , JFYI  the latest documentation is upto date here [pymodbus-n](http://pymodbus-n.readthedocs.io/en/latest/), I do not have access to the original pymodbus documentation

dhoomakethu, 2017-10-06:
@MartinBorg @markmevans I have added a gitattributes files which should exclude the documentation/tests and other un releated stuff the generate release tarball. Refer PR #223 .

Please let me if this addresses the issue you raised in this issue.
[.gitattributes](https://github.com/riptideio/pymodbus/pull/223/files#diff-fc723d30b02a4cca7a534518111c1a66)

MartinBorg, 2017-10-09:
Thanks for tackling this issue! I never used `.gitattributes`, so I'm not sure what it does. But it looks like it would leave out the documentation, examples etc. entirely, right? If so, this is not what I would want, because we do ship the documentation and examples in Debian. We just want to build them ourselves, not use prebuild software or documenation (= "generated files"). I suggest not to have generated files files (e.g. the HTML files of the docs) in git at all, but only the source files (= .rst etc.).

dhoomakethu, 2017-10-10:
@MartinBorg thanks for the feedback, I will see what needs to be done.

dhoomakethu, 2017-12-20:
@ghost, I have removed all unwanted files from the doc folder in this branch [tar-ball-without-docs](https://github.com/riptideio/pymodbus/tree/tar-ball-without-docs) . Please give it a try and see if that addresses your requirement with this issue. Sorry for keeping it open for a long time.

commit 7929316d (2017-10-06): Update .gitattributes

Fix #210
PR #252 (2017-12-21): Tar ball without docs
Fixes #210 and replaces #223  as PR candidate.
commit 942e9065 (2017-12-21): Tar ball without docs (#252)

* Trying to fix #210
1. Using if __name__ == "__main__" clause in example to avoid running them while generating docs
2. Trying to enforce PEP-8 compliant code + PYTHON2-3 compatibale code with examples as much as possible
3. Renaming examples with "-" to "_" for the ease of documentation generation
4. Minor changes  with setup and setup commands
5. Support documentation creation from make file
6. Remove unnecessary html and pdf files from the doc folder

* Fix makefile

* Remove unwanted files and folders from doc folder

* Add seperated requirements for Docs and [... cut]
commit 621a90e3 (2017-12-22): Tar ball without docs (#252)

* Trying to fix #210
1. Using if __name__ == "__main__" clause in example to avoid running them while generating docs
2. Trying to enforce PEP-8 compliant code + PYTHON2-3 compatibale code with examples as much as possible
3. Renaming examples with "-" to "_" for the ease of documentation generation
4. Minor changes  with setup and setup commands
5. Support documentation creation from make file
6. Remove unnecessary html and pdf files from the doc folder

* Fix makefile

* Remove unwanted files and folders from doc folder

* Add seperated requirements for Docs and [... cut]
closed 2018-01-03 by dhoomakethu