pymodbus-dev/pymodbus #212: InvalidMessageRecievedException when try to communicate with a modbus slave device (RTU)

thaxy, 2017-09-08:
Hello,

When I try to read modbus registers from a slave through a serial port via pymodbus:
I get the following error:

```
thaxy@thaxy:~/Documents/client$ docker-compose up
Starting client_modbus_slave_1 ... 
Starting client_modbus_slave_1 ... done
Starting client_modbus_master_1 ... 
Starting client_modbus_master_1 ... done
Attaching to client_modbus_slave_1, client_modbus_master_1
modbus_master_1  | [11:53:53.504] DEBUG [__main__.<module>:61] Initializing the client
modbus_master_1  | [11:53:53.520] DEBUG [master.config.validator._validate_with_yaml_schema:54] Config file valid according to yml schema.
modbus_master_1  | [11:53:53.520] DEBUG [master.config.core._load:63] Finished loading yaml file. Config is valid
modbus_master_1  | [11:53:53.521] DEBUG [__main__.__init__:31] Initializing serial port: '/dev/ttyS0', {'baudrate': 9600, 'parity': 'N', 'bytesize': 8, 'timeout': 1.0, 'stopbits': 1}
modbus_master_1  | [11:53:53.522] DEBUG [pymodbus.client.async.connectionMade:74] Client connected to modbus server
modbus_master_1  | [11:53:53.522] INFO [master.protocol.connectionMade:33] [/dev/ttyS0] Connection made. Reading settings are [{'start_address': 0, 'offset': 30001, 'device_id': 1, 'count': 10}]
modbus_master_1  | [11:53:53.522] DEBUG [__main__.<module>:92] Starting the reactor
modbus_master_1  | [11:53:54.524] DEBUG [master.protocol.fetch_modbus_table:42] [/dev/ttyS0] Start to fetch modbus table(s)
modbus_master_1  | [11:53:54.524] DEBUG [pymodbus.transaction.addTransaction:309] adding transaction 1
modbus_master_1  | [11:53:54.585] DEBUG [pymodbus.factory._helper:238] Factory Response[1]
modbus_master_1  | Unhandled Error
modbus_master_1  | Traceback (most recent call last):
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/log.py", line 103, in callWithLogger
modbus_master_1  |     return callWithContext({"system": lp}, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/log.py", line 86, in callWithContext
modbus_master_1  |     return context.call({ILogContext: newCtx}, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/context.py", line 122, in callWithContext
modbus_master_1  |     return self.currentContext().callWithContext(ctx, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/context.py", line 85, in callWithContext
modbus_master_1  |     return func(*args,**kw)
modbus_master_1  | --- <exception caught here> ---
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/posixbase.py", line 597, in _doReadOrWrite
modbus_master_1  |     why = selectable.doRead()
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/_posixserialport.py", line 61, in doRead
modbus_master_1  |     return fdesc.readFromFD(self.fileno(), self.protocol.dataReceived)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/fdesc.py", line 94, in readFromFD
modbus_master_1  |     callback(output)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/client/async.py", line 93, in dataReceived
modbus_master_1  |     self.framer.processIncomingPacket(data, self._handleResponse)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/transaction.py", line 682, in processIncomingPacket
modbus_master_1  |     self._process(callback, error=True)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/transaction.py", line 711, in _process
modbus_master_1  |     raise InvalidMessageRecievedException(result)
modbus_master_1  | pymodbus.exceptions.InvalidMessageRecievedException: Modbus Error: [Invalid Message] ReadBitResponse(48)
modbus_master_1  | 
modbus_master_1  | [11:53:54.590] ERROR [master.protocol.connectionLost:37] [/dev/ttyS0] Connection lost
^CGracefully stopping... (press Ctrl+C again to force)
Stopping client_modbus_master_1 ... done
Stopping client_modbus_slave_1  ... done
```

As you can see I get a response from the slave but pymodbus got problems decoding the response.
I use the following framer:
`framer = ModbusRtuFramer(ClientDecoder())`
When I emulate a slave with pymodbus using the ``StartSerialServer`` I successfully retrieve the registers. Any suggestions on that. Am I missing something?

Thanks in advance
thaxy

dhoomakethu, 2017-09-08:
what version of Pymodbus are you using ?

thaxy, 2017-09-08:
Sorry, my bad v1.3.2

dhoomakethu, 2017-09-08:
Thanks,  It is interesting to know that the read works fine with emulated pymodbus server and modbus sync client. Would it be possible for you to share your client code and the debug logs with  sync client ?

thaxy, 2017-09-08:
It seems that the read always works fine with the emulated ``pymodbus`` slave no matter if use sync or async (via ``twisted``) master.
But when I connect to a slave device emulated by http://modbustools.com/modbus_slave.html only the sync request of ``libmodbus`` seems to work the async gets a response but has a problem decoding it correctly.

How can I send you the source code in private?

thaxy, 2017-09-08:
Maybe this is intersting too (sync client):

```
>>> rr = client.read_holding_registers(1, 1, unit=1)
DEBUG:pymodbus.transaction:Running transaction 3
DEBUG:pymodbus.transaction:send: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
WARNING:pymodbus.client.sync:cleanup recv buffer before send: 0x58 0x0 0x22 0x0 0x22 0x0 0x22 0x1 0x59 0x1 0x59 0x18 0xca 0x0 0x22 0xe0 0x5d
DEBUG:pymodbus.transaction:recv: 0x1 0x3 0x2 0x0 0x12 0x38 0x49
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 3
>>> rr
<pymodbus.register_read_message.ReadHoldingRegistersResponse object at 0x7f261b2994e0>
>>> rr.__dict__
{'unit_id': 1, 'protocol_id': 0, 'skip_encode': False, 'registers': [18], 'check': 0, 'transaction_id': 0}
>>> rr.__dict__
{'unit_id': 1, 'protocol_id': 0, 'skip_encode': False, 'registers': [18], 'check': 0, 'transaction_id': 0}
>>> rr = client.read_holding_registers(1, 0, unit=1)
DEBUG:pymodbus.transaction:Running transaction 4
DEBUG:pymodbus.transaction:send: 0x1 0x3 0x0 0x1 0x0 0x0 0x14 0xa
DEBUG:pymodbus.transaction:recv: 0x1 0x3 0x0 0x20 0xf0
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 4
>>> rr.__dict__
{'unit_id': 1, 'protocol_id': 0, 'skip_encode': False, 'registers': [], 'check': 0, 'transaction_id': 0}
>>> rr = client.read_holding_registers(1, 2, unit=1)

```

dhoomakethu, 2017-09-09:
Since, I am working on linux based system , I could not test against modbustool ,but I did tried against [this](https://github.com/riptideio/kivy-modbus-simu) and [this](https://github.com/dhoomakethu/modbus_sim_cli) for slave simulation and they did worked without any issue.

thaxy, 2017-09-11:
Seems like pymodbus doesn't wait until all bytes are received. 

serial cable length: 1.5m
holding registers in slave: [1, 23, 56, 456, 0, 0, 0, 0, 0, 0]

slave: modbus slave (windows software)
request: 01 03 00 00 00 0a c5 cd
response (async): 01 03 14 00 6f 00 17 00
response (sync):  01 03 14 00 01 00 17 00 38 01 c8 00 00 00 00 00 00 00 00 00 00 00 00 07 f5

slave: pymodbus slave
request: 0x1 0x3 0x0 0x0 0x0 0xa 0xc5 0xcd
response (async): 0x1 0x3 0x14 0x0 0x1 0x0 0x17 0x0
response (sync):  0x1 0x3 0x14 0x0 0x1 0x0 0x17 0x0 0x38 0x1 0xc8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xf5

ASYNC

```
[08:54:59.210] DEBUG [__main__.__init__:31] Initializing serial port: '/dev/ttyS0', {'stopbits': 1, 'timeout': 5, 'baudrate': 9600, 'parity': 'N', 'bytesize': 8}
modbus_master_1  | [08:54:59.211] DEBUG [pymodbus.client.async.connectionMade:75] Client connected to modbus server
modbus_master_1  | [08:54:59.212] INFO [master.protocol.connectionMade:33] [/dev/ttyS0] Connection made. Reading settings are [{'device_id': 1, 'count': 10, 'offset': 30001, 'start_address': 0}]
modbus_master_1  | [08:55:00.213] DEBUG [master.protocol.fetch_modbus_table:42] [/dev/ttyS0] Start to fetch modbus table(s)
modbus_master_1  | [08:55:00.214] DEBUG [pymodbus.client.async.execute:104] Running transaction 1
modbus_master_1  | [08:55:00.214] DEBUG [pymodbus.client.async.execute:109] send: 0x1 0x3 0x0 0x0 0x0 0xa 0xc5 0xcd
modbus_master_1  | [08:55:00.214] DEBUG [pymodbus.transaction.addTransaction:309] adding transaction 1
modbus_master_1  | [08:55:00.282] DEBUG [pymodbus.client.async.dataReceived:96] recv: 0x1 0x3 0x14 0x0 0x1 0x0 0x17 0x0
modbus_master_1  | [08:55:00.282] DEBUG [pymodbus.factory._helper:238] Factory Response[1]
modbus_master_1  | Unhandled Error
modbus_master_1  | Traceback (most recent call last):
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/log.py", line 103, in callWithLogger
modbus_master_1  |     return callWithContext({"system": lp}, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/log.py", line 86, in callWithContext
modbus_master_1  |     return context.call({ILogContext: newCtx}, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/context.py", line 122, in callWithContext
modbus_master_1  |     return self.currentContext().callWithContext(ctx, func, *args, **kw)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/python/context.py", line 85, in callWithContext
modbus_master_1  |     return func(*args,**kw)
modbus_master_1  | --- <exception caught here> ---
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/posixbase.py", line 597, in _doReadOrWrite
modbus_master_1  |     why = selectable.doRead()
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/_posixserialport.py", line 61, in doRead
modbus_master_1  |     return fdesc.readFromFD(self.fileno(), self.protocol.dataReceived)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/twisted/internet/fdesc.py", line 94, in readFromFD
modbus_master_1  |     callback(output)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/client/async.py", line 97, in dataReceived
modbus_master_1  |     self.framer.processIncomingPacket(data, self._handleResponse)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/transaction.py", line 682, in processIncomingPacket
modbus_master_1  |     self._process(callback, error=True)
modbus_master_1  |   File "/usr/local/lib/python3.5/site-packages/pymodbus/transaction.py", line 711, in _process
modbus_master_1  |     raise InvalidMessageRecievedException(result)
modbus_master_1  | pymodbus.exceptions.InvalidMessageRecievedException: Modbus Error: [Invalid Message] ReadBitResponse(48)
modbus_master_1  |
modbus_master_1  | [08:55:00.283] ERROR [master.protocol.connectionLost:37] [/dev/ttyS0] Connection lost
```

SYNC

```
from pymodbus.client.sync import ModbusSerialClient as ModbusClient

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(method='rtu', port='/dev/ttyS0', baudrate=9600, bytesize=8, parity='N', stopbits=1, timeout=1)
client.connect()
client.read_holding_registers(0, 10, unit=1).__dict__

DEBUG:pymodbus.transaction:Running transaction 16
DEBUG:pymodbus.transaction:send: 0x1 0x3 0x0 0x0 0x0 0xa 0xc5 0xcd
DEBUG:pymodbus.transaction:recv: 0x1 0x3 0x14 0x0 0x1 0x0 0x17 0x0 0x38 0x1 0xc8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xf5
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 16
{'unit_id': 1, 'transaction_id': 0, 'registers': [1, 23, 56, 456, 0, 0, 0, 0, 0, 0], 'check': 0, 'skip_encode': False, 'protocol_id': 0}

client.read_holding_registers(0, 10, unit=1).__dict__
DEBUG:pymodbus.transaction:Running transaction 10
DEBUG:pymodbus.transaction:send: 0x1 0x3 0x0 0x0 0x0 0xa 0xc5 0xcd
WARNING:pymodbus.client.sync:cleanup recv buffer before send: 0x38 0x1 0xc8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xf5
DEBUG:pymodbus.transaction:recv: 0x1 0x3 0x14 0x0 0x1 0x0 0x17 0x0 0x38 0x1 0xc8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x7 0xf5
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 0
DEBUG:pymodbus.transaction:getting transaction 10
{'registers': [1, 23, 56, 456, 0, 0, 0, 0, 0, 0], 'protocol_id': 0, 'skip_encode': False, 'transaction_id': 0, 'unit_id': 1, 'check': 0}

```

dhoomakethu, 2017-09-11:
A lot of enhancements have gone in to sync client which would make it pretty stable and many of those changes (e.g waiting for the complete response, having 3.5 char space between transactions and others ) have not yet gone in to async client. Since I work with sync client + given the fact that I donot have the setup to reproduce the issue, I would appreciate if you can have a look and figure out the issue with async client . You can start at [ModbusClientProtocol](https://github.com/riptideio/pymodbus/blob/master/pymodbus/client/async.py#L54) in pymodbus/async.py. The flow would be like this.
```
ModbusClientProtocol.execute --> Sends request on the wire
ModbusClientProtocol._buildResponse -> add transaction to the que
ModbusClientProtocol.dataReceived -> process Incoming request
ModbusClientProtocol._handleResponse -> deal with the response

```
If you can see the [execute](https://github.com/riptideio/pymodbus/blob/master/pymodbus/transaction.py#L105) method from the transaction.py, there are multiple things we are doing to ensure we are getting the correct/meaningful response. We need to adapt a few to the async client aswell.

Also, Just curious is the behaviour if you increase the read time out to 2 or more from 1sec ?

thaxy, 2017-09-11:
I am currently working with ``timeout=None``

> timeout = None: wait forever / until requested number of bytes are received

From:  http://pyserial.readthedocs.io/en/latest/pyserial_api.html#serial.Serial

I will review the logic of the sync client and make a PR to the ``dev`` branch in the next weeks since I am working on another project. Please have a look at my current changes maybe you can say something to the ``while`` loop and the ``self._process(callback, error=True)`` which makes absolutely no sense to me. You know the code better than me ;)

Is there any other thing according to the async client which is behind the sync client?

dhoomakethu, 2017-09-11:
all those changes are required for the sync client to work fine, It is not supposed to be breaking async client, but looks like it is. I will try to look in to the issue sometime this week. Your current changes in the PR could break the sync client. Thanks for looking .

thaxy, 2017-09-19:
We want to use this lib in our project but since this code is used by the sync client too we are unsure if we can overview the impact on the codebase  Are you currently working on a solution or would it be our turn?

dhoomakethu, 2017-09-19:
If you have a working solution, please raise a PR on dev branch so that somebody would review .  As of now , I do not have a definite time line for an official fix.

thaxy, 2017-09-22:
Just to give you some feedback here: Our project lead decided not to invest any time to fix this bug. My hands are tied, you are on your own. Good luck... :confused:

dhoomakethu, 2018-05-03:
Fixed with v1.5.0

closed 2018-05-03 by dhoomakethu