pymodbus-dev/pymodbus #214: Using pymodbus to issue wrong requests

bracco23, 2017-09-13:
Hello,

I'm working on a specification-compliant implementation of Modbus and I was trying to use this package to develop an automatic test environment to validate my implementation.

I successfully implemented normal requests, but now I wanted to test responses at things like wrong function codes, wrong data formats, wrong checksum and things like that.

Can I integrate that with the module somehow? do I need to fall back to serial communication ( I'm working over RS485 RTU) or I can extend the module to allow me to do that for me? Is there a documentation where I can find the relevant information?

Thanks for the information and for the wonderful package.

dhoomakethu, 2017-09-14:
You can very well use pymodbus for your tasks, You can create your invalid requests by creating classes derived out of `ModbusRequest` and extend `ModbusSerialClient` to include your requests, so that you can do
 ```
from my_client_module import MyClient
my_client = MyClient(...)

res = my_client.invalid_request()
# or
from my_invalid_request_module import invalid_request
ir = invalid_request(unit=<unit_id>)
res = my_client.execute(ir)

```

bracco23, 2017-09-14:
Thanks, to try to understand if it was possible I used pyreverse to create a class diagram to help me navigate the package and I arrived to the same solution, and it worked fine. I also found some interesting things about the library itself, but I will talk about that in a new issue.

closed 2017-09-14 by bracco23