pymodbus-dev/pymodbus #216: Pymodbus synchronoud client receives packet partially

madhusudhanjk, 2017-09-17:
Here is the complete transaction, refer Highlighted B+I for the observation

pymodbus.server.sync: DEBUG: Started thread to serve client at ('172.26.70.41', 43829)
pymodbus.server.sync: DEBUG: Client Connected [172.26.70.41:43829]
pymodbus.server.sync: DEBUG: Started thread to serve client at ('172.26.70.42', 34000)
pymodbus.server.sync: DEBUG: Client Connected [172.26.70.42:34000]
pymodbus.transaction: DEBUG: Running transaction 1
pymodbus.transaction: DEBUG: send: 0x0 0x1 0x0 0x0 0x0 0x6 0x15 0x1 0x0 0x1 0x0 0x1
pymodbus.transaction: DEBUG: Running transaction 2
pymodbus.transaction: DEBUG: send: 0x0 0x2 0x0 0x0 0x0 0x6 0x16 0x1 0x0 0x1 0x0 0x1
pymodbus.server.sync: DEBUG: 0x0 0x1 0x0 0x0 0x0 0x6 0x15 0x1 0x0 0x1 0x0 0x1
pymodbus.transaction: DEBUG: 0x0 0x1 0x0 0x0 0x0 0x6 0x15 0x1 0x0 0x1 0x0 0x1
pymodbus.factory: DEBUG: Factory Request[1]
pymodbus.datastore.context: DEBUG: validate[1] 2:1
pymodbus.datastore.context: DEBUG: getValues[1] 2:1
pymodbus.server.sync: DEBUG: send: 00010000000415010101
pymodbus.server.sync: DEBUG: 0x0 0x2 0x0 0x0 0x0 0x6 0x16 0x1 0x0 0x1 0x0 0x1
pymodbus.transaction: DEBUG: 0x0 0x2 0x0 0x0 0x0 0x6 0x16 0x1 0x0 0x1 0x0 0x1
pymodbus.factory: DEBUG: Factory Request[1]
pymodbus.datastore.context: DEBUG: validate[1] 2:1
pymodbus.datastore.context: DEBUG: getValues[1] 2:1
pymodbus.server.sync: DEBUG: send: 00020000000416010101
pymodbus.transaction: DEBUG: recv: 0x0 0x1 0x0 0x0 0x0 0x4 0x15 0x1 0x1 0x1
pymodbus.transaction: DEBUG: 0x0 0x1 0x0 0x0 0x0 0x4 0x15 0x1 0x1 0x1
pymodbus.factory: DEBUG: Factory Response[1]
pymodbus.transaction: DEBUG: adding transaction 1
pymodbus.transaction: DEBUG: getting transaction 1
pymodbus.transaction: DEBUG: Running transaction 3
pymodbus.transaction: DEBUG: send: 0x0 0x3 0x0 0x0 0x0 0x6 0x15 0x5 0x0 0x0 0xff 0x0
pymodbus.transaction: DEBUG: recv: 0x0 0x2 0x0 0x0 0x0 0x4 0x16 0x1 0x1 0x1
pymodbus.transaction: DEBUG: 0x0 0x2 0x0 0x0 0x0 0x4 0x16 0x1 0x1 0x1
pymodbus.factory: DEBUG: Factory Response[1]
pymodbus.transaction: DEBUG: adding transaction 2
pymodbus.transaction: DEBUG: getting transaction 3
pymodbus.transaction: DEBUG: getting transaction 0
pymodbus.transaction: DEBUG: Running transaction 4
pymodbus.transaction: DEBUG: send: 0x0 0x4 0x0 0x0 0x0 0x6 0x15 0x1 0x0 0x0 0x0 0x1
pymodbus.server.sync: DEBUG: 0x0 0x3 0x0 0x0 0x0 0x6 0x15 0x5 0x0 0x0 0xff 0x0
pymodbus.transaction: DEBUG: 0x0 0x3 0x0 0x0 0x0 0x6 0x15 0x5 0x0 0x0 0xff 0x0
pymodbus.factory: DEBUG: Factory Request[5]
pymodbus.datastore.context: DEBUG: validate[5] 1:1
pymodbus.datastore.context: DEBUG: setValues[5] 1:1
pymodbus.datastore.context: DEBUG: getValues[5] 1:1
pymodbus.server.sync: DEBUG: send: 00030000000615050000ff00
_**pymodbus.transaction: DEBUG: recv: 0x0 0x3 0x0 0x0 0x0 0x6 0x15 0x5 0x0 0x0**_
pymodbus.transaction: DEBUG: 0x0 0x3 0x0 0x0 0x0 0x6 0x15 0x5 0x0 0x0
pymodbus.transaction: DEBUG: getting transaction 2
_**pymodbus.transaction: DEBUG: recv: 0xff 0x0**_
pymodbus.transaction: DEBUG: 0xff 0x0
pymodbus.transaction: DEBUG: Running transaction 5
pymodbus.transaction: DEBUG: send: 0x0 0x5 0x0 0x0 0x0 0x6 0x16 0x5 0x0 0x0 0xff 0x0

dhoomakethu, 2017-09-18:
what version of pymodbus are you using ? Also do you mean to say the server is receiving incomplete request ?

madhusudhanjk, 2017-09-18:
pymodbus version : 1.3.2. Its's client who is receiving incomplete

dhoomakethu, 2017-09-18:
Let me clarify it again, A client is the one which sends requests and a slave/server is something which responds to those requests, The logs you have provided talks about synchronous server, If I understand correctly , you have three `ModbusTcpClient` instances which are querying against one `ModbusTcpServer`. And The `ModbusTcpServer` sometimes receives only partial requests. Please correct me if my understanding is incorrect.

madhusudhanjk, 2017-09-18:
I start three ModbusTcpServer  conencting to a remote gateway , this in log corresponds to
pymodbus.server.sync: DEBUG: Started thread to serve client at ('172.26.70.41', 43829)
pymodbus.server.sync: DEBUG: Client Connected [172.26.70.41:43829]
pymodbus.server.sync: DEBUG: Started thread to serve client at ('172.26.70.42', 34000)
pymodbus.server.sync: DEBUG: Client Connected [172.26.70.42:34000] 
Then i initiate request from one ModbusTcpClient directing to one of three ModbusTcpServer.
Since both ModbusTcpClient and ModbusTcpServer is of synchronous type, i suspected that either of them should not accept less or more bytes as defined in MBAP header length field.

dhoomakethu, 2017-09-18:
Could you provide your server and client code?

dhoomakethu, 2017-10-03:
similart to #217 , will be tracked there

closed 2017-10-03 by dhoomakethu