pymodbus-dev/pymodbus #217: Issue with Modbus Error: [Input/Output] No Response received from the remote unit

pawelmarkowski, 2017-09-20:
Hello,
I am checking out pymodbus.__version__ '1.3.2'  library and attached code does not work, once code copy in pymodbus3 works fine.

What do I wrong?

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

from pymodbus.client.sync import ModbusTcpClient
from pymodbus.payload import BinaryPayloadDecoder
from pymodbus.constants import Endian
#
#
#
client = ModbusTcpClient('192.168.174.82', port = 502)
registers_set = 1
# offset, adres początkowy od której ramki/rejestru zaczynam czytać
address = 7505
# ile rejestrów przeczytać (ramki)
count = registers_set * 2
# adres modbus'owy urzadzenia
unit = 1
#
result = client.read_holding_registers(address, count, unit = unit)
print(result)
print(result.registers)
#
for i in range(registers_set):
    j = i * 2
    k = (i + 1) * 2
    print (j, k, "register range")
    decoder = BinaryPayloadDecoder.from_registers(result.registers[j:k], endian = Endian.Big)
    print("{0:.2f}".format(decoder.decode_32bit_float()))
client.close()

DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x1d 0x51 0x0 0x2
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x45 0xae 0x6f 0x8c
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x45 0xae 0x6f 0x8c
DEBUG:pymodbus.transaction:getting transaction 1
Traceback (most recent call last):
Modbus Error: [Input/Output] No Response received from the remote unit
  File "/home/pawel/workspace/scripts/python/modbus/lumel.py", line 23, in <module>
    print(result.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

______________________________
import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

from pymodbus3.client.sync import ModbusTcpClient
from pymodbus3.payload import BinaryPayloadDecoder
from pymodbus3.constants import Endian
#
#
#
client = ModbusTcpClient('192.168.174.82', port = "502")
registers_set = 1
# adres 32BIT dla sterownika LUMEL
address = 7505
# ile rejestrów przeczytać (ramki)
count = registers_set * 2
# adres modbus'owy urzadzenia
unit = 1
#
result = client.read_holding_registers(address, count, unit = unit)
print(result)
print(result.registers)
#
for i in range(registers_set):
    j = i * 2
    k = (i + 1) * 2
    print (j, k, "register range")
    decoder = BinaryPayloadDecoder.from_registers(result.registers[j:k], endian = Endian.Big)
    print("{0:.2f}".format(decoder.decode_32bit_float()))
client.close()

DEBUG:pymodbus3.transaction:Running transaction 1
DEBUG:pymodbus3.factory:Factory Response[3]
DEBUG:pymodbus3.transaction:adding transaction 1
DEBUG:pymodbus3.transaction:getting transaction 1
ReadRegisterResponse (4)
[17838, 28486, 16777, 24065]
0 2 register range
5581.91

Thanks for help.

dhoomakethu, 2017-09-21:
@pawelmarkowski , would you mind sharing your findings? Was it a problem with your setup?

pawelmarkowski, 2017-09-21:
no it was not a problem with setup. It still does not work ... interesting thing is that the device has address = 7505 with 32bit registry ... I found other 16bit address that works and I can read a data so I closed the ticket. It is strange that a pymodbus3 works fine.

dhoomakethu, 2017-09-21:
Are you working with a real modbus device ? If yes can you please share the details/specs?

pawelmarkowski, 2017-09-21:
http://www.lumel.com.pl/download/Z2Z4L2x1bWVsL2VuL2RlZmF1bHRfbXVsdGlsaXN0YV9wbGlrb3cudjAvNTIx/n4309a.pdf

dhoomakethu, 2017-09-22:
@pawelmarkowski the device talks only about serial interface but your code is for Modbus TCP, Is there a RTU-TCP gateway you are talking to ?

pawelmarkowski, 2017-09-22:
Gateway, but I will confirm it on Monday.

pawelmarkowski, 2017-10-02:
Hello, sorry for 1 week delay :). I updated above link - device has modbus TCP interface.

lixiangabcd08, 2017-10-03:
I have the same problem with version '1.3.2'. The length of received package through sync TCP client is not correct.

Following is the logging information:

DEBUG:root:Read a holding register
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x7e 0x3 0x9c 0x42 0x0 0x1
**DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x3 0x2 0x0**
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x3 0x2 0x0
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit
error
'ModbusIOException' object has no attribute 'registers'

DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x0 0x2 0x0 0x0 0x0 0x6 0x7e 0x3 0x9c 0x42 0x0 0x1
**DEBUG:pymodbus.transaction:recv: 0x1**
DEBUG:pymodbus.transaction:0x1
error
'len'

DEBUG:pymodbus.transaction:Running transaction 3
DEBUG:pymodbus.transaction:send: 0x0 0x3 0x0 0x0 0x0 0x6 0x7e 0x3 0x9c 0x42 0x0 0x1
**DEBUG:pymodbus.transaction:recv: 0x0 0x2 0x0 0x0 0x0 0x5 0x7e 0x3 0x2 0x0 0x1**
DEBUG:pymodbus.transaction:0x0 0x2 0x0 0x0 0x0 0x5 0x7e 0x3 0x2 0x0 0x1
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 2
DEBUG:pymodbus.transaction:getting transaction 3
DEBUG:pymodbus.transaction:getting transaction 0
None
error
'NoneType' object has no attribute 'registers'

The recv data in the first transaction was truncated into 2 transactions and caused all the following transaction to be incorrect even though the length is actually correct.

In addition, I managed to get the correct data by using pymodbus3 package instead.

dhoomakethu, 2017-10-03:
@lixiangabcd08 In your case I see you are trying to read the absolute address (40002), Could you just pass the index (0x2) as address and see if you can get the expected result?

dhoomakethu, 2017-10-03:
@pawelmarkowski could you give a try with this branch and let me know if it works ?

https://github.com/riptideio/pymodbus/tree/217-fix-tcp-client

lixiangabcd08, 2017-10-03:
@dhoomakethu thank you very much for your suggestion, but changing the base address doesn't work. Same problem persists. As I have said, there are some problems regarding the number of bytes in the first 2 received packages. From 3rd package onwards, the received package length becomes ok. But since it's synchronous, it doesn't work out very nicely.

lixiangabcd08, 2017-10-03:
@dhoomakethu By the way, the code works correctly when I initiate the ModbusTcpClient with library from pymodbus3 and use the read_input_registers function from pymodbus. So I believe there is some bug inside initiation of client code. Hope it helps debugging.

dhoomakethu, 2017-10-03:
@lixiangabcd08 please see the comment above , please try the pymodbus from that branch or use the tar file to install pymodbus.

> @pawelmarkowski could you give a try with this branch and let me know if it works ?

https://github.com/riptideio/pymodbus/tree/217-fix-tcp-client or with this
pymodbus-1.4.0rc1.tar.gz

pawelmarkowski, 2017-10-03:
Still the same error.

1st transaction 6010 address: 16bit registry 2 frames
address = 6010
count = 2
result = self.client.read_holding_registers(address, count, unit = self.unit)

DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x17 0x7a 0x0 0x2
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xf9 0x84 0x45 0xc0
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xf9 0x84 0x45 0xc0
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 1
DEBUG:pymodbus.transaction:getting transaction 1
ReadRegisterResponse (2)
[63876, 17856]
6175.19

2nd transaction 7505 address: 32bit registry (1 frame)
address = 7505
count = 1
result = self.client.read_holding_registers(address, count, unit = self.unit)

DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x1d 0x51 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x45 0xc0
DEBUG:pymodbus.transaction:0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x45 0xc0
DEBUG:pymodbus.transaction:getting transaction 2
Modbus Error: [Input/Output] No Response received from the remote unit

pawelmarkowski, 2017-10-03:
@dhoomakethu 
It works in your version. Only what I need is to change decoding because in this registry data is stored in reverse order. 

DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x17 0x7a 0x0 0x2
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xfc 0xbb 0x45 0xc0
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0xfc 0xbb 0x45 0xc0
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 1
DEBUG:pymodbus.transaction:getting transaction 1
ReadRegisterResponse (2)
[64699, 17856]
6175.59
DEBUG:pymodbus.transaction:Running transaction 2
DEBUG:pymodbus.transaction:send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x1d 0x51 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x45 0xc0 0xfc 0xbb
DEBUG:pymodbus.transaction:0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x3 0x4 0x45 0xc0 0xfc 0xbb
DEBUG:pymodbus.factory:Factory Response[3]
DEBUG:pymodbus.transaction:adding transaction 2
DEBUG:pymodbus.transaction:getting transaction 2
ReadRegisterResponse (2)
[17856, 64699]
-7778993684926939764378185074730860544.00

dhoomakethu, 2017-10-03:
@pawelmarkowski thanks for confirming, I will wait on @lixiangabcd08 to confirm so that I can push this to master and close the issue

lixiangabcd08, 2017-10-03:
@dhoomakethu I am very sorry that I am unable to verify it until next week:(

dhoomakethu, 2017-10-03:
will be tracked with #221

closed 2017-10-03 by dhoomakethu

lixiangabcd08, 2017-10-12:
@dhoomakethu Sorry for the late reply, I have tried your version. It does not solve the problem. However the error is a bit different now.
Instead of receiving 10 packages in the first transaction, 1 package in the second transaction and 11 (correct number) packages in the rest of transactions, now I receive 10 package in every transaction.

Debugging infomation as followed:
169.254.82.110:502
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x7e 0x4 0x9d 0x7 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x4 0x2 0x0
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x4 0x2 0x0
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit

169.254.82.110:502
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x7e 0x4 0x9d 0x7 0x0 0x1
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x4 0x2 0x0
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x5 0x7e 0x4 0x2 0x0
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit

And this TCP connection works fine with pymodbus3.

dhoomakethu, 2017-10-12:
@lixiangabcd08  , Are you sure you are using the correct pymodbus version ? The behaviour you are observing is not expected , with the changes ModbusTCP client would now will receive up to 1024 bytes of data in one read. Please uninstall and install pymodbus again (the version with patch in this comment thread above ) and confirm. 
Thanks

lixiangabcd08, 2017-10-12:
@dhoomakethu Thank you for your reply! I believe I am using the correct pymodbus library since I have uninstall and install for a few time. The received modbus data from execution using the offical version and the version in this thread are different. And I only observe this behavior for one type of our TCP device while the other TCP devices work fine with pymodbus.

dhoomakethu, 2017-10-12:
@lixiangabcd08 Thats interesting to know, could you please share the details of the device ? Also Are you getting correct response if using any other modbus packages (for e.g Modbus-tk ?)

lixiangabcd08, 2017-10-13:
@dhoomakethu The problematic device is SMA Solar PV inverter (STP 5000TL-20) with SunSpec Modbus Interface. For the same manufacturer, SMA cluster controller works fine with pymodbus. I have not tried Modbus-tk (I only have access to the inverter once a week), but QModMaster and pymodbus3 give corrects response. Appreciate all the effort you put in.

dhoomakethu, 2017-10-13:
Thanks for the update, good to know it works with pymodbus3, I will have a look.

Amritsoul, 2017-10-24:
I have tried using pymodbus3 also, but i am getting the same error

pe2a, 2018-12-05:
I have added time.sleep function ant it has fixed.