pymodbus-dev/pymodbus #220: Code hangs after akward? factory response [135]

zikann, 2017-09-28:
Hi,
this is my setup:
pymodbus: 1.3.2
Modbus RTU over RS485
3 devices on network 
continously doing what's below: 
4 registers queried from 1. device in 4 requests
2 registers queried from 2. device in 2 requests
2 registers queried from 3. device in 2 requests

I belive one device has some problems with communication( the oldest one) and when this happens the whole comunication stops until application restart.

In the logs it reveals itself in the following way:

everything OK in 1 device:
```
2017-09-26 11:42:25,165 DEBUG:Running transaction 21587
2017-09-26 11:42:25,165 DEBUG:send: 0x1 0x3 0x5 0xdb 0x0 0x2 0xb4 0xfc
2017-09-26 11:42:25,165 DEBUG:will sleep to wait for 3.5 char
2017-09-26 11:42:25,181 DEBUG:recv: 0x1 0x3 0x4 0x3f 0x7d 0xfb 0x3d 0xe4 0xde
2017-09-26 11:42:25,181 DEBUG:Factory Response[3]
2017-09-26 11:42:25,181 DEBUG:adding transaction 0
2017-09-26 11:42:25,181 DEBUG:getting transaction 21587
```
no response from second device:
```
2017-09-26 11:42:25,197 DEBUG:send: 0x3f 0x3 0x7 0x8 0x0 0x2 0x40 0x63
2017-09-26 11:42:25,805 DEBUG:recv: 
2017-09-26 11:42:25,805 DEBUG:getting transaction 21588
2017-09-26 11:42:25,805 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-09-26 11:42:25,805 ERROR:Modbus exception received:Modbus Error: [Input/Output] No Response received from the remote unit
2017-09-26 11:42:25,805 DEBUG:Running transaction 21589
2017-09-26 11:42:25,805 DEBUG:send: 0x3f 0x3 0x13 0xa0 0x0 0x1 0x84 0x72
2017-09-26 11:42:25,805 DEBUG:will sleep to wait for 3.5 char
2017-09-26 11:42:26,430 DEBUG:recv: 
2017-09-26 11:42:26,430 DEBUG:getting transaction 21589
2017-09-26 11:42:26,430 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-09-26 11:42:26,430 ERROR:Modbus exception received:Modbus Error: [Input/Output] No Response received from the remote unit
```
no response from third device:
```2017-09-26 11:42:26,430 DEBUG:Running transaction 21590
2017-09-26 11:42:26,430 DEBUG:send: 0x40 0x3 0x7 0x8 0x0 0x2 0x4b 0xac
2017-09-26 11:42:26,430 DEBUG:will sleep to wait for 3.5 char
2017-09-26 11:42:27,053 DEBUG:recv: 
2017-09-26 11:42:27,053 DEBUG:getting transaction 21590
2017-09-26 11:42:27,053 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-09-26 11:42:27,053 ERROR:Modbus exception received:Modbus Error: [Input/Output] No Response received from the remote unit
```
final log entry until restart:
```
2017-09-26 11:42:27,053 DEBUG:Running transaction 21591
2017-09-26 11:42:27,053 DEBUG:send: 0x40 0x3 0x13 0xa0 0x0 0x1 0x8f 0xbd
2017-09-26 11:42:27,053 DEBUG:will sleep to wait for 3.5 char
2017-09-26 11:42:27,115 DEBUG:Starting new HTTPS connection (1): things.ubidots.com
2017-09-26 11:42:27,677 DEBUG:recv: 0x87
2017-09-26 11:42:27,677 DEBUG:Factory Response[135]
```
Notice the last line with: **DEBUG:Factory Response[135]**
I think that the received value 0x87 is some kind of glitch in communication and it's cousing the issue but is should not stop the communication.

Any ideas?

Thanks for help

dhoomakethu, 2017-09-29:
Ok, Questions.

1. What client are you using (sync/async) ?
2. How are you doing the continuous query , Could you share your client code?
3. Are the three devices connected on the same serial port or on different ports ?

Thanks

zikann, 2017-09-29:
Ad1. Sync client running in separate thread.
Ad2. Simple loop over dictionary having the settings needed to query devices( device id, adresses etc... ).
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

import logging
import threading
import time

from pymodbus.constants import Endian
from pymodbus.exceptions import ConnectionException
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.payload import BinaryPayloadDecoder

from config import Configuration
from strain_gauge.strain_gauge import Measurement

def run_in_thread(devices):
    # SERIAL CLIENT
    client = ModbusClient(method='rtu',
                          port=Configuration.config_dict['Bus settings']['port'],
                          baudrate=Configuration.config_dict['Bus settings']['baudrate'],
                          parity=Configuration.config_dict['Bus settings']['parity'],
                          timeout=Configuration.config_dict['Bus settings']['timeout'])
    while True:
        for device in devices.itervalues():
            if not device.enabled:
                continue
            for data_name, data_desc in device.data.iteritems():
                try:
                    rr = client.read_holding_registers(data_desc.address, data_desc.size, unit=device.modbus_dev_id)
                except ConnectionException:
                    logging.error('Cannot connect to client,port:{}.address:{}, device_id:{}'
                              .format(client.port,
                                      data_desc.address,
                                      device.modbus_dev_id
                                      )
                              )
                    device.set_error('cannot_connect')
                    break
                if rr is not None:
                    # TODO Ogarnąc exception response
                    try:
                        print rr
                        decoder = BinaryPayloadDecoder.fromRegisters(rr.registers, endian=Endian.Big)
                        print rr.registers

                        if data_desc.size == 2:
                            decoded = decoder.decode_32bit_float()
                        elif data_desc.size == 1:
                            decoded = decoder.decode_16bit_int()
                        else:
                            logging.error('Bad data size specified for {}, on {}'.format(data_name, data_desc.address))
                            continue
                        print decoded
                        data_desc.value = Measurement(value_converted, decoded, time.time())

                    except Exception as e:
                        print e
                        logging.error(str(e))
                        print 'Modbus exception received:{}'.format(rr)
                        logging.error('Modbus exception received:{}'.format(rr))
                        device.set_error('exception_response')
                else:
                    logging.error('No response from port:{}.address:{}, device_id:{}'
                              .format(client.port,
                                      data_desc.address,
                                      device.modbus_dev_id
                                      )
                              )
                    device.set_error('no_response')

        time.sleep(Configuration.config_dict['General']['modbus_query_interval'])
    client.close()

def run_against_devices(devices):
    t = threading.Thread(target=run_in_thread, args=(devices,))
    t.daemon = True
    t.start()

```
Ad3. Same serial port over RS485 network.

dhoomakethu, 2017-09-29:
Thanks, If it is possible, Could you please connect one device at a time and see if you can get response with out errors for all three devices ? From your logs , device 2 and 3 are not responding. This happens even when the connections are not proper.

zikann, 2017-09-29:
This is only a small piece od log showing the time just before the hang.
The statystics from the day it happened are folowing:

device 1:
requests per day = 183682
propper responses per day= 182962
success ratio = 99,6

device 2:
requests per day = 91930
propper responses per day= 90952
% = 98,9%

device 3:
requests per day = 91926
proper responses per day= 87857
success ratio = 95%

After that day I slowed down the serial communication from 57600 to 19200 and the stats improved to:
device 1: 100%
device 2: 100%
device3: 97%

Additionaly I think i should mention that the "hanging" happens one time i 2 days in average.

dhoomakethu, 2017-10-03:
Thanks for the data , What is the load on the CPU when the reads start failing ? Would you mind share the configuration of the device on which the client is running ?

zikann, 2017-10-03:
I dont't remember the exact value but I remember it was "normal"  not 100%
or something like that. The client is a  normal PC 2 cores Windows 10 etc...
Its running over a moxa usb-rs485 converter in half duplex 2 wire
configuration.

zikann, 2017-10-15:
Hi,
I've got new, perhaps interesting logs:
```
2017-10-14 11:13:59,927 DEBUG:send: 0x40 0x3 0x7 0x8 0x0 0x2 0x4b 0xac
2017-10-14 11:13:59,927 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:00,552 DEBUG:recv: 
2017-10-14 11:14:00,552 DEBUG:Factory Response[64]
2017-10-14 11:14:00,552 ERROR:Unable to decode response Modbus Error: Unknown response 64
2017-10-14 11:14:00,568 DEBUG:Transaction failed. (Modbus Error: [Input/Output] Unable to decode request) 
2017-10-14 11:14:00,568 DEBUG:getting transaction 40510
2017-10-14 11:14:00,568 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-10-14 11:14:00,568 ERROR:Modbus exception received:Modbus Error: [Input/Output] Modbus Error: [Input/Output] Unable to decode request
2017-10-14 11:14:00,661 DEBUG:Running transaction 40511
2017-10-14 11:14:00,661 DEBUG:send: 0x40 0x3 0x13 0xa0 0x0 0x1 0x8f 0xbd
2017-10-14 11:14:00,661 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:01,286 DEBUG:recv: 
2017-10-14 11:14:01,286 DEBUG:getting transaction 40511
2017-10-14 11:14:01,286 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-10-14 11:14:01,286 ERROR:Modbus exception received:Modbus Error: [Input/Output] No Response received from the remote unit
2017-10-14 11:14:01,286 DEBUG:Running transaction 40512
2017-10-14 11:14:01,286 DEBUG:send: 0x1 0x3 0x5 0xe1 0x0 0x2 0x94 0xf1
2017-10-14 11:14:01,286 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:01,301 DEBUG:recv: 0x1 0x3 0x4 0x3f 0x86 0xdc 0x69 0x8f 0x20
2017-10-14 11:14:01,301 DEBUG:Factory Response[3]
2017-10-14 11:14:01,301 DEBUG:adding transaction 0
2017-10-14 11:14:01,301 DEBUG:getting transaction 40512
2017-10-14 11:14:02,331 DEBUG:Running transaction 40513
2017-10-14 11:14:02,331 DEBUG:send: 0x1 0x3 0x5 0xde 0x0 0x2 0xa4 0xfd
2017-10-14 11:14:02,331 DEBUG:recv: 0x1 0x3 0x4 0x3f 0x86 0x1 0xf6 0x96 0x18
2017-10-14 11:14:02,331 DEBUG:Factory Response[3]
2017-10-14 11:14:02,345 DEBUG:adding transaction 0
2017-10-14 11:14:02,345 DEBUG:getting transaction 40513
2017-10-14 11:14:02,345 DEBUG:Running transaction 40514
2017-10-14 11:14:02,345 DEBUG:send: 0x1 0x3 0x5 0xe4 0x0 0x2 0x84 0xf0
2017-10-14 11:14:02,752 DEBUG:recv: 0x1 0x3 0x4 0x3f 0x83 0xe7
2017-10-14 11:14:02,752 DEBUG:Factory Response[1]
2017-10-14 11:14:02,767 DEBUG:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(32)) 
2017-10-14 11:14:02,861 DEBUG:send: 0x1 0x3 0x5 0xe4 0x0 0x2 0x84 0xf0
2017-10-14 11:14:02,861 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:03,484 DEBUG:recv: 
2017-10-14 11:14:03,484 DEBUG:Factory Response[1]
2017-10-14 11:14:03,500 DEBUG:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(32)) 
2017-10-14 11:14:03,594 DEBUG:send: 0x1 0x3 0x5 0xe4 0x0 0x2 0x84 0xf0
2017-10-14 11:14:03,594 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:04,217 DEBUG:recv: 
2017-10-14 11:14:04,217 DEBUG:Factory Response[1]
2017-10-14 11:14:04,233 DEBUG:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(32)) 
2017-10-14 11:14:04,233 DEBUG:getting transaction 40514
2017-10-14 11:14:04,233 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-10-14 11:14:04,233 ERROR:Modbus exception received:Modbus Error: [Input/Output] Modbus Error: [Invalid Message] ReadBitResponse(32)
2017-10-14 11:14:04,328 DEBUG:Running transaction 40515
2017-10-14 11:14:04,328 DEBUG:send: 0x1 0x3 0x5 0xdb 0x0 0x2 0xb4 0xfc
2017-10-14 11:14:04,328 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:04,951 DEBUG:recv: 
2017-10-14 11:14:04,951 DEBUG:getting transaction 40515
2017-10-14 11:14:04,951 ERROR:'ModbusIOException' object has no attribute 'registers'
2017-10-14 11:14:04,951 ERROR:Modbus exception received:Modbus Error: [Input/Output] No Response received from the remote unit
2017-10-14 11:14:04,951 DEBUG:Running transaction 40516
2017-10-14 11:14:04,951 DEBUG:send: 0x3f 0x3 0x7 0x8 0x0 0x2 0x40 0x63
2017-10-14 11:14:04,951 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:05,778 DEBUG:recv: 0xf8
2017-10-14 11:14:05,778 DEBUG:Factory Response[248]
```
"2017-10-14 11:14:05,778 DEBUG:Factory Response[248]" is the last entry in log.

dhoomakethu, 2017-10-16:
@zikann Could you add the following flags to your client `retry_on_empty=True, retries=3 or more`

```
# * retries - Specify how many retries to allow per transaction (default = 3)
# * retry_on_empty - Is an empty response a retry (default = False)
#
# Here is an example of using these options::
#
#    client = ModbusClient(...., retries=3, retry_on_empty=True)
```

zikann, 2017-10-22:
Flags added, unfortunately the behaviour is still the same.
For now I don't have fresh logs to show.

After further investigating the previous logs I find it strange that things like this happen:
```
2017-10-14 11:13:59,927 DEBUG:send: 0x40 0x3 0x7 0x8 0x0 0x2 0x4b 0xac
2017-10-14 11:13:59,927 DEBUG:will sleep to wait for 3.5 char
2017-10-14 11:14:00,552 DEBUG:recv: 
2017-10-14 11:14:00,552 DEBUG:Factory Response[64]
```
How is it possible that it says that factory response is 64 when the previous line shows that there is no response?

```
2017-10-14 11:14:02,345 DEBUG:send: 0x1 0x3 0x5 0xe4 0x0 0x2 0x84 0xf0
2017-10-14 11:14:02,752 DEBUG:recv: 0x1 0x3 0x4 0x3f 0x83 0xe7
2017-10-14 11:14:02,752 DEBUG:Factory Response[1]
2017-10-14 11:14:02,767 DEBUG:Transaction failed. (Modbus Error: [Invalid Message] ReadBitResponse(32)) 
```
How is it possible that it says that factory response is 1 when the previous line shows that the factory response is "3".

zikann, 2017-11-04:
> How is it possible that it says that factory response is 64 when the previous line shows that there is no response?

I've found in code that when the response is not valid the frame is decoded as "raw":
https://github.com/riptideio/pymodbus/blob/62b1350b721ce01bec35de7a828fbc92a3084744/pymodbus/transaction.py#L465-L466
so the mentioned log means that instead of a valid frame a not valid frame starting with 0x01 was received.

But this doesn't explain the original issue with "hanging" of the thread that is responsible for pooling the slave devices.

Status for now:
After changing the timeout to 0.5s, connecting all devices on the bus to a common UPS and adding termination resistors on the bus, the errors in transmissin are gone as the effect of hanging thread, but I belive that the cause for hanging is still there somewhere in the code...

dhoomakethu, 2017-11-05:
@zikann thanks for the follow-up, I am halfway debugging this issue but I am having tough time to reproduce this. I will keep this open for the time being

lromor, 2017-12-05:
Hello,
same problem here, no response and hangs.

-l

dhoomakethu, 2017-12-05:
@iromor, could you please provide some information regarding your setup. Any logs before and after hang is highly appreciated.

janiversen, 2022-04-01:
Is this issue still open ? I tested with a usb-rs485 adapter and 3 devices using version 2.5.3, tested both sync and async, all without problems.