pymodbus-dev/pymodbus #221: Re do/enhances calculating expected response for various protocols

dhoomakethu, 2017-10-03:
ModbusTCP seems to be having a problem reading everything in the response, refer #217, #200 , #202

izak, 2017-12-01:
I seem to be having a problem related to this issue. If not helpful, I apologise. I'm trying to communicate with an SMA solar inverter using the Sunspec protocol. The response is supposed to be `0x53 0x75 0x6e 0x53` (literally `SunS`), but it seems the last byte is chopped off and then I get the same `No Response` error.

```
import logging
from pymodbus.client.sync import ModbusTcpClient as ModbusClient

def main(host, port, unit):
    logging.basicConfig()
    logging.getLogger('pymodbus').setLevel(logging.DEBUG)
    with ModbusClient(host, port=port, timeout=20) as client:
    print client.read_holding_registers(40000, 2, unit=unit)

if __name__ == "__main__":
    main('192.168.0.131', 502, 126)
```

And the output:

```
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x7e 0x3 0x9c 0x40 0x0 0x2
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x7e 0x3 0x4 0x53 0x75 0x6e
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x7 0x7e 0x3 0x4 0x53 0x75 0x6e
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit
```

dhoomakethu, 2017-12-01:
@izak, could you please give with the code on `dev` branch ? It is supposed to be having a fix with respect to modbus TCP.

izak, 2017-12-01:
This happens with the code on the dev branch too. That was the first thing I tried, because I noticed some work had gone into this already.

The trouble seems to be that `socket.recv` receives a byte short, which is perfectly legit, that's how the unix syscall works. But because we're running a SocketFramer, it does something else I don't quite yet comprehend. I'm only about 2 hours into the code so I don't yet fully understand how it is designed.

dhoomakethu, 2017-12-01:
Ok, if that is the case could you please try the read transaction twice and see if the missing byte is cleared before the second read transaction ? Please share the logs here.

izak, 2017-12-01:
By literally just calling  `self._recv` a second time in transaction.py and appending the result, it does indeed work.

```
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x0 0x1 0x0 0x0 0x0 0x6 0x7e 0x3 0x9c 0x40 0x0 0x2
DEBUG:pymodbus.transaction:recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x7e 0x3 0x4 0x53 0x75 0x6e 0x53
DEBUG:pymodbus.transaction:0x0 0x1 0x0 0x0 0x0 0x7 0x7e 0x3 0x4 0x53 0x75 0x6e 0x53
```

dhoomakethu, 2017-12-01:
Just in case useful, I found an example for sunspec client here https://github.com/riptideio/pymodbus/blob/29d236edaccd48ddc35b5503298d157be2428cd1/examples/contrib/sunspec_client.py, the example is very old and might be buggy, give it a try.

izak, 2017-12-01:
My use case was really to poke around a bit to see what I have to work with. I'm actually working on a project with code written in C, that is, I'm not actually going to use pymodbus in this project (but we are using it in [this one](https://github.com/izak/Bornay-venus-driver), where it works fine). I since came up with a different way to poke around: Just sniff the traffic from the wire with wireshark. So I'm probably good to go.

dhoomakethu, 2017-12-01:
Cool, I will keep this case in mind while fixing this issue, thanks for bringing this up.

dhoomakethu, 2017-12-01:
@izak If you don't mind could you please give a try on [this branch](https://github.com/riptideio/pymodbus/tree/%23221-Tcp-read) and check if it works fine with your setup.

izak, 2017-12-01:
Indeed that seems to solve it. Thanks!

dhoomakethu, 2017-12-01:
Thanks for confirming

commit 689708a6 (2018-01-26): Fix #221 timing enhancements, #188 workarounds
commit 9d2c864b (2018-04-26): Pymodbus 1.5.0 (#285)

*  Modbus sync client timing enhancements #221
 Fix TCP server #256, #260

* Fix #221 timing enhancements, #188 workarounds

* 1. #284 Async servers - Option to start reactor outside Start<server>Server function
2. #283 Fix BinaryPayloadDecoder/Builder issues when using with pymodbus server
3. #278 Fix issue with sync/async servers failing to handle requests with transaction id > 255
4. #221 Move timing and transcational logic to framers for sync clients
5. #221 More debug logs for sync clients
6. Misc updates with examples and minor enhancements

* 1. #277 MEI message r [... cut]
closed 2018-05-03 by dhoomakethu