pymodbus-dev/pymodbus #228: Gateway example

mariogirasole, 2017-10-14:
Dear all, I'm trying to do a simple Modbus TCP gateway that read data from up to 5 remote devices and publish these information at modbus tables on the server.
I not seeing dificults to read and write messages to remote devices using the modbus functions created.
For this project, my customer need to read the data on the server (my gateway) on the Input Status and Input Register tables (FC2 and FC4).
How can I write data for these tables using my python script since these areas are read only?
The setValues methods from the store and context objects not allow this operation using fx=4 (input register). 
I also tryed to use the method setValues from the remote store, created by (RemoteSlaveContext) but without success.
I'm using the version 2.7.13 from Python.
Thanks in advance.

 and My dificult is I

dhoomakethu, 2017-10-14:
You can not read and write to Input registers or discrete inputs as per the modbus spec, they are read only. You can use Holding Registers and Coils  if you would like to do R/W operations. But your problem statement is kind of confusing. 

1. Are the 5 slave devices real or simulated ,what mode are they running (RTU/TCP/ASCII) ?
2. How is your customer intend to read the data on the server you are developing (again with Modbus transactions or any other mode )?
3.Are you trying to map every register to Inputregister and evey input (coil,discrete inputs) on the remote slaves to discrete inputs on your server?

If possible, please share the code snippet of your remote store to understand what you are trying to do.

mariogirasole, 2017-10-14:
Hello, I'll decribe an example from my need:
My 5 slaves devices are real in Modbus TCP mode.
My slaves devices providing holding registers information (range 30101 to
30147).
Almost all information are available in bits at the holding registers at
source maps (slaves), for this reason I'm using modbus FC3 to read these
information.
Once read the information, I need to put them on my server context, but
according to the map defined by my customer, see examples below:

Example #1 (Status and alarms)
Slave side:
device: 1
register/bit: 30101 bit 0 (Holding Register)
server side:
device:1
register/bit: 10001 (Input Status)

Note: In this case, the data will be read only on the server side.

Example #2 (Comands)
server side:
device:1
register/bit: 00001 (Coil)
Slave side:
device: 1
register/bit: 30131 bit 0

Note: In this case, the data will be write only on the server side.

See attached my python scripts for client and server sides.

Thanks in advance.

dhoomakethu, 2017-10-16:
@mariogirasole Thanks for the explanation but no attachments , please update your comments with the attachment

mariogirasole, 2017-10-16:
Here are my python codes for Server and Client Sides:

Server Side
=======
#!/usr/bin/env python
# -*- coding: utf-8 -*-

# Definições do programa
nSlaves = 3 # Número de eleavadores (Slaves) máx 10
master= [0]*1
master[0] = ['10.146.4.16',502,1] # Endereços do servidor (Master)
maceth = "b8:27:eb:bd:5a:3e" #endereço mac address do adaptador ethernet (proteção)
# License
mac = open('/sys/class/net/eth0/address').readline()
if (mac[0:17]) == maceth:
	#---------------------------------------------------------------------------# 
	# import the modbus libraries we need
	#---------------------------------------------------------------------------# 
	from pymodbus.server.async import StartTcpServer
	from pymodbus.device import ModbusDeviceIdentification
	from pymodbus.datastore import ModbusSequentialDataBlock
	from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext
	import time

	#---------------------------------------------------------------------------# 
	# configure the service logging
	#---------------------------------------------------------------------------# 
	import logging
	logging.basicConfig()
	log = logging.getLogger()
	log.setLevel(logging.DEBUG)

	#---------------------------------------------------------------------------# 
	# initialize your data store
	#---------------------------------------------------------------------------# 
	slavesdb = ModbusSlaveContext(
			di = ModbusSequentialDataBlock(0,[0]*60),
			co = ModbusSequentialDataBlock(0,[0]*10),
			hr = ModbusSequentialDataBlock(0,[0]*5),
			ir = ModbusSequentialDataBlock(0,[0]*5),zero_mode=True)
	
	context = ModbusServerContext(slaves=slavesdb, single=True)

	#---------------------------------------------------------------------------# 
	# initialize the server information
	#---------------------------------------------------------------------------# 
	identity = ModbusDeviceIdentification()
	identity.VendorName  = 'AZZURRA ENGENHARIA'
	identity.ProductCode = 'GW-MBTCP-MS'
	identity.VendorUrl   = 'www.azzurraengenharia.com'
	identity.ProductName = 'Gateway Modbus TCP Master/Slave'
	identity.ModelName   = 'GW-MBTCP-MS'
	identity.MajorMinorRevision = '1.0'

	#---------------------------------------------------------------------------# 
	# run the server
	#---------------------------------------------------------------------------# 
	time.sleep(10)
	StartTcpServer(context, identity=identity, address=(master[0][0],master[0][1]))

else:
	print("Dispositivo não licenciado para executar o aplicativo!\nContate o fornecedor:\nwww.azzurraengenharia.com")

CLIENT SIDE
========

#!/usr/bin/env python
# -*- coding: utf-8 -*-

# Definições do programa
[... 180 lines cut ...]
	valor = rq.getRegister(2)
	if valor == 1:
		x = 1
	else:
		x = 0
	ws = context[1].setValues(2,(s*12)+11,0)				
	

#except:
	# Falha de Comunicação com Slave
	if falha[s] > 5:
	   fu = context[1].setValues(2,(s*12)+10,[1])
	   log.debug('Falha Comunicação com Slave'+str(s)+' - IP: '+slave[s][0])
	else:
	   falha[s] = falha[s] + 1            

#---------------------------------------------------------------------------# 
# run process (each slaves)
#---------------------------------------------------------------------------# 
context = [0]
for i in range(nSlaves):
	loop = LoopingCall(f=updating_process, s=i)
	loop.start(cycle, now=False) # initially delay by time

reactor.run()

dhoomakethu, 2017-10-18:
@mariogirasole , are you expecting the values you read and write in `updating_process` on the client side to be reflecting on your server code (Server side) ? . Ideally, Your gateway client and server needs to be using the same context in order to get the correct values. I will give a try of your code this week and will update .

dhoomakethu, 2017-10-19:
@mariogirasole , unfortunately there is no straight forward solution for your problem, A remote slave context is one to one mapping of the datablocks on the remotes and it uses the regular read/write functions to read/write from the remote hosts (refer https://github.com/riptideio/pymodbus/blob/master/pymodbus/datastore/remote.py#L86) , Hence when you read a holding register in `updating_process` and write on to a input register/discrete input, the remote slave context uses the `write_registers` method which is invalid and the operation fails. You will probably have to come up with your own datastore/remote server context (refer https://github.com/riptideio/pymodbus/blob/master/examples/contrib/database-datastore.py or https://github.com/riptideio/pymodbus/blob/master/examples/contrib/redis-datastore.py or  https://github.com/riptideio/pymodbus/blob/master/examples/contrib/remote_server_context.py ) Where in your remote server reads, process and writes data to input/registers.

closed 2017-10-23 by dhoomakethu