pymodbus-dev/pymodbus #230: pymodbus RTU not working

candronikos, 2017-10-17:
### Versions

* Python: Python 2.7.13
* OS: Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
* Pymodbus: 1.3.2
* Modbus Hardware (if used): N/A

### Description

Trying to test the pymodbus driver in RTU mode. In an serial viewer no data is received. Testing raw data packets with pyserial works, setting the client in 'ascii' mode at least shows something's received. When set to RTU mode, nothing shows up. How can I verify that the driver is being used correctly.

### Code

```python
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
#import pymodbus.client.sync import ModbusTCPClient as ModbusClient

import logging
logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

client = ModbusClient(method='rtu', port='/dev/ttyS0', timeout=1)
#client = ModbusClient(method='ascii', port='/dev/ttyS0', timeout=1)
#client = ModbusClient(method='10.', )

client.connect()

rr = client.read_holding_registers(10,count=12,unit=5)

print(rr)

client.close()
```

### Output

```
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:send: 0x5 0x3 0x0 0xa 0x0 0xc 0x64 0x49
DEBUG:pymodbus.client.sync:will sleep to wait for 3.5 char
DEBUG:pymodbus.transaction:recv: 
DEBUG:pymodbus.transaction:getting transaction 1
Modbus Error: [Input/Output] No Response received from the remote unit
```

dhoomakethu, 2017-10-17:
Is there a physical modbus device connected on the port `/dev/ttyS0` ? Modbus RTU requires RS485 support, please check if Raspberry PI can work in RS485 mode , if not use USB to RS485 converters , which should work just fine.

candronikos, 2017-10-17:
Are you saying that 'rtu' needs RS485 and 'ascii' doesn't? My colleague has verified that 'ascii' mode works over the same port when doing the `rr = client.read_holding_registers(10,count=12,unit=5)`.

Also, how can I confirm that the RPI is using RS485? It's connected using the UART pins if that helps.

dhoomakethu, 2017-10-17:
Yes 'rtu' mode works only on RS485 , As far as I know Raspberry Pi does not support RS485 out of the box(correct me if I am wrong). You need something like this https://www.sparkfun.com/products/13706 to get the RS485 working.

closed 2017-10-18 by dhoomakethu